tbsdtv/linux_media #2: build error after 4d616f9

CvH, 2016-07-27:
I got this error after commit 4d616f9 (_update the si2168 snr,ber... interface_).
gcc 5.3 - cross compiling
```
/tbs-oss-2016-07-26/media_build/v4l/si2168.c: In function 'si2168_read_snr':
/tbs-oss-2016-07-26/media_build/v4l/si2168.c:201:5: error: 'struct si2168_dev' has no member named 'snr'
  dev->snr=52428;
     ^
/tbs-oss-2016-07-26/media_build/v4l/si2168.c:202:45: error: 'struct si2168_dev' has no member named 'snr'
  *snr = (dev->fe_status & FE_HAS_LOCK) ? dev->snr : 0;
                                             ^
/tbs-oss-2016-07-26/media_build/v4l/si2168.c: In function 'si2168_read_ucblocks':
/tbs-oss-2016-07-26/media_build/v4l/si2168.c:247:9: error: 'struct si2168_dev' has no member named 'stat_resp'
  if (dev->stat_resp & 0x10) {
         ^
/tbs-oss-2016-07-26/media_build/v4l/si2168.c: In function 'si2168_init':
/tbs-oss-2016-07-26/media_build/v4l/si2168.c:500:9: error: 'struct si2168_dev' has no member named 'warm'
  if (dev->warm) {
         ^
/tbs-oss-2016-07-26/media_build/v4l/si2168.c: In function 'si2168_sleep':
/tbs-oss-2016-07-26/media_build/v4l/si2168.c:738:9: error: 'struct si2168_dev' has no member named 'version'
  if (dev->version > ('B' << 24 | 4 << 16 | 0 << 8 | 11 << 0))
         ^
/tbs-oss-2016-07-26/media_build/v4l/si2168.c:739:6: error: 'struct si2168_dev' has no member named 'warm'
   dev->warm = false;
      ^
scripts/Makefile.build:264: recipe for target '/tbs-oss-2016-07-26/media_build/v4l/si2168.o' failed
make[4]: *** [/tbs-oss-2016-07-26/media_build/v4l/si2168.o] Error 1
```

tbsiptv, 2016-07-27:
Hi 

Sorry about this please update your code to the latest one
https://github.com/tbsdtv/linux_media/commit/a45e590dc6ba77655ad0042b62c1ce77e11a12d6

Thanks

Kind Regards

steven

CvH, 2016-07-27:
tx - worked like a charm !

closed 2016-07-27 by CvH