tbsdtv/linux_media #3: Not all the frontends are registering - Signal in astra cesbo not correct- frontend order

nightshadow36, 2016-08-28:
1: I have a mix of cards in my server 6905, 690, 6985.
I have a total of 8 tbs 6985 tuners 2 x cards as you can see only 1 is registering how can i fix this?
2: In astra cesbo the TBS 6985 is not showing the correct Signal / BER
3: Also the rontend does not stay in order on reboots or any time.
Thank you

13.376247]  [<ffffffffc06be87c>] saa716x_tbs6985_frontend_attach+0x7c/0x290 [saa716x_budget]
[   13.376325]  [<ffffffffc06be87c>] saa716x_tbs6985_frontend_attach+0x7c/0x290 [saa716x_budget]
[   13.376524]  [<ffffffffc06be87c>] saa716x_tbs6985_frontend_attach+0x7c/0x290 [saa716x_budget]
[   13.376592]  [<ffffffffc06be87c>] saa716x_tbs6985_frontend_attach+0x7c/0x290 [saa716x_budget]
[   13.667619] SAA716x Budget 0000:0b:00.0: DVB: registering adapter 12 frontend 0 (TurboSight TBS 6985 DVB-S/S2)...
[   37.306148] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.418134] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 13 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.537891] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 14 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.649365] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 15 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.769246] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 16 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.839632] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 17 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.910876] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 18 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.988692] TBSECP3 driver 0000:08:00.0: DVB: registering adapter 19 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   38.115250] TBSECP3 driver 0000:0a:00.0: DVB: registering adapter 20 frontend 0 (TurboSight TBS 6905 DVB-S/S2 )...
[   38.236218] TBSECP3 driver 0000:0a:00.0: DVB: registering adapter 21 frontend 0 (TurboSight TBS 6905 DVB-S/S2 )...

crazycat69, 2016-08-28:
default maximum DVB adapters overrided in media_build script, default changed to 48 - https://github.com/tbsdtv/media_build/commit/0e42dc6b0c4daf60fbf978b584985a4618bfc839

so maybe you have > 48 ? :)

nightshadow36, 2016-08-28:
I have 1 x 6909 / 2 x 6905 / 2 x 6985. That should be a total of 24 Frontends.

crazycat69, 2016-08-28:
Please attach full dmesg log

nightshadow36, 2016-08-28:
Hope this is the correct Log that you want. Thank you.

[dmesg.tar.gz](https://github.com/tbsdtv/linux_media/files/441219/dmesg.tar.gz)

crazycat69, 2016-08-29:
You can assign some adapter nums for certain adapter driver. Put this options to **/etc/modprobe.d/dvb.conf**

`options saa716x_core adapter_nr=0,1,2,3`

So saa716x-based 6985 always attached as adapter 0,1,2,3. Other FPGA-based cards 4,5,6,7......

nightshadow36, 2016-08-29:
Thanks i have tried before and again now but it does not work.
[image]

ovidiu31, 2016-09-03:
old hardware with new hardware... not the best suitable option ... there are quite major differences on how it was designed to work 6985 and how is supposed to work the 6905 or 6904 not to mention 6909. In theory should be ok ... but that is just theory

nightshadow36, 2016-09-04:
Thanks for the feedback, I have decided to update my cards

crazycat69, 2016-09-07:
> As you can see I made a mistake the TBS6985 do register but i get all the attach messages under?
"saa716x" Is this normal? Also issue 2 & 3 is there, I have tried updating.

i think need disable using I2C_MUX adapter inside tas2101 driver.

nightshadow36, 2016-09-07:
OK last question and thank you for your help.
"i think need disable using I2C_MUX adapter inside tas2101 driver."
What does that do? and how to find out the information to do it?
Thank you

crazycat69, 2016-09-07:
03a5a706b9b938c22e54432d3b530d653a390149

closed 2017-03-10 by crazycat69