tbsdtv/linux_media #4: TBS 6909 low sinal compared to 6905

nightshadow36, 2016-09-04:
6909 lower signal.

Test channel = Channel 5 using Astra Cesbo
Card 6909 Signal = 53% SNR = 50%
Card 6905 Signal = 62% SNR = 50%
The SNR is the same and the signal is a about 10% less on the 6909
Now the difference in signal is that due to the Card Tuner model or the Driver?
Thank you

steven1758, 2016-09-05:
Hi 

Thanks for your feedback ,6909 and 6905 using different tuner and demo so the signal strength and quality maybe different ,if there is no packet loss do not need to care about this .

Thank

crazycat69, 2016-09-05:
> Test channel = Channel 5 using Astra Cesbo
Card 6909 Signal = 53% SNR = 50%
Card 6905 Signal = 62% SNR = 50%
The SNR is the same and the signal is a about 10% less on the 6909
Now the difference in signal is that due to the Card Tuner model or the Driver?

Both demod drivers stv0910 and mxl5xx report legacy (dvb api v3) signal strength (%) as RFLevel(dBm) + 100, and SNR(%)=SNR(dB) * 5. Most important is SNR.

nightshadow36, 2016-09-05:
Ok so i dont need to care about the signal output and the snr is the most inportant.
But is the 6909 a good tuner?
I am asking because i am in spain and i am using a 2.4 dish  even with this i am on the frindge of reception. am waiting for my other 6909

crazycat69, 2016-09-07:
6909 not have any real switch or splitter :) MXL584 chip have four tuners and 8 demods. Each demod can be routed to any tuner (so any input). So "switch" inside 6909 is just "virtual switch" inside mx5xx frontend driver :) No any additional noise (signal loss) like real multiswitch.

About signal quality - current TBS cards have similar performance. This is stv0910-based 6903/5/8, tas2101-based 6902/4, 6910 and mxl584-based 6909. So anyway you have SNR ~10dB (~50%) with your dish.

closed 2016-11-27 by crazycat69