tbsdtv/linux_media #7: Mix cards TBSECP3 frontend 0 attach failed

nightshadow36, 2016-09-09:
Hello.
I have just upgraded with 2 x 6909 and 2 x 6905
With the 6905 I 
[image]
get TBSECP3 frontend 0 attach failed - TBSECP3 frontend 1 attach failed
Please see screen shot, any idea how to fix?
Thank you.

crazycat69, 2016-09-15:
please update drivers and check again.

victor-accarini, 2016-10-10:
Hi,

I bought a 6522 and a 6528 to try them out. But no matter each one I plug in I get the same error.
I tried with another machine with the same OS and I get the same error.
I'm running a CentOS7 3.10.0-327.36.1.el7.x86_64 and installed the latest drivers.
Can both of them be faulty? They are brand new.

crazycat69, 2016-10-10:
what error ?

victor-accarini, 2016-10-11:
`dmesg | grep 06:00.0:`

```
[    0.125564] pci 0000:06:00.0: [544d:6178] type 00 class 0x048000
[    0.125576] pci 0000:06:00.0: reg 0x10: [mem 0x92c00000-0x92c3ffff]
[    0.402474] pci 0000:06:00.0: Signaling PME through PCIe PME interrupt
[    9.023046] TBSECP3 driver 0000:06:00.0: enabling device (0140 -> 0142)
[    9.023091] TBSECP3 driver 0000:06:00.0: TurboSight TBS 6528 DVB-S/S2 + CI 
[    9.023189] TBSECP3 driver 0000:06:00.0: irq 30 for MSI/MSI-X
[    9.091826] TBSECP3 driver 0000:06:00.0: MAC address 00:22:ab:e0:65:45
[    9.092793] TBSECP3 driver 0000:06:00.0: TBSECP3 frontend 0 attach failed
[    9.092795] TBSECP3 driver 0000:06:00.0: frontend attach failed
[    9.092885] TBSECP3 driver 0000:06:00.0: adapter0 attach failed
[    9.092887] TBSECP3 driver 0000:06:00.0: TurboSight TBS 6528 DVB-S/S2 + CI : PCI 0000:06:00.0, IRQ 30, MMIO 0x92c00000
```

crazycat69, 2016-10-11:
maybe si2183 demod firmware not installed ?

victor-accarini, 2016-10-11:
Is this correct?

```
> find /lib/firmware/ -name *2183*
/lib/firmware/dvb-demod-si2183-b60-01.fw
```

crazycat69, 2016-10-11:
yes

victor-accarini, 2016-10-11:
I updated my kernel to 4.4 and the folder /dev/dvb/adapter# appeared, but I can't tune into a channel.
I tested my reception signal with another device and everything is ok.

sending dmesg before and after tunning.
[dmesg2.txt](https://github.com/tbsdtv/linux_media/files/522736/dmesg2.txt)
[dmesg.txt](https://github.com/tbsdtv/linux_media/files/522737/dmesg.txt)

crazycat69, 2016-10-11:
dvb adapter have two frontends - first is ter/cable, second is satellite

victor-accarini, 2016-10-11:
I'm using terrestrial (frontend0).

```
$ scan -a 0 -f 0 channel.conf
scanning channel.conf
using '/dev/dvb/adapter0/frontend0' and '/dev/dvb/adapter0/demux0'
ERROR: invalid enum value '6MHZ'
initial transponder 641142857 3 9 9 6 2 4 0
>>> tune to: 641142857:INVERSION_AUTO:BANDWIDTH_AUTO:FEC_AUTO:FEC_AUTO:QAM_AUTO:TRANSMISSION_MODE_AUTO:GUARD_INTERVAL_AUTO:HIERARCHY_NONE
WARNING: >>> tuning failed!!!
>>> tune to: 641142857:INVERSION_AUTO:BANDWIDTH_AUTO:FEC_AUTO:FEC_AUTO:QAM_AUTO:TRANSMISSION_MODE_AUTO:GUARD_INTERVAL_AUTO:HIERARCHY_NONE (tuning failed)
WARNING: >>> tuning failed!!!
ERROR: initial tuning failed
dumping lists (0 services)
Done.
```

```
$ cat channel.conf
T 641142857 6MHZ AUTO AUTO AUTO AUTO AUTO NONE
```

closed 2016-10-11 by nightshadow36

crazycat69, 2016-10-11:
some strange mux list :) at least you must specify freq, bandwidth and delivery system :)
i recommend use scan-s2 for scan - https://bitbucket.org/CrazyCat/scan-s2
and szap-s2/tzap-t2 for zapping - https://bitbucket.org/CrazyCat/szap-s2

victor-accarini, 2016-10-11:
I use the same configuration for my other ISDB-T(Brazil) devices and they are working...
I tried scan-s2 and got the same error as before. thanks for the recommendation :)

Changed 6MHZ to 6MHz to stop the error message.

crazycat69, 2016-10-11:
so you want scan isdb-t ?look like scan util not support it and si2183 multi-standard driver require specify delivery system.

crazycat69, 2016-10-11:
you can you dvb-fe-tool for changing current delivery system and scan with old app like scan
**dvb-fe-tool -a 0 -f 0 -d ISDB-T**
in my case is adapter 0, frontend 0

nightshadow36, 2016-10-11:
How do I stop receving emails?

Its driving me nuts

 

Email

victor-accarini, 2016-10-11:
@nightshadow36 github -> unsubscribe button (bottom-right panel)
@crazycat69 Ok, I will try it later

victor-accarini, 2016-11-04:
@crazycat69 I'm back. I bought a TBS6814 and was able to make it work on Ubuntu 16.04, but I still get an error when using my TBS6528.
I installed Ubuntu Server 16.04, installed the newest drivers and firmwares but I can't set the parameters to tune my 6528 board.

```
$ uname -a
Linux localhost 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:07:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

I'm trying to tune using `dvbv5-zap`, but when I execute it I get the following error on kernel.
```
$ dvbv5-zap -c channel.conf "EPTV Campinas" -P -r
$ cat /etc/var/syslog
...
kernel: [ 5977.720769] si2183 0-0067: err setting tuner params
kernel: [ 5977.721033] si2183 0-0067: set_params failed=-22
...
```

When I use scan I get an "tuning failed" error. I also executed with the `dvb-fe-tool` command.