tbsdtv/linux_media #8: Assign inputs to tuner pairs

fe31nz, 2016-10-05:
Here in New Zealand, we have only one real satellite source, Optus D1, and all the transponders on it that are used for free-to-air channels (Freeview) or pay channels (Sky TV) are transmitted from it using horizontal polarisation.  Our receiving dishes are normally using just a single local oscillator - that is what Sky TV installs when you sign up with them and also what most third party installers use.  The latest dish variant (which I have) has two LNBs, so that you Sky can sell you a second set-top box without having to change the dish.  So I currently have one LNB connected to my STB, and the other to a splitter and then to two DVB-S2 tuners in my MythTV PC.  I have bought a TBS6909 8 tuner DVB-S2 card to replace the two separate DVB-S2 tuners, but if I use the current drivers in mode=1, I would have to provide a signal on all four of the card's inputs.  To do that, I would need to split the signal in a 4-way splitter, which might cause the signal level to be too low for good reception.  Since the MLX5xx chip used on the TBS6909 allows internal mapping of the inputs to the tuner pairs, I thought it would be much better if I could just modify the drivers to allow me to use a single input to run all 8 tuners.  When I looked at the mlx5xx.c file, I found that was in fact pretty easy to do, and I now have it working.  There is an additional "inputs" parameter on the mlx5xx driver which allows assigning of the inputs to the tuner pairs when using mode=1, but defaults to the current mode=1 setup if it is missing:

`root@lith:~/projects/tbs6909-drivers/media_build# modinfo mxl5xx
filename:       /lib/modules/4.4.0-38-generic/kernel/drivers/media/dvb-frontends/mxl5xx.ko
license:        GPL
author:         Ralph Metzler
description:    MXL5XX driver
srcversion:     6D1D7AE4BB71F44228F09B0
depends:        
vermagic:       4.4.0-38-generic SMP mod_unload modversions 
parm:           mode:Multi-switch mode: 0=quattro/quad 1=normal direct connection (int)
parm:           inputs:Input number (0-3) for each of the 4 tuner pairs.  Default is 0,1,2,3.  Only used if mode=1. (array of short)
`
I have attached a git diff of my changes.
[diff.txt](https://github.com/tbsdtv/linux_media/files/510698/diff.txt)

crazycat69, 2016-10-05:
You not need mode 1. Just connect cable to H/low or H/High input (depend from transponders freq if used universal Ku LNB) and mxl5xx driver route all logical demods to single input.

crazycat69, 2016-10-05:
Ok, so all freq Sky@ 160e is high-band, and your LNB is single LOF. So connect to any Horizontal input.

fe31nz, 2016-10-06:
Since we do not use multiswitches here in New Zealand, I had no idea how they work, so I had completely missed the idea of using mode=0 and the H/L input (input 2).  I have now tested this and it works exactly as I want it to.  The H/H input (input 3) does not work, as that has the 22 kHz tone on it, and while the LNBs on our dishes only have one local oscillator, they do respond to the 22 kHz tone by turning that local oscillator off, depite there not being another oscillator to turn on.

steven1758, 2016-10-08:
Do you mean that you only have one cable from your LNB so the low band and high band can not used at the same time?

fe31nz, 2016-10-08:
There are two cables, from two separate LNBs.  Each LNB has only one local oscillator, so there is no high band support as such - we only get the frequency range supported by the one oscillator.  That frequency range covers all the transponders on Optus D1 that are supported as being available in New Zealand.

closed 2016-11-27 by crazycat69