tbsdtv/linux_media #9: Not compilable with kernel 4.8

martinjakl, 2016-10-22:
It's not possible to compile drivers with kernel 4.8 it looks like that dma attributes change from structure to unsigned int.

postmaxin, 2016-10-23:
This is going to affect everybody using Ubuntu 16.04, which is especially bad because Ubuntu 16.04 is an LTS release.

This problem could have been mitigated if the TBSDTV drivers were being contributed to Ubuntu as a DKMS kernel module package. If TBS had this package in the Ubuntu repo, the Ubuntu kernel team's autobuild would have picked up on the breakage and either notified TBS or contributed a fix. As things are right now, every TBSDTV user that uses 16.04 is in for a rude surprise in the next few days.

crazycat69, 2016-10-23:
Just need merge with upstream media_tree and media_build.
But this make troubles for <4.6 users because cxd2820r rewriten with RegMap API and all cxe2820r-depend modules was disabled for <4.6 kernels ( including saa716x_budget for many TBS PCI-E cards).
So wait next LTS kernel for 14.04,  16.04.

postmaxin, 2016-10-23:
crazycat, for some reason merge + distclean + rebuild wasn't enough. I had to blow the repos away and check them out again. But I'm good now.

This would be really nice to have available as DKMS on a PPA though, and to eventually get it into ubuntu mainline.  plug+play is important! :)

crazycat69, 2016-10-23:
If someone want maintain some DKMS on PPA - always welcome :)

postmaxin, 2016-10-23:
In order to do that sanely,

1. This code needs to be buildable with just your latest linux-headers package and *one* git repo, not two
2. "make dist" needs to create a tarball of that repo that anybody can do #1 with, so that git is no longer a dependency to build your code.

Can you do that?

CHBMB, 2016-10-23:
I compile these for Unraid and the most recent kernel version is 4.8.2 on their V6.3 RC2 release and I am also seeing this issue.

If it is at all helpful here is a log 
http://pastebin.com/raw/fFuW34nv

postmaxin, 2016-10-24:
crazycat? Are you willing to normalize the kernel build process a little bit as I detailed above? If you do that it would go a long way towards somebody (maybe even me!) DKMS-ifying it so it stands a chance of making it into a PPA or ubuntu mainline.

davidrnewman, 2016-10-29:
It compiles under 4.8.0, but the driver doesn't work on a TBS6285. When set to scan DTV channels in the UK, it cannot find any channels.

CHBMB, 2016-10-29:
@davidrnewman I couldn't get it to compile correctly without a lot of errors on 4.8.2....  Did you check the logs?  It finishes the compile process but not without a lot of errors.

davidrnewman, 2016-10-29:
I was wrong. There are lots of warnings during compilation, such as:

warning: 'struct dma_attrs'declared inside parameter list will not be visible outside of this definition or declaration
  void  *(*alloc)(struct device *dev, const struct dma_attrs *attrs,

and one actual error:

/home/dave/src/TV/tbsdtv/media_build/v4l/videobuf2-core.c: In function '__vb2_buf_mem_alloc':
/home/dave/src/TV/tbsdtv/media_build/v4l/videobuf2-core.c:211:5: error: passing argument 2 of '_q->mem_ops->alloc' from incompatible pointer type [-Werror=incompatible-pointer-types]
     q->dma_attrs, size, dma_dir, q->gfp_flags);

crazycat69, 2016-11-02:
media_build updated, minimal version for cxd2820r-depend modules changed from 4.6 to 4.4.
linux_media merged with 4.8 kernel sources. some conflicts in cxd2820r and cx231xx sources resolved, need more check with TBS 528x, 5990.

davidrnewman, 2016-11-03:
It compiled OK against kernel 4.8.0, without errors but with lots of warnings. 

[make.error.txt](https://github.com/tbsdtv/linux_media/files/568021/make.error.txt)

When I tried to get the TBS6285 card to scan DVB channels in the UK (from mythtv_setup) it couldn't find any channels. I notices that the signal strength indicators were at 40% to 70%, much lower than when scanning under previous kernels with the closed source driver.

crazycat69, 2016-11-03:
warnings is ok :) your problems with si2168 demod driver maybe related to app (need set delivery system DVB-T or T2 or C)

davidrnewman, 2016-11-06:
mythtv-setup detects that the card is type DVB-T/T2/C, subtype DVB-T2. That is OK. But when I run a channel scan I get multiple errors like:

DiSEqCDevTree: No device tree for cardid 1
Failed to get status#012#011#011#011eno: Remote I/O error (121)

Those could be firmware errors. How do I make sure the right firmware for TBS6285 is downloaded?

crazycat69, 2016-11-06:
Check kernel log (dmesg).

CHBMB, 2016-11-06:
Can confirm these compiled fine for me using a TBS 6205 on kernel version 4.8.4

@crazycat69 I've let TBS know that their closed source drivers aren't working on kernel 4.8+ at the moment and I was using your CrazyCat build as well.  Do you have an opensource version of your CrazyCat build with the extra drivers?  Thanks for all your work on the DVB stuff, awesome!

crazycat69, 2016-11-06:
what my build ?

CHBMB, 2016-11-08:
Sorry @crazycat69 I wasn't clear.

This version is compiling fine on kernel 4.8.4

I have been using your CrazyCat drivers from [here](https://bitbucket.org/CrazyCat/linux-tbs-drivers/), but wondered if there was a version for the opensource TBS drivers.  

The reason I ask is I actually compile a number of builds for Unraid users, of which yours is one.  Currently I've been compiling stock in kernel stuff with firmware from LibreELEC, TBS Official Closed Source (Both DVB-S/T & DVB-C), Digital Devices (Github), TBS OpenSource & your version of the TBS drivers.

davidrnewman, 2016-11-08:
So which versions work in kernet 4.8.*? The one here? The one in the ljalves tree? Something downloaded from CrazyCat?

In particular, which ones will not just compile, but also work to scan channels then demodulate them in weak signal strength areas?

CHBMB, 2016-11-08:
So currently, TBS open source are working for me, new version of closed source are available here, I'm just compiling them now and seem to be fine.  Got them from [here](http://www.tbsiptv.com/download/common/tbs-linux-drivers_v161031.zip).  Will try the CrazyCat bitbucket drivers shortly, but long story short, they're all compiling fine at the moment, and opensource are working fine for me.  Obviously haven't tested the others yet.

crazycat69, 2016-11-09:
"semi closed-source" linux-tbs-driver really not updated since april 2016, just pre-processor workarounds for new kernel compatibility. I think this must be canceled finally and now disable access to [my linux-tbs-drivers fork repo](https://bitbucket.org/CrazyCat/linux-tbs-drivers) :)

CHBMB, 2016-11-09:
@crazycat69 OK, thanks for clearing that up and for all the hard work you put in.

davidrnewman, 2016-11-11:
I have switched to the closed source version, because its AGC is working for me, while this open source version isn't sensitive enough to even find channels on a channel scan. One of the make error warnings I posted above refers to AGC.

closed 2016-11-27 by crazycat69