tbsdtv/linux_media #14: Please help how to install on centos 6

nightshadow36, 2016-11-04:
Hi.
I have a server that i changed from ubuntu to centos as ubuntu always updates the kernal
I am unable to get it installed on centos using the guide.
Is there anything extra i need to install or add to get it installed?
Thank you

steven1758, 2016-11-07:
Hi 

Please have a try lke this :
1.update to new kernel
#rpm --import https://www.elrepo.org/RPM-GPG-KEY-elrepo.org
#rpm -Uvh http://www.elrepo.org/elrepo-release-7.0-2.el7.elrepo.noarch.rpm
#yum --enablerepo=elrepo-kernel install kernel-lt-devel kernel-lt -y
#grub2-set-default 0 

2.
#reboot

3.
#yum -y install perl-devel perl-ExtUtils-CBuilder perl-ExtUtils-MakeMaker
#yum -y install patchutils
#yum -y install patch

4.
#wget https://sourceforge.net/projects/moiproamd/files/Files/Digest-SHA-5.93.tar.gz/download
#tar xvf   Digest-SHA-5.95.tar.gz
#perl Makefile.PL
#make && make install

5.
#wget https://sourceforge.net/projects/moiproamd/files/Files/perl-proc-processtable-master.zip/download
#unzip perl-proc-processtable-master.zip
#perl Makefile.PL
#make && make install

6.Install the open sourece driver:
check this 
https://github.com/tbsdtv/linux_media/wiki

thanks

ovidiu31, 2016-11-22:
@steven1758 

its working Steve , you can close this issue.

closed 2016-11-27 by crazycat69