tbsdtv/linux_media #15: ISDBT: Err setting tuner params

victor-accarini, 2016-11-08:
I have two TBS6522 and two TBS6528, I can't make them tune into an ISDBT channel in Linux.

I tried installing them on 5 different machines with fresh Ubuntu and CentOS install, and with different kernels going from 3.10 to 4.4.

Using kernel 4.4 and Ubuntu the driver was able to register the adapters and frontends but is unable to tune/lock into any frequency.

```
$ uname -a
Linux localhost 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:07:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

Here I'm using dvbv5-zap, but I tried with several other tools and all of them get the same result.
```
$ dvbv5-zap -c channel.conf "EPTV Campinas" -P -r
$ cat /etc/var/syslog
...
kernel: [ 5977.720769] si2183 0-0067: err setting tuner params
kernel: [ 5977.721033] si2183 0-0067: set_params failed=-22
...
```

When I use scan I get an "tuning failed" error. I also executed with the `dvb-fe-tool -d ISDBT` command.
I never worked with linux drivers but if someone can point where I have to look, maybe I can find the problem...

crazycat69, 2016-11-09:
Look like si2157 tuner driver not support ISDB-T - so you have tuner setup error :) And look like Si2183 demod driver support ISDB-T, but not tested with real signal :) Need remote access to your PC with ISDB-T connected ant/signal for debug and driver improvement. Before i have good experience with DVB-C annex B support with South American users.

victor-accarini, 2016-11-09:
Ok, I will set up a machine for you and send how to connect.
Can I send in your email crazycat69@narod.ru?

crazycat69, 2016-11-09:
yes

closed 2016-11-09 by crazycat69