tbsdtv/linux_media #17: TBS6909 ignores satellite input parameter (-S 2,3,4) on high band frequencies

tonygables, 2016-11-25:
Hello,

TBS6909 is in multiswitch mode (mode 0). It can lock to different satellites (-S 2, -S 3, -S 4) when the frequency is low (<11700 MHz). However when a high frequency (>11700) is entered, -S parameter is ignored and the adapter tries to lock with -S 1.

crazycat69, 2016-11-25:
This very bad idea using DiseqC 1.0 cmd in mode 0 :) Usual before send diseqc cmd forced LNB voltage 18V and off 22KHz continous tone, so mxl5xx routed to H/Low input

tonygables, 2016-11-27:
Perhaps I couldn't make myself clear. Please see below for details:

Satellite 2 + Low Band locks. Satellite 2 + High Band doesn't lock.

This works: dvblast -a 0 _**-f 10992000**_ -s 27500000 -v 13 _**-S 2**_
This doesn't: dvblast -a 0 _**-f 12226000**_ -s 27500000 -v 13 _**-S 2**_

Satellite 1 locks both for high band and low band.

This works: dvblast -a 0 _**-f 11096000**_ -s 30000000 -v 13 _**-S 1**_
This works: dvblast -a 0 _**-f 12034000**_ -s 27500000 -v 13 _**-S 1**_

crazycat69, 2016-11-27:
So as i say - buggy  diseqc support in mode 0.
Use mode 1 if you want connect to multiswitch with 2-4 satellites.