tbsdtv/linux_media #18: TBS 6522: Add DVB-S2X Support (feature request)

caveman99, 2016-11-30:
Since the driver installer replaces the media subsystem of the kernel anyway, can DVB-S2X Support added without too much hassle? As i understand it, this would mean adding new types of mudulation, rolloff and FEC values but no changes to the base DVB-S2 system.

crazycat69, 2016-11-30:
So add this to dvb headers, anyway si2183 frontend driver not need it because always autodetect modulation params and s2x is some part of s2 (same delivery system).

caveman99, 2016-12-06:
Thanks for the suggestion, you are partly right about the autodetect. I use TVHeadend on the machine, and the tuners exported via sat-ip can read the test transponder on 19,2E successfully (using TransEdit beta from DVBViewer). Using the mux settings in TVHeadend itself does not work, it can somehow read the transponder but doesa not find usable streams. Scanning result ist 'partially ok'. I might need to dig deeper into this issue but as you mentioned it should (tm) work out of the box even without adding the new parameters.

closed 2017-03-10 by crazycat69