tbsdtv/linux_media #20: Patch for kernel 4.9

Xxyzzy, 2016-12-12:
At your own risk. No support.
[patch-kernel-4.9.txt](https://github.com/tbsdtv/linux_media/files/646646/patch-kernel-4.9.txt)

CvH, 2016-12-12:
I guess `CONFIG_SAA716X_SUPPORT` is still not working with that ?

crazycat69, 2016-12-12:
tbsdtv/media_tree and tbsdtv/media_build updated with latest for 4.9 compatibility.

CvH, 2016-12-12:
@crazycat69 does the mismatch in `drivers/media/dvb-frontends/stv090x.c` get fixed ? 
besides that, the patch against plain 4.9 works after i added http://pastebin.com/JN8smBEy to the master branch 👍

crazycat69, 2016-12-13:
All conflicts resolved
5c21ed51b3b110dcad6671a522a9bef1b9d2c904

CvH, 2016-12-13:
sorry expressed too imprecise, if I create a patch `git diff master latest > ../tbs-49.patch` this patch can't be merged wit kernel 4.9 because of some mismatch in `stv090x.c` and `dw2102.c`. 
The master branch has not the 4.9 files for this two files (it includes addition commits).

CvH, 2016-12-13:
@crazycat69 SAA716x is still (at least 4.8 won't work either) not working -> [error](http://pastebin.com/bVRetWrL) ([full log](http://pastebin.com/Y6etshst))
if I build without `SAA716x` it works like a charm 👍

closed 2017-03-10 by crazycat69