tbsdtv/linux_media #21: Installation error

YaPaY, 2016-12-14:
git clone https://github.com/tbsdtv/media_build.git
Cloning into 'media_build'...
fatal: unable to access 'https://github.com/tbsdtv/media_build.git/': Peer's Certificate has expired.

the other question to work my card with universal multiswitch which key should I use. According to the tutorial there is support only for Quattro/Quad and LNB connection. Can't I use this driver wih Multiswitch?

steven1758, 2016-12-15:
Hi YaPay 

Please have a try this offline package : http://www.tbsdtv.com/download/document/linux/media_build-1215.tar.bz2

YaPaY, 2016-12-15:
> Hi YaPay
>
> Please have a try this offline package : http://www.tbsdtv.com/
> download/document/linux/media_build-1215.tar.bz2

Hi steven,

How can I install this. I installed ./install and copied firmwarebut after
reboot i cant see any dvb card. could you please help me about the
installation or is there any possibility to realize all command from you
via putty.

many thanks

YaPaY, 2016-12-15:
hi steven

./install.sh still compiling the driver (more than 3o minutes) i think something goes wrong