tbsdtv/linux_media #24: TBS 5927 does not load firmware

dem0nlord, 2016-12-24:
Worked/still works with previous 4.8 kernel, but not since 4.9 was put into the tree.

Dec 24 07:00:54 backend kernel: [    4.676551] usbcore: registered new interface driver tbs5927
Dec 24 07:10:30 backend kernel: [  590.171382] usb 1-4: New USB device found, idVendor=734c, idProduct=5927
Dec 24 07:10:30 backend kernel: [  590.172544] dvb-usb: found a 'TurboSight TBS 5927 DVB-S/S2' in cold state, will try to load a firmware
Dec 24 07:10:30 backend kernel: [  590.172551] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5927.fw'
Dec 24 07:10:30 backend kernel: [  590.172553] tbs5927: start downloading TBS5927 firmware
Dec 24 07:10:30 backend kernel: [  590.172565] tbs5927: error while transferring firmware
Dec 24 07:10:30 backend kernel: [  590.172628] tbs5927: could not restart the USB controller CPU.
Dec 24 07:10:30 backend kernel: [  590.172679] tbs5927: could not restart the USB controller CPU.

crazycat69, 2016-12-24:
Need disable Virtually-mapped stack feature in 4.9+ kernel.
https://github.com/LibreELEC/LibreELEC.tv/pull/1076
https://github.com/LibreELEC/LibreELEC.tv/commit/93a3df396ef89357c1c1fc980345d994a3842ff6

dem0nlord, 2016-12-25:
Thanks, that took care of it. Since you don't have a .config in your tree, nothing to do here so I will close.

closed 2016-12-25 by dem0nlord