tbsdtv/linux_media #25: Does not compile for Linux Kernel 4.8.0-2

Hoeze, 2016-12-26:
Hi,
I'm using Debian with Kernel 4.8.0-2 and tried to install this driver as described in the wiki.
However, I just get a message that patches could not be applied when I run 'make'.

Using the 'master'-branch instead of 'latest' seems to work, as I can compile and install, but when I run 'make reload' a big proportion of the modules does not exist.
Especially when i run 'modprobe saa716x_budget int_type=1' the module does not exist.

crazycat69, 2017-01-03:
You must use latest branch. Master just official kernel + media_tree without tbs support.

Hoeze, 2017-02-26:
Yes, i tried lastest, but it still does not compile.

Latest Debian Kernel 4.9.0-1 (4.9.6-3):
http://pastebin.com/hxvPi9A5

Same problem with Kernel 4.8...

closed 2017-03-10 by crazycat69