tbsdtv/linux_media #26: TBS 6285 not scanning DVB-C

jgaskell, 2017-01-03:
Setup: unRAID 6.2.4 w/ Open Source drivers.

Issue: Scanning the DVB-C source, no information back from TVHeadend.
Action 1:  Reverted to a Windows 10 machine, running NextPVR on the same card, same settings - and it worked, pulling down several hundred channels.

Action 2: Reverted to unRAID 6.2.4 compiled with the TBS Official DVB-C drivers and retried. Success. Same amount of channels have been retrieved as per the Windows instance. 

Conclusion: So, there's some kind of bug in the open source drivers for the TBS 6285 card in DVB-C mode. The error messages consistently stated a Poll Timeout.

Additional Info:
Frequency of Mux: 571000000
Symbol Rate: 6952000
Delivery: DVB-C

Drivers - latest used as of 23/12/2016.

I'm now running unRAID 6.2.4 with the TBS Official drivers for over 24 hours it's solid, but it's not a continuing solution for me, as I will also be putting some DVB-S signals in there and I'm not sure if the DVB-C drivers will work with the DVB-S. 

So, if anyone has any suggestions on how to implement the Open Source drivers and have them work with a DVB-C source and also with a DVB-S and DVB-T, I'd be most grateful.

crazycat69, 2017-01-03:
See more messages in dmesg. Maybe tuner/demod firmware not nstalled ?

jgaskell, 2017-01-03:
I'll have to switch the drivers out again (can only do it when the missus is out!) to get you a dmesg output. But, with the Open Source drivers, TVHeadend did recognise the card with it's DVB-T, ATSC-C and DVB-C capabilities. When I moved to the current official drivers, it is only recognised as a DVB-C card.

jgaskell, 2017-01-20:
Ok, I've done a dmesg output... 

[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CBFFF write-protect
[    0.000000]   CC000-DFFFF uncachable
[    0.000000]   E0000-E7FFF write-through
[    0.000000]   E8000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask F800000000 write-back
[    0.000000]   1 base 0800000000 mask FC00000000 write-back
[    0.000000]   2 base 0C00000000 mask FFC0000000 write-back
[    0.000000]   3 base 00C0000000 mask FFC0000000 uncachable
[    0.000000]   4 base 00BF800000 mask FFFF800000 uncachable
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000]   8 disabled
[    0.000000]   9 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] e820: update [mem 0xbf800000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xbf780 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [ffff                                                                        8800000ff780]
[    0.000000] Base memory trampoline at [ffff880000093000] 93000 size 24576
[    0.000000] Using GB pages for direct mapping
[    0.000000] BRK [0x01a73000, 0x01a73fff] PGTABLE
[    0.000000] BRK [0x01a74000, 0x01a74fff] PGTABLE
[    0.000000] BRK [0x01a75000, 0x01a75fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x7ad8f000-0x7fffffff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FAC50 000024 (v02 ACPIAM)
[    0.000000] ACPI: XSDT 0x00000000BF790100 00008C (v01 SMCI            2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: FACP 0x00000000BF790290 0000F4 (v03 111612 FACP1749 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI BIOS Warning (bug): 32/64X length mismatch in FADT/Gpe0Block                                                                        : 128/64 (20150930/tbfadt-623)
[    0.000000] ACPI: DSDT 0x00000000BF7906A0 00664D (v01 10006  10006000 0000000                                                                        0 INTL 20051117)
[    0.000000] ACPI: FACS 0x00000000BF79E000 000040
[    0.000000] ACPI: FACS 0x00000000BF79E000 000040
[    0.000000] ACPI: APIC 0x00000000BF790390 00011E (v01 111612 APIC1749 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: MCFG 0x00000000BF7904B0 00003C (v01 111612 OEMMCFG  2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: SLIT 0x00000000BF7904F0 000030 (v01 111612 OEMSLIT  2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: OEMB 0x00000000BF79E040 000086 (v01 111612 OEMB1749 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: SRAT 0x00000000BF79A6A0 000250 (v01 111612 OEMSRAT  0000000                                                                        1 INTL 00000001)
[    0.000000] ACPI: HPET 0x00000000BF79A8F0 000038 (v01 111612 OEMHPET  2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: DMAR 0x00000000BF79E0D0 000128 (v01 AMI    OEMDMAR  0000000                                                                        1 MSFT 00000097)
[    0.000000] ACPI: SSDT 0x00000000BF7A35F0 000363 (v01 DpgPmm CpuPm    0000001                                                                        2 INTL 20051117)
[    0.000000] ACPI: EINJ 0x00000000BF79A930 000130 (v01 AMIER  AMI_EINJ 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: BERT 0x00000000BF79AAC0 000030 (v01 AMIER  AMI_BERT 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: ERST 0x00000000BF79AAF0 0001B0 (v01 AMIER  AMI_ERST 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: HEST 0x00000000BF79ACA0 0000A8 (v01 AMIER  ABC_HEST 2012111                                                                        6 MSFT 00000097)
[    0.000000] ACPI: Local APIC address 0xfee00000
[    0.000000] SRAT: PXM 0 -> APIC 0x00 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x02 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x04 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x10 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x12 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x14 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x01 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x03 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x05 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x11 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x13 -> Node 0
[    0.000000] SRAT: PXM 0 -> APIC 0x15 -> Node 0
[    0.000000] SRAT: PXM 1 -> APIC 0x20 -> Node 1
[    0.000000] SRAT: PXM 1 -> APIC 0x22 -> Node 1
[    0.000000] SRAT: PXM 1 -> APIC 0x24 -> Node 1
[    0.000000] SRAT: PXM 1 -> APIC 0x30 -> Node 1
[    0.000000] SRAT: PXM 1 -> APIC 0x32 -> Node 1
[    0.000000] SRAT: PXM 1 -> APIC 0x34 -> Node 1
[    0.000000] SRAT: PXM 1 -> APIC 0x21 -> Node 1
[... cut ...]
[  187.385003] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b                                                                        40-01.fw'
[  187.797697] si2168 2-0066: downloading firmware from file 'dvb-demod-si2168-b                                                                        40-01.fw'
[  188.210802] si2168 2-0064: downloading firmware from file 'dvb-demod-si2168-b                                                                        40-01.fw'
[  188.623888] si2168 2-0064: downloading firmware from file 'dvb-demod-si2168-b                                                                        40-01.fw'

jgaskell, 2017-01-24:
Anybody able to make anything of this? Thanks in advance.

ovidiu31, 2017-01-30:
@jgaskell 
Crazycat asked you an plain simple question regarding firmware install .... still no answer from you.

the firmware procedure below 

Firmware

wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

jgaskell, 2017-02-03:
Installed the firmware on the Open Source as above - still no joy.
DMESG: http://pastebin.com/fV6ZwPzX

TVHeadend output: http://pastebin.com/fHH0Pzkd

I've now reverted back to the Official Drivers and all is good.

ovidiu31, 2017-02-03:
@jgaskell 

I think 6285 not yet supported by the oss drivers.
The chinese engineers are still in holiday till monday (Chinese New year)

jgaskell, 2017-02-07:
That's odd. I'm sure I saw it in the Wiki on LinuxTV that this card was supported.

CvH, 2017-02-07:
@ovidiu31 https://github.com/tbsdtv/linux_media/wiki
[image]

:)

jgaskell, 2017-02-12:
Don't suppose there's anything else I can try?

bas-t, 2017-03-27:
You can try to build a vanilla kernel with the driver in-tree, or if you are running debian or ubuntu, you can recompile your stock kernel with the driver in-tree. I've been doing so for several years and it always worked for me. I've got 3x TBS6285 up and running.

Look at https://github.com/bas-t/dvbloopback/blob/master/configure

lines 886 to 945 will give you an idea of how I'm used to do this. Mind you, I use a stripped down version of saa716x, it only supports TBS 6281 and TBS 6285

Good luck!

jgaskell, 2018-10-05:
I can confirm the latest pull I made (not just the latest commit) has fixed this. I had a year or so away from these drivers and decided to try again. All good and working well. Thank you so much,

closed 2018-10-05 by jgaskell