tbsdtv/linux_media #27: Recompile on Kernel update

golya87, 2017-01-03:
Hello Guys,

Is there a faster way to get media tree after kernel update, than a script that checks the right folders and starts compiling if neccessary?
I've read in other forums, the media tree is too big for dkms.

crazycat69, 2017-01-03:
offline.sh script in media_build make archive with offline sources + backport patches. So need only compile and install without online access to git repo or source snapshot.

golya87, 2017-01-03:
Sound good, but is there a way to "recycle" the previously compiled parts from the old kernel?

closed 2017-03-10 by crazycat69

491994, 2018-12-26:
Hello All,
            Tried to install the driver of TBS5927 in centos 7 kenel version 4.4 and get error before the installation which is-

/root/tbsdriver/media_build/v4l/videobuf-dma-sg.c:188:9: error: too many arguments to function 'get_user_pages'
         dma->pages, NULL);

what to do next for this?

eardkdw, 2019-01-25:
I have the same issue as 491994. CentOS 7 with `4.4.171-1.el7.elrepo.x86_64` .