tbsdtv/linux_media #28: "ts discontinuity" issue with TBS6991SE

popraduvmp, 2017-01-08:
Hi,

I have the same issues as seen in https://github.com/tbsdtv/linux_media/issues/16 but for TBS6991SE but only for some channels such as: AutoMotorSport HD on S16E0 11470 V 30000. I get around 500-600 discontinuities per second.
The same channels tunned on a 6909 works without any issues.

MOI Pro AMD, FW: 2.0.0.1
OS: CentOS Linux release 7.2.1511
Kernel: 4.4.19-1.el7.elrepo.x86_64
DVBlast 3.1

popraduvmp, 2017-01-19:
@crazycat69 
I'll provide some more info in order to help with testing the issue:
command:
`dvblast -a 2 -f 11470000 -s 30000000 -v 13 -m qpsk -c adapter2.cfg -U -W -Y -i 1`

adapter2.cfg:
`127.0.0.1:10204         1       610`

I attached a few seconds of output.
[output.txt](https://github.com/tbsdtv/linux_media/files/716295/output.txt)

BTW: I also tested with the proprietary TBS drivers and I see the same issues with TBS6991SE. With TBS6909 everything works ok with both drivers.

Thank you.

crazycat69, 2017-01-19:
Maybe high-bitrate transponder issue with serial TS bus.

popraduvmp, 2017-01-19:
Thanks for the quick reply, is there any way I can fix this?