tbsdtv/linux_media #29: TBS5281 card DVB-C network: invalid checksum problem

schneci, 2017-01-24:
I'm using a TBS5281 card with Tvheadend 4.2 for receiving DVB-C channels and the Silicon Labs Si2168 opensource TBS driver at most of the channels generate the following error messages in log file of Tvheadend:

> TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2VIDEO @ #201 Continuity counter error (total 1557)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2AUDIO @ #210 Continuity counter error (total 252)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: TELETEXT @ #220 Continuity counter error (total 191)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2AUDIO @ #211 Continuity counter error (total 236)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2VIDEO @ #201 Corrupted PES header (errors 1)
pmt: 250MHz in DVB-C network: invalid checksum (len 55, errors 1)
eit: 250MHz in DVB-C network: invalid checksum (len 1718, errors 40)

If I use the proprietary TBS driver everything works correctly. Could help somebody to me.

BR,
Schneci

jgaskell, 2017-01-30:
I'm having issues with the Open Source driver as well, but perfect with the stock TBS drivers - link here for info: https://github.com/tbsdtv/linux_media/issues/26

It might be worth you dumping a dmesg output to see if it works correctly.

ovidiu31, 2017-01-30:
@jgaskell 
@schneci 
both drivers open source and closed source are TBS proprietary.

To be able to help and assist with the issues the informations provided need to be as much complete as possible 

For example

1) kernel version used
uname -r

2) install pastebinit and use the folowing command to paste link to the kernel logs

dmesg | pastebinit

It will output you an hyperlink which can be copy pasted into the raised issue and prevent crazy scrolling around into useless log entries.

schneci, 2017-01-30:
My problem has solved by using the following options in Libreelec 7.95.1 version which uses 4.9x kernel version:

`options cx231xx transfer_mode=0`

In Libreelec or Openelec environment you can set this option using following command:

`echo "options cx231xx transfer_mode=0" > /storage/.config/modprobe.d/cx231xx.conf`

closed 2017-02-02 by crazycat69