tbsdtv/linux_media #32: TBS 5680 - Add support in driver

undertextz, 2017-02-02:
Is it possible to add support for the [TBS 5680 DVB-C USB Tuner](http://www.tbsdtv.com/products/tbs5680-dvb-c-tv-tuner-ci-usb.html)? I ask nicely on my knees :)

Currently using LibreELEC 7.95.1 and it would be lovely to be able to use the 5680 out-of-the-box with their releases in the future. Log (dmesg) of the non-detected 5680 here: http://pastebin.com/JwybV0wa

ovidiu31, 2017-02-03:
@undertextz 
I think there are few special builds of LE with included TBS drivers ... check the LE forum for those images.

CvH, 2017-02-03:
@ovidiu31
if there is no support at `github.com/tbsdtv/linux_media/` there is no support at LE ...
afaik they are not supported atm here (LE official is using media_build with tbs drivers till they get hopefully upstreamed to the kernel) so nothing we can add even if we want

closed 2017-03-10 by crazycat69

HenkPoley, 2020-04-13:
I notice that using `./build` from [tbsdtv/media_build](https://github.com/tbsdtv/media_build) does not seem to install the TBS5680 driver.

And I'd swear I used to have a working driver using either this, or: https://bitbucket.org/CrazyCat/linux-tbs-drivers/src/master/

Could you point me how to fix this?