tbsdtv/linux_media #34: Compile fails with Kernel 4.10.0

caveman99, 2017-02-20:
CC [M]  /root/build/tbsdriver/media_build/v4l/videobuf-dma-sg.o
/root/build/tbsdriver/media_build/v4l/videobuf-dma-sg.c: In function 'videobuf_vm_fault':
/root/build/tbsdriver/media_build/v4l/videobuf-dma-sg.c:442:21: error: 'struct vm_fault' has no member named 'virtual_address'
   (unsigned long)vmf->virtual_address,
                     ^
/root/build/tbsdriver/media_build/v4l/videobuf-dma-sg.c:56:42: note: in definition of macro 'dprintk'
   printk(KERN_DEBUG "vbuf-sg: " fmt , ## arg)
                                          ^
/root/build/tbsdriver/media_build/v4l/videobuf-dma-sg.c:448:46: error: 'struct vm_fault' has no member named 'virtual_address'
  clear_user_highpage(page, (unsigned long)vmf->virtual_address);
                                              ^
make[3]: *** [/root/build/tbsdriver/media_build/v4l/videobuf-dma-sg.o] Error 1
make[2]: *** [_module_/root/build/tbsdriver/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/kernels/4.10.0-1.el7.elrepo.x86_64'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/root/build/tbsdriver/media_build/v4l'
make: *** [all] Error 2

caveman99, 2017-02-22:
While this makes the tree compile again, i think a proper fix ist not commenting out the compile error, but to adapt this package to the 4.10 media tree. I'll follow up on this when i get a change to actually boot into the new kernel tonight.

CvH, 2017-02-23:
@caveman99 
```
--- a/drivers/media/v4l2-core/videobuf-dma-sg.c
+++ b/drivers/media/v4l2-core/videobuf-dma-sg.c
@@ -439,13 +439,12 @@ static int videobuf_vm_fault(struct vm_area_struct *vma, struct vm_fault *vmf)
 	struct page *page;
 
 	dprintk(3, "fault: fault @ %08lx [vma %08lx-%08lx]\n",
-		(unsigned long)vmf->virtual_address,
-		vma->vm_start, vma->vm_end);
+		vmf->address, vma->vm_start, vma->vm_end);
 
 	page = alloc_page(GFP_USER | __GFP_DMA32);
 	if (!page)
 		return VM_FAULT_OOM;
-	clear_user_highpage(page, (unsigned long)vmf->virtual_address);
+	clear_user_highpage(page, vmf->address);
 	vmf->page = page;
 
 	return 0;

```
taken from upstream media_tree, builds for me

crazycat69, 2017-03-10:
[linux_media](https://github.com/tbsdtv/linux_media) and [media_build](https://github.com/tbsdtv/media_build) updated.