tbsdtv/linux_media #35: Not compiling in Debian 9 / 4.9 Kernel

demontanaro, 2017-03-11:
Syncing with dir ../../media
make[2]: Leaving directory '/root/media_build/linux'
make -C /lib/modules/4.9.0-2-amd64/build SUBDIRS=/root/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-4.9.0-2-amd64'
  CC [M]  /root/media_build/v4l/cec-adap.o
/root/media_build/v4l/cec-adap.c: In function 'cec_transmit_msg_fh':
/root/media_build/v4l/cec-adap.c:600:17: error: 'CEC_MSG_FL_REPLY_TO_FOLLOWERS' undeclared (first use in this function)
   msg->flags &= CEC_MSG_FL_REPLY_TO_FOLLOWERS;
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/media_build/v4l/cec-adap.c:600:17: note: each undeclared identifier is reported only once for each function it appears in
/root/media_build/v4l/cec-adap.c: In function 'cec_config_thread_func':
/root/media_build/v4l/cec-adap.c:1273:21: error: 'CEC_LOG_ADDRS_FL_CDC_ONLY' undeclared (first use in this function)
       (las->flags & CEC_LOG_ADDRS_FL_CDC_ONLY))
                     ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/media_build/v4l/cec-adap.c: In function '__cec_s_log_addrs':
/root/media_build/v4l/cec-adap.c:1401:25: error: 'CEC_LOG_ADDRS_FL_CDC_ONLY' undeclared (first use in this function)
  if (log_addrs->flags & CEC_LOG_ADDRS_FL_CDC_ONLY) {
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/media_build/v4l/cec-adap.c: In function 'cec_receive_notify':
/root/media_build/v4l/cec-adap.c:1626:6: error: implicit declaration of function 'cec_is_cdc_only' [-Werror=implicit-function-declaration]
  if (cec_is_cdc_only(&adap->log_addrs) &&
      ^~~~~~~~~~~~~~~
/root/media_build/v4l/cec-adap.c:1699:33: error: 'CEC_LOG_ADDRS_FL_ALLOW_RC_PASSTHRU' undeclared (first use in this function)
       !(adap->log_addrs.flags & CEC_LOG_ADDRS_FL_ALLOW_RC_PASSTHRU))
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/media_build/v4l/cec-adap.c:1797:33: error: 'CEC_MSG_FL_REPLY_TO_FOLLOWERS' undeclared (first use in this function)
  if (is_reply && !(msg->flags & CEC_MSG_FL_REPLY_TO_FOLLOWERS))
                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.9.0-2-common/scripts/Makefile.build:298: recipe for target '/root/media_build/v4l/cec-adap.o' failed
make[5]: *** [/root/media_build/v4l/cec-adap.o] Error 1
/usr/src/linux-headers-4.9.0-2-common/Makefile:1507: recipe for target '_module_/root/media_build/v4l' failed
make[4]: *** [_module_/root/media_build/v4l] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[3]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.9.0-2-amd64'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/root/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

xiao191cao, 2017-03-13:
1. 
line 426
		//if (dvb->count == 1)
			/*return cx231xx_init_bulk_ts2(dev, CX231XX_DVB_MAX_PACKETS,
					CX231XX_DVB_NUM_BUFS,
					dev->ts2_mode.max_pkt_size,
					dvb_bulk_copy_ts2);
			*/		
		//else
			return cx231xx_init_bulk(dev, CX231XX_DVB_MAX_PACKETS,
					CX231XX_DVB_NUM_BUFS,
					dev->ts1_mode.max_pkt_size,
					dvb_bulk_copy);

2. 
line 451
		/*if (dvb->count == 1)
			cx231xx_uninit_bulk_ts2(dev);
		else*/

3.
line 828
demod_i2c = cx231xx_get_i2c_adap(dev, dev->board.demod_i2c_master[i]);
--->demod_i2c = cx231xx_get_i2c_adap(dev, dev->board.demod_i2c_master);

4.
line 1362

                tbs_cx_mac(cx231xx_get_i2c_adap(dev, dev->board.demod_i2c_master[0]), i, mac);
		dev_info(dev->dev, "MAC address %pM\n", mac);
		memcpy(dev->dvb[i]->adapter.proposed_mac, mac, 6);
---------------->
		/*tbs_cx_mac(cx231xx_get_i2c_adap(dev, dev->board.demod_i2c_master[0]), i, mac);
		dev_info(dev->dev, "MAC address %pM\n", mac);
		memcpy(dev->dvb[i]->adapter.proposed_mac, mac, 6);
                */
                if (i == 0) {
			tbs_cx_mac(&dev->i2c_bus[1].i2c_adap, 0, mac);
		} 
                if (i == 1) {
			memcpy(dev->dvb[0]->adapter.proposed_mac, mac, 6);
			printk(KERN_INFO "TurboSight TBS5990 MAC Addresse bas: %pM\n", mac);
			mac[5] +=1;
			memcpy(dev->dvb[1]->adapter.proposed_mac, mac, 6);
		 }

dem0nlord, 2017-03-13:
This module does now build for me after the latest commits, however my WinTV-HVR-955Q now becomes non functional, not seen by tvheadend. Go back to the previous 4.9.0-rc6 and it still works. lsusb -v is identical in both. I'm guessing possibly the i2c changes to lgdt3306a.

crazycat69, 2017-03-13:
See more details in dmesg output.17:11, 13 марта 2017 г., dem0nlord <notifications@github.com>:This module does now build for me after the latest commits, however my WinTV-HVR-955Q now becomes non functional, not seen by tvheadend. Go back to the previous 4.9.0-rc6 and it still works. lsusb -v is identical in both. I'm guessing possibly the i2c changes to lgdt3306a.

dem0nlord, 2017-03-14:
All my adapters are working again with your newest commits, hopefully the issue creator will be fixed also.

demontanaro, 2017-03-14:
The drivers now compile without problems, and the cards working fine.

For me, the bug is solved now.