tbsdtv/linux_media #37: 5281 usb dual tuner stopped working

feroxy, 2017-03-12:
I am using a tbs5281 dual usb tuner alongside two tbs5220 usb sticks in a dvr system. This setup was working fine, with 4 adapters showing up as expected under /dev/dvb.

Updated to latest git and now only the two 5220s show up under /dev/dvb and in tvheadend (and work fine). However the 5281 now shows up under /dev/v4l and is not detected in tvheadend.

crazycat69, 2017-03-13:
what kernel version ?

dem0nlord, 2017-03-13:
He said latest git. I'd expect this is part of the current cx231xx build failures mentioned in that other issue, which I am seeing as well.

crazycat69, 2017-03-13:
Please check again with latest changes
43e0fdfa3324345ffd42d197a464fb67fe106c19

feroxy, 2017-03-13:
No fix with latest changes @crazycat69 and I have uploaded dmesg log at http://termbin.com/ojd1 (call traces at the end seem related)

Also may have spoken too soon re: 5220. One of them is glitching after system has been running for a while and will no longer tune. The other one still works at that point though:

```
Mar 13 19:20:02 nucnasdvr tvheadend[1237]: mpegts: 498MHz in Freeview - scan no data, failed
Mar 13 19:20:03 nucnasdvr tvheadend[1237]: subscription: 00D4: service instance is bad, reason: No input detected
Mar 13 19:20:07 nucnasdvr tvheadend[1237]: subscription: 00D4: No input source available for subscription "HTTP" to channel "BBC ONE NI HD"
```

crazycat69, 2017-03-13:
Please check again with latest changes
7b1ae2111e28f4639ec4d83310f9e753535bec72

P.S. Maybe you can give me remote access for inspect/debug driver ?

feroxy, 2017-03-13:
That seems to have fixed it; all 4 adapters showing up and working as expected. I'll keep an eye on it and check it stays that way, but looks good so far.

Since it seems to be working now, I'll assume remote access is not needed? But if need be I could work something out (complicated a little in that this box is doing multiple duty as a nas etc; ie not something I would typically be allowing outside my lan)

crazycat69, 2017-03-14:
Ok, not need access.

closed 2017-03-15 by crazycat69