tbsdtv/linux_media #40: TBS5182 suspend problem

schneci, 2017-03-21:
I'm using the TBS8152 dvb-c USB module with Tvheadend 4.2 in Libreelec 8.0.0 environment. This Libreelec version uses the Linux 4.9.8 version.

I would like to sleep the system is used by Kodi Krypton suspend power-off option, but the system is freezed. If I have disconnected the USB modul from system the suspend process has worked correctly.
In next step I have rebooted and reconnected the USB hardware to Libreelec and I have tested the `cx231xx` module dependency tree by `lsmod` command:
```
lsmod | grep cx
cx231xx_dvb_ci         14873  0
tas2101                 8235  1 cx231xx_dvb_ci
dvb_core               82018  1 cx231xx_dvb_ci
cx25840               138349  1
cx231xx               147335  1 cx231xx_dvb_ci
videobuf_vmalloc        4231  1 cx231xx
videobuf_core          14600  2 cx231xx,videobuf_vmalloc
cx2341x                15535  1 cx231xx
i2c_mux                 4330  2 si2168,cx231xx
v4l2_common             3125  3 cx2341x,cx231xx,cx25840
tveeprom               13169  1 cx231xx
videodev              113087  4 cx2341x,v4l2_common,cx231xx,cx25840
rc_core                17350  4 `imon,rc_imon_pad,cx231xx
```
If I try to remove `cx231xx` dvb driver by using `rmmod` before suspend, it has sent the following message:
```
rmmod cx231xx
rmmod: ERROR: Module cx231xx is in use by: cx231xx_dvb_ci
```
After this I try to remove the `cx231xx_dvb_ci` driver, but `rmmod` sent a `Killed` message and after this doesn't removed the module:
```
rmmod cx231xx_dvb_ci
Killed
```
I think it doesn't normal message, may be it is a Linux kernel modul problem.

Can somebody help me?

PS: I'm sending some additional information:
```
modinfo cx231xx
filename:       /lib/modules/4.9.8/updates/cx231xx.ko
version:        0.0.3
license:        GPL
description:    Conexant cx231xx based USB video device driver
author:         Srinivasa Deevi <srinivasa.deevi@conexant.com>
firmware:       v4l-cx23885-enc.fw
srcversion:     69AE4B51D811435B01A920C
...
depends:        videobuf-core,videodev,v4l2-common,tveeprom,cx2341x,videobuf-vmalloc,rc-core
vermagic:       4.9.8 SMP mod_unload
parm:           pcb_debug:enable pcb config debug messages [video] (int)
parm:           mpegbufs:number of mpeg buffers, range 2-32 (int)
parm:           mpeglines:number of lines in an MPEG buffer, range 2-32 (int)
parm:           mpeglinesize:number of bytes in each line of an MPEG buffer, range 512-1024 (int)
parm:           v4l_debug:enable V4L debug messages (int)
parm:           core_debug:enable debug messages [core] (int)
parm:           reg_debug:enable debug messages [URB reg] (int)
parm:           alt:alternate setting to use for video endpoint (int)
parm:           tuner:tuner type (int)
parm:           transfer_mode:transfer mode (1-ISO or 0-BULK) (int)
parm:           disable_ir:disable infrared remote support (int)
parm:           i2c_scan:scan i2c bus at insmod time (int)
parm:           i2c_debug:enable debug messages [i2c] (int)
parm:           isoc_debug:enable debug messages [isoc transfers] (int)
parm:           card:card type (array of int)
parm:           video_nr:video device numbers (array of int)
parm:           vbi_nr:vbi device numbers (array of int)
parm:           radio_nr:radio device numbers (array of int)
parm:           video_debug:enable debug messages [video] (int)
```
```
modinfo cx231xx_dvb_ci
filename:       /lib/modules/4.9.8/updates/cx231xx-dvb-ci.ko
license:        GPL
author:         Srinivasa Deevi <srinivasa.deevi@conexant.com>
description:    driver for cx231xx based DVB cards
depends:        cx231xx,dvb-core,tas2101
vermagic:       4.9.8 SMP mod_unload
parm:           debug:enable debug messages [dvb] (int)
parm:           adapter_nr:DVB adapter numbers (array of short)
```
```
modinfo si2168
filename:       /lib/modules/4.9.8/updates/si2168.ko
firmware:       dvb-demod-si2168-b40-01.fw
firmware:       dvb-demod-si2168-a30-01.fw
firmware:       dvb-demod-si2168-a20-01.fw
license:        GPL
description:    Silicon Labs Si2168 DVB-T/T2/C demodulator driver
author:         Antti Palosaari <crope@iki.fi>
alias:          i2c:si2168
depends:
vermagic:       4.9.8 SMP mod_unload
```
```
modinfo si2157
filename:       /lib/modules/4.9.8/updates/si2157.ko
firmware:       dvb-tuner-si2141-a10-01.fw
firmware:       dvb-tuner-si2158-a20-01.fw
license:        GPL
author:         Antti Palosaari <crope@iki.fi>
description:    Silicon Labs Si2146/2147/2148/2157/2158 silicon tuner driver
alias:          i2c:si2141
alias:          i2c:si2146
alias:          i2c:si2157
depends:
vermagic:       4.9.8 SMP mod_unload
```

crazycat69, 2017-03-21:
First you must stop tvheadend.

schneci, 2017-03-21:
I tried anyway, but it did not help.
```
systemctl stop service.tvheadend42

systemctl is-active service.tvheadend42
inactive
```
After this randomly sometimes `rmmod cx231xx_dvb_ci` command answers `rmmod: ERROR: Module cx231xx_dvb_ci is in use`
or
`Killed.`

BTW I will use the `89-tvheadend.power` script file in `/storage/.config.sleep.d` for stopping and restarting the dvb modules, but until doesn't solve above problem I couldn't use it. The `REMOVE_MODULE` variable can be set in Tvheaden42 addon configuration panel which will be included the removing module name sequance.

```
. /etc/profile

oe_setup_addon service.tvheadend42

SERVICE="service.tvheadend42"

case "$1" in
  pre)
    if systemctl is-active "$SERVICE" &>/dev/null ; then
      systemctl stop "$SERVICE"
      for module in $REMOVE_MODULES ; do
        rmmod $module
      done
    fi
    ;;
  post)
    if systemctl is-enabled "$SERVICE" &>/dev/null ; then
      for module in $REMOVE_MODULES ; do
        modprobe $module
      done
      systemctl start "$SERVICE"
    fi
    ;;
esac
```

crazycat69, 2017-03-21:
So no luck :-) Many current dvb drivers prevent unloading and not handle suspend/wakeup.

schneci, 2017-03-29:
I think the following messages in dmesg are caused the rmmod Killed or Segfault output:

```
[ 50.184375] BUG: unable to handle kernel NULL pointer dereference at 0000000000000010
[ 50.184546] IP: [<ffffffffa09f7306>] dvb_fini+0xf6/0x150 [cx231xx_dvb_ci]
```

You can see in end of dmesg what happened after rmmod cx231xx_dvb_ci:
http://sprunge.us/FNJj

schneci, 2017-03-30:
I think I have found the main problem. The TBS 5281 usb card includes dual adapter. The TBS proprietary driver use the adap_cnt counter variable:
 
```
    [CX231XX_BOARD_TBS_5281] = {
        .name = "TurboSight TBS 5281",
        .tuner_type = TUNER_ABSENT,
        .decoder = CX231XX_AVDECODER,
        .output_mode = OUT_MODE_VIP11,
        .demod_xfer_mode = 0,
        .ctl_pin_status_mask = 0xFFFFFFC4,
        .agc_analog_digital_select_gpio = 0x00,
        .tuner_sif_gpio = -1,
        .tuner_scl_gpio = -1,
        .tuner_sda_gpio = -1,
        .gpio_pin_status_mask = 0x4001000,
        .tuner_i2c_master = 2,
        .demod_i2c_master = 1,
        .has_dvb = 1,
        .adap_cnt = 2,
        .demod_addr = 0x64,
        .norm = V4L2_STD_PAL_M,
        
        .input = {{
            .type = CX231XX_VMUX_TELEVISION,
            .vmux = CX231XX_VIN_3_1,
            .amux = CX231XX_AMUX_VIDEO,
            .gpio = NULL,
        }, {
            .type = CX231XX_VMUX_COMPOSITE1,
            .vmux = CX231XX_VIN_2_1,
            .amux = CX231XX_AMUX_LINE_IN,
            .gpio = NULL,
        }, {
            .type = CX231XX_VMUX_SVIDEO,
            .vmux = CX231XX_VIN_1_1 |
                (CX231XX_VIN_1_2 << 8) |
                CX25840_SVIDEO_ON,
            .amux = CX231XX_AMUX_LINE_IN,
            .gpio = NULL,
        } },
    },
```

The kernel BUG message in dmesg shows the problem is caused by dvb_fini process with kernel NULL point exception during rmmod. I have seen the TBS proprietary driver source and this calculates the adapter number in adapter release and unregister calling, but the opensource cx231xx driver not:

```
static int dvb_fini(struct cx231xx *dev)
{
    int i;

    if (!dev->board.has_dvb) {
        /* This device does not support the extension */
        return 0;
    }

    for (i = 0; i < dev->board.adap_cnt; i++) {
    if (dev->dvb[i]) {
        tbscxci_release(dev->dvb[i]);
        unregister_dvb(dev->dvb[i]);
        dev->dvb[i] = NULL;
    }
    }

    return 0;
}
```

What is your opinion? Can anybody help to solve this problem?

crazycat69, 2017-03-30:
this code present in dvb_fini().
https://github.com/tbsdtv/linux_media/blob/latest/drivers/media/usb/cx231xx/cx231xx-dvb.c#L1415

crazycat69, 2017-03-30:
Check this fix
f10b80837b205be12e3b511a14c184f1873ce1d6

schneci, 2017-03-31:
Hi crazycat69,

How can I test this version in Libreelec8.0.0 environment, because LE8 use the standard linux 4.9.8 distribution witch includes the cx231xx driver. That version generally can handle TBS5281, but the `rmmod` is chrased this module when I want to put the system to suspend.

CvH, 2017-03-31:
A test version (with this commit) is already at the designated forum thread at LE.

CvH, 2017-04-01:
@crazycat69
https://github.com/tbsdtv/linux_media/commit/f10b80837b205be12e3b511a14c184f1873ce1d6 was reported working from @schneci at the LE forum thread

crazycat69, 2017-04-01:
ok :)

closed 2017-04-01 by crazycat69