tbsdtv/linux_media #41: TBS-6290 SE DVB CAM not recognized

stole204, 2017-03-24:
Hello all,

Yesterday I just compile those nice TBS open source drivers the card works ok, but my SMIT module is not recognized.

dmesg log says:

dvb_ca_en50221: dvb_ca adapter 0: DVB CAM link initialisation failed :(

uname -r
4.4.56-1.el7.elrepo.x86_64
CentOS Linux release 7.3.1611 (Core)

steven1758, 2017-03-26:
Hi 

Please have a try like this change this file 

linux/drivers/media/dvb-core/dvb_ca_en50221.c at line 354

old line:     if ((ret = dvb_ca_en50221_wait_if_status(ca, slot, STATUSREG_DA, HZ / 10)) != 0)

new line:   if ((ret = dvb_ca_en50221_wait_if_status(ca, slot, STATUSREG_DA, HZ)) != 0)

then recompile the driver and install check whether will be ok 

Thanks 

kind Regards

stole204, 2017-03-26:
I already did that, not working.

steven1758, 2017-03-27:
Hi can you tell us that which version of closed TBS drivers you are used to test which works ?

stole204, 2017-03-27:
These are drivers from TBS **tbs-linux-drivers_v170206.zip**

dvb_ca adapter 0: DVB CAM detected and initialised successfully

Davin622, 2017-03-28:
Is it CI+ of your cards?Do you have other CAM?

steven1758, 2017-03-28:
Hi 

Please send email to our support@tbsdtv.com 

Thanks 

Kind Regards