tbsdtv/linux_media #42: TBS5990 Single interface working only

cwsutherland, 2017-03-30:
Using Linux raspberrypi 4.4.50-v7+ Kernel

Build works, both adapters and frontends appear.

root@raspberrypi:/etc/modprobe.d# ls /dev/dvb
adapter0  adapter1

However only single adaptor works. in dmesg continuous

[129049.167218] cx231xx 1-1.4:1.1: Unknown tuner type configuring SIF

pi@raspberrypi:~ $ lsusb
Bus 001 Device 004: ID 734c:5990 TBS Technologies China 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
cfg80211              428871  0 
rfkill                 16037  1 cfg80211
av201x                  3701  2 
cx231xx_dvb_ci         14788  0 
tas2101                 8975  3 cx231xx_dvb_ci
dvb_core               85726  1 cx231xx_dvb_ci
cx231xx_alsa            5821  0 
snd_pcm                75762  1 cx231xx_alsa
snd_timer              19288  1 snd_pcm
snd                    51908  3 cx231xx_alsa,snd_timer,snd_pcm
cx25840                48623  1 
cx231xx               148837  2 cx231xx_alsa,cx231xx_dvb_ci
cx2341x                17304  1 cx231xx
videobuf_vmalloc        4893  1 cx231xx
videobuf_core          16667  2 cx231xx,videobuf_vmalloc
tveeprom               10846  1 cx231xx
rc_core                17983  1 cx231xx
i2c_mux                 2457  2 cx231xx,tas2101
v4l2_common             3929  3 cx2341x,cx231xx,cx25840
videodev              140825  4 cx2341x,cx231xx,cx25840,v4l2_common
media                  19205  4 cx231xx,cx25840,videodev,dvb_core
bcm2835_gpiomem         2976  0 
bcm2835_wdt             3225  0 
uio_pdrv_genirq         3164  0 
uio                     8000  1 uio_pdrv_genirq
ipv6                  347620  30 

/etc/modprobe.d/cx231xx.conf
options cx231xx transfer_mode=0
options disable infrared remote support

sharky1337, 2017-06-25:
How did you get that far ? I'm not able to get the module working. System runs raspbian on a rpi3 .

Which driver you are using ? please have a try with this one :
#wget http://www.tbsdtv.com/download/document/linux/media_build-0119.tar.bz2
# sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/
#tar jxvf media_build-0119.tar.bz2
#cd media_build 
#./install.sh 
#reboot

cwsutherland, 2017-06-26:
I'm guessing you are missing some dependencies

check you have raspberrypi-kernel-headers are installed