tbsdtv/linux_media #43: Debian 9 /kernel 4.9.13-1 not compile -again-

demontanaro, 2017-03-30:
/root/tbs_github/media_build/v4l/pt3.c: In function 'pt3_fetch_thread':
/root/tbs_github/media_build/v4l/pt3.c:466:9: error: incompatible types when assigning to type 'ktime_t {aka union ktime}' from type 'long int'
   delay = PT3_FETCH_DELAY * NSEC_PER_MSEC;
         ^
/usr/src/linux-headers-4.9.0-2-common/scripts/Makefile.build:298: recipe for target '/root/tbs_github/media_build/v4l/pt3.o' failed
make[5]: *** [/root/tbs_github/media_build/v4l/pt3.o] Error 1
make[5]: *** Waiting for unfinished jobs....
/usr/src/linux-headers-4.9.0-2-common/Makefile:1507: recipe for target '_module_/root/tbs_github/media_build/v4l' failed
make[4]: *** [_module_/root/tbs_github/media_build/v4l] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[3]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.9.0-2-amd64'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/root/tbs_github/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

steven1758, 2017-03-31:
Hi please try the latest code should fixed this issue

demontanaro, 2017-03-31:
I run git remote update and git pull and no new modifications.

I updated the source before build it (about a 6 or 7 hours ago)