tbsdtv/linux_media #44: TBS 6909 Kernel 4.4.0-71 - TBSECP3 frontend 0 attach failed.

nightshadow36, 2017-04-03:
Hi.
Just updated my kernel and updated my card drivers now i get TBSECP3 frontend  attach failed.
I don`t know what`s going on it says registering and failed on the same numbers but not showing 32 tuners?
What to do thank you.

[   36.502707] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   36.579398] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   36.700700] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   36.796576] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   36.916593] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 4 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.028659] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 5 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.109126] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 6 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   37.230788] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.453879] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 8 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.552673] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 9 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.625264] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 10 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.695869] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 11 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.799870] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 12 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.919852] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 13 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   62.988065] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 14 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   63.084404] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 15 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[   63.209636] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 0 attach failed
[   63.209637] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   63.365895] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 1 attach failed
[   63.365896] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   63.481883] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 2 attach failed
[   63.481884] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   63.593715] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 3 attach failed
[   63.593716] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   63.753893] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 4 attach failed
[   63.753894] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   63.905860] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 5 attach failed
[   63.905861] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   64.025849] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 6 attach failed
[   64.025850] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   64.185843] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 7 attach failed
[   64.185844] TBSECP3 driver 0000:09:00.0: frontend attach failed
[   64.297841] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 0 attach failed
[   64.297842] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   64.453818] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 1 attach failed
[   64.453819] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   64.561545] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 2 attach failed
[   64.561546] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   64.685385] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 3 attach failed
[   64.685387] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   64.841859] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 4 attach failed
[   64.841860] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   64.961833] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 5 attach failed
[   64.961834] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   65.073801] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 6 attach failed
[   65.073802] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[   65.185676] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 7 attach failed
[   65.185677] TBSECP3 driver 0000:0a:00.0: frontend attach failed

ovidiu31, 2017-04-03:
@nightshadow36 

delete the media and media_build folders and re-git them then follow the installation and compile proces
https://github.com/tbsdtv/linux_media/wiki

nightshadow36, 2017-04-03:
I have done this already about 4 times. I think i have to change DVB_MAX_ADAPTERS but trying to find out where

nightshadow36, 2017-04-03:
Please anyone know the problem. I have just dont a complete fresh install on my IPTV server and i have the same problem. I thought it might have been max DVD adapters but that is set to 48.

CvH, 2017-04-03:
https://github.com/tbsdtv/media_build/blob/master/v4l/scripts/make_kconfig.pl#L618
It is set to 48, we (at LE) have set it for a while to 64 now (linux kernel default is 8) with no visible side effects.

nightshadow36, 2017-04-03:
Yes the settings of 48 is fine, this was how it was set before. But with the kernel update and the latest driver i have the errors above. Problem is my IPTV server has been off line 14 hours, very tired and dont know how to fix this problem

ovidiu31, 2017-04-03:
first of all my advice would be to disable the kernel updating ... had same problems as you ... then disabled kernel updates recompiled and reinstall drivers all is ok now.

to avoid for future such stuff disable on production servers the kernel updates... some people might say security breach and stuff ... just make sure you have strong passwords and do the kernel update disable.

nightshadow36, 2017-04-03:
Its been a nightmare, been about 16 hours - In the end I gave up. Done the same as you.
Thanks

momomo, 2017-07-25:
@nightshadow36 @ovidiu31 

I am having similar issues I believe: 

I posted a question on tvheadend forum prior to finding this issue: https://tvheadend.org/boards/5/topics/27406

I am not so well oriented in this field. 

Care to explain what is the best fix for this?