tbsdtv/linux_media #47: TBS5520 no frontend was attached error

Tnix76, 2017-04-05:
Hi all,
I compiled the open source drivers successfully but when I reboot or replug the USB I get the next errors (excerpt):

```
[    4.441206] dvb-usb: found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware
[    4.443600] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[    4.443638] iTCO_wdt: Found a Lynx Point TCO device (Version=2, TCOBASE=0x1860)
[    4.443939] dvb-usb: downloading firmware from file 'dvb-usb-tbs5520-01.fw'
[    4.443940] tbs5520: start downloading tbs5520 firmware
....
[    4.567295] tbs5520: end downloading tbs5520 firmware
[    4.567300] dvb-usb: found a 'TBS 5520 USB2.0' in warm state.
[    4.567360] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    4.567370] dvbdev: DVB: registering new adapter (TBS 5520 USB2.0)
[    4.567371] dvb-usb: MAC address: 00:00:00:00:00:00
[    4.570003] i2c i2c-9: avl6882: attach failed family id mismatch
[    4.570031] TBS5520: frontend attach failed
[    4.570032] dvb-usb: no frontend was attached by 'TBS 5520 USB2.0'
[    4.570101] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-2/input/input16
[    4.570129] dvb-usb: schedule remote query interval to 150 msecs.
[    4.570130] dvb-usb: TBS 5520 USB2.0 successfully initialized and connected.
[    4.570151] usbcore: registered new interface driver tbs5520
[    4.587549] dvb-usb: TBS 5520 USB2.0 successfully deinitialized and disconnected.
[    4.679804] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[    4.683377] iwlwifi 0000:03:00.0 wlp3s0: renamed from wlan0
[    4.898198] Adding 16777212k swap on /dev/sdb2.  Priority:-1 extents:1 across:16777212k FS
[    5.596719] EXT4-fs (sdb3): mounted filesystem with ordered data mode. Opts: (null)
[    5.643190] systemd-journald[552]: Received request to flush runtime journal from PID 1
[    5.919327] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
[    6.105886] EXT4-fs (sdb4): mounted filesystem with ordered data mode. Opts: (null)
[    6.241300] usb 3-2: new high-speed USB device number 8 using xhci_hcd
[    6.409808] usb 3-2: New USB device found, idVendor=734c, idProduct=5520
[    6.409810] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.409811] usb 3-2: Product: TBS 5520
[    6.409812] usb 3-2: Manufacturer: TBS-Tech
[    6.410259] dvb-usb: found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware
[    6.410283] dvb-usb: downloading firmware from file 'dvb-usb-tbs5520-01.fw'
[    6.410284] tbs5520: start downloading tbs5520 firmware
[    6.519233] tbs5520: end downloading tbs5520 firmware
[    6.519236] dvb-usb: found a 'TBS 5520 USB2.0' in warm state.
[    6.519275] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    6.519329] dvbdev: DVB: registering new adapter (TBS 5520 USB2.0)
[    6.519331] dvb-usb: MAC address: 00:00:00:00:00:00
[    6.522579] i2c i2c-9: avl6882: found id=0xb family_id=0x68624955
[    6.528843] i2c i2c-9: r848: Rafael R848 successfully attached
[    6.528844] tbs5520: TBS5520: frontend attached

[    6.528993] dvb-usb: no frontend was attached by 'TBS 5520 USB2.0'
[    6.529147] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-2/input/input17
[    6.529177] dvb-usb: schedule remote query interval to 150 msecs.
[    6.529178] dvb-usb: TBS 5520 USB2.0 successfully initialized and connected.
```

With all any DVB device is present in the system.

Some data about my system:
Fedora 25 Workstation
4.10.8-200.fc25.x86_64

Thanks.

Davin622, 2017-04-06:
Hi,
i have check your log file, "frontend attached" successfully. but then show"dvb-usb: no frontend was attached by 'TBS 5520 USB2.0" i don not know why. would you mind wite a email to our support(support@tbsdtv.com) then give us a SSH?

Tnix76, 2017-04-06:
Hi,
I'm going to write an email to support@tbsdtv.com with the same case I've wrote before but I can't understand why you need a SSH ... I'll send you the logs and information you'll need.

Thanks.

Tnix76, 2017-04-25:
Hi,
after the last kernel updates, the problem still persists. Now with kernel 4.10.11-200.fc25.x86_64. I paste the system error log when I re-connect the USB device:

`abr 25 23:18:30 mymachine kernel: usb 3-2: new high-speed USB device number 8 using xhci_hcd 
abr 25 23:18:30 mymachine kernel: usb 3-2: New USB device found, idVendor=734c, idProduct=5520 
abr 25 23:18:30 mymachine kernel: usb 3-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0 
abr 25 23:18:30 mymachine mtp-probe[27729]: checking bus 3, device 8: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-2" 
abr 25 23:18:30 mymachine mtp-probe[27729]: bus: 3, device: 8 was not an MTP device 
abr 25 23:18:30 mymachine kernel: dvb_usb_tbs5520: loading out-of-tree module taints kernel. 
abr 25 23:18:30 mymachine kernel: dvb_usb_tbs5520: module verification failed: signature and/or required key missing - tainting kernel 
abr 25 23:18:30 mymachine kernel: dvb-usb: found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware 
abr 25 23:18:30 mymachine kernel: dvb-usb: downloading firmware from file 'dvb-usb-tbs5520-01.fw' 
abr 25 23:18:30 mymachine kernel: tbs5520: start downloading tbs5520 firmware 
abr 25 23:18:30 mymachine kernel: usb 3-2: USB disconnect, device number 8 
abr 25 23:18:30 mymachine kernel: tbs5520: end downloading tbs5520 firmware 
abr 25 23:18:30 mymachine kernel: dvb-usb: found a 'TBS 5520 USB2.0' in warm state. 
abr 25 23:18:30 mymachine kernel: dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer. 
abr 25 23:18:30 mymachine kernel: dvbdev: DVB: registering new adapter (TBS 5520 USB2.0) 
abr 25 23:18:30 mymachine kernel: dvb-usb: MAC address: 00:00:00:00:00:00 
abr 25 23:18:30 mymachine baloo_file[2856]: QObject::connect: invalid null parameter 
abr 25 23:18:30 mymachine baloo_file[2856]: QObject::connect: invalid null parameter 
abr 25 23:18:30 mymachine kernel: i2c i2c-9: avl6882: attach failed family id mismatch 
abr 25 23:18:30 mymachine kernel: TBS5520: frontend attach failed 
abr 25 23:18:30 mymachine kernel: dvb-usb: no frontend was attached by 'TBS 5520 USB2.0' 
abr 25 23:18:30 mymachine kernel: input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-2/input/input16 
abr 25 23:18:30 mymachine kernel: dvb-usb: schedule remote query interval to 150 msecs. 
abr 25 23:18:30 mymachine kernel: dvb-usb: TBS 5520 USB2.0 successfully initialized and connected. 
abr 25 23:18:30 mymachine kernel: usbcore: registered new interface driver tbs5520 
abr 25 23:18:30 mymachine kdeinit5[2739]: QObject::connect: invalid null parameter 
abr 25 23:18:30 mymachine kdeinit5[2739]: QObject::connect: invalid null parameter 
abr 25 23:18:30 mymachine kdeinit5[2739]: QObject::connect: invalid null parameter 
abr 25 23:18:30 mymachine baloo_file[2856]: QObject::connect: invalid null parameter 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (II) config/udev: Adding input device IR-receiver inside an USB DVB receiver (/dev/input/event15) 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (**) IR-receiver inside an USB DVB receiver: Applying InputClass "evdev keyboard catchall" 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (**) IR-receiver inside an USB DVB receiver: Applying InputClass "libinput keyboard catchall" 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (**) IR-receiver inside an USB DVB receiver: Applying InputClass "system-keyboard" 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (II) Using input driver 'libinput' for 'IR-receiver inside an USB DVB receiver' 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (EE) systemd-logind: failed to take device /dev/input/event15: No such device 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (**) IR-receiver inside an USB DVB receiver: always reports core events 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (**) Option "Device" "/dev/input/event15" 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (**) Option "_source" "server/udev" 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (EE) client bug: Invalid path /dev/input/event15 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (EE) libinput: IR-receiver inside an USB DVB receiver: Failed to create a device for /dev/input/event15 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (EE) PreInit returned 2 for "IR-receiver inside an USB DVB receiver" 
abr 25 23:18:30 mymachine /usr/libexec/gdm-x-session[2548]: (II) UnloadModule: "libinput" 
abr 25 23:18:30 mymachine kernel: dvb-usb: TBS 5520 USB2.0 successfully deinitialized and disconnected. 
abr 25 23:18:32 mymachine kernel: usb 3-2: new high-speed USB device number 9 using xhci_hcd 
abr 25 23:18:32 mymachine kernel: usb 3-2: New USB device found, idVendor=734c, idProduct=5520 
abr 25 23:18:32 mymachine kernel: usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0 
abr 25 23:18:32 mymachine kernel: usb 3-2: Product: TBS 5520 
abr 25 23:18:32 mymachine kernel: usb 3-2: Manufacturer: TBS-Tech 
abr 25 23:18:32 mymachine kernel: dvb-usb: found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware 
abr 25 23:18:32 mymachine kernel: dvb-usb: downloading firmware from file 'dvb-usb-tbs5520-01.fw' 
abr 25 23:18:32 mymachine kernel: tbs5520: start downloading tbs5520 firmware 
abr 25 23:18:32 mymachine kernel: tbs5520: end downloading tbs5520 firmware 
[... cut ...]
abr 25 23:18:32 mymachine /usr/libexec/gdm-x-session[2548]: (II) XINPUT: Adding extended input device "IR-receiver inside an USB DVB receiver" (type: KEYBOARD, id 18) 
abr 25 23:18:32 mymachine /usr/libexec/gdm-x-session[2548]: (**) Option "xkb_layout" "es" 
abr 25 23:18:32 mymachine /usr/libexec/gdm-x-session[2548]: (II) input device 'IR-receiver inside an USB DVB receiver', /dev/input/event15 is tagged by udev as: Keyboard 
abr 25 23:18:32 mymachine /usr/libexec/gdm-x-session[2548]: (II) input device 'IR-receiver inside an USB DVB receiver', /dev/input/event15 is a keyboard

`

Thanks.