tbsdtv/linux_media #48: TBS 6205 and 6209

locka-hd, 2017-04-22:
Hello 
I have TBS 6205 and 6209 insttaled with latest driver .. But DVB-C not work .. i can use only DVB-T ..
any idea

crazycat69, 2017-04-25:
what software used ?

locka-hd, 2017-04-26:
mkdir tbsdriver
cd tbsdriver
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make distclean
make -j4
make install
reboot

firmware:
wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/

Upgrading sources and re-installing：

~ $ cd media
~/media $ git remote update
~/media $ git pull
~/media $ cd ../media_build
~/media_build $ git remote update
~/media_build $ git pull
~/media_build $ make
~/media_build $ sudo make install
(now you can reboot or unload/reload modules manually:)
~/media_build $ reboot

crazycat69, 2017-04-27:
What soft used for tuning/straming ? Tvheadend, dvblast, vdr, vlc, astra and etc.

locka-hd, 2017-04-28:
i try astra and tvheadend .. in tvheadend i can serch Only DVB-T .. i think this drivers cant work For dvb-C

crazycat69, 2017-04-30:
What Dvb-c params ? Annex a/c or b ? Modulation 64qam or 256qam ?Sometimes dvb-c signal very strong and need some attenuator.

chrisfudge, 2017-06-14:
Hi
Newbie here. I am using opensuse Tumbleweed 4.11.4.1 with MythTv and I have a couple of problems: Mythtv reports the signal strength as 40% and sometime fails to record HD channels. When I have compiled TBS driver v170330 for Kernel 4.10.13 Mythtv reports signal strength at 90% and no failures on HD channels but with the Github driver on this kernel I get 40% and HD failures. 
I notice when compiling Github it warns of using "V4L/DVB backport tree, with experimental drivers
backported to run on legacy kernels" is there a way to use a current V4L/DVB tree for current 4.11 kernel.
Thanks
Chris Fudge

crazycat69, 2017-06-14:
What input signal ? Dvb-T/T2 or C ? Signal strengt and SNR ?

chrisfudge, 2017-06-15:
Hi
	Thanks for replying. I'm using a Terrestrial  aerial(U.K), I've attached two text 
files, created using dvb-fe-tool to show the differences. 
Tbs-tbs.txt is output when the driver from TBS is used.
Tbs-Github.txt is the output when the driver from 
Github tbsdtv/linux_media is used.

I believe this will provides Signal Strength and S/N ratio for each adaptor on the card.

Regards
Chris

On Wednesday, 14 June 2017 17:55:02 BST CrazyCat wrote:
> What input signal ? Dvb-T/T2 or C ? Signal strengt and SNR ?

Password: 
linux-steam:~ # cd /home/Myth/WorkArea
linux-steam:/home/Myth/WorkArea # ./tbs6205-test.sh
Linux linux-steam 4.10.13-1-default #1 SMP PREEMPT Thu Apr 27 12:23:31 UTC 2017 (e5d11ce) x86_64 x86_64 x86_64 GNU/Linux
Device TurboSight TBS 6205 DVBT/T2 frontend (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_4_5
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_RECOVER
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.3, Current v5 delivery system: DVBT
Supported delivery system: 
    [DVBT]
Warning: new delivery systems like ISDB-T, ISDB-S, DMB-TH, DSS, ATSC-MH will be miss-detected by a DVBv5.4 or earlier API call
FREQUENCY = 521964330
MODULATION = QAM/64
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = 3/4
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = 8K
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Device TurboSight TBS 6205 DVBT/T2 frontend (/dev/dvb/adapter1/frontend0) capabilities:
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_4_5
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_QAM_16
[... 252 lines cut ...]
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.10, Current v5 delivery system: DVBT
Supported delivery systems: 
    [DVBT]
     DVBT2
     DVBC/ANNEX_A
     DVBC/ANNEX_B
FREQUENCY = 521833330
MODULATION = QAM/AUTO
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = AUTO
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = AUTO
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.75dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0

chrisfudge, 2017-06-16:
Hi Thanks for replying. I'm using a Terrestrial aerial(U.K), I've attached two text files, created using dvb-fe-tool to show the differences. Tbs-tbs.txt is output when the driver from TBS is used. Tbs-Github.txt is the output when the driver from Github tbsdtv/linux_media is used. I believe this will provides Signal Strength and S/N ratio for each adaptor on the card. 
Regards Chris
Tbs-Tbs
`linux-steam:~ # cd /home/Myth/WorkArea
linux-steam:/home/Myth/WorkArea # ./tbs6205-test.sh
Linux linux-steam 4.10.13-1-default #1 SMP PREEMPT Thu Apr 27 12:23:31 UTC 2017 (e5d11ce) x86_64 x86_64 x86_64 GNU/Linux
Device TurboSight TBS 6205 DVBT/T2 frontend (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_4_5
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_RECOVER
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.3, Current v5 delivery system: DVBT
Supported delivery system: 
    [DVBT]
Warning: new delivery systems like ISDB-T, ISDB-S, DMB-TH, DSS, ATSC-MH will be miss-detected by a DVBv5.4 or earlier API call
FREQUENCY = 521964330
MODULATION = QAM/64
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = 3/4
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = 8K
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Device TurboSight TBS 6205 DVBT/T2 frontend (/dev/dvb/adapter1/frontend0) capabilities:
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_4_5
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_RECOVER
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.3, Current v5 delivery system: DVBT
Supported delivery system: 
    [DVBT]
Warning: new delivery systems like ISDB-T, ISDB-S, DMB-TH, DSS, ATSC-MH will be miss-detected by a DVBv5.4 or earlier API call
FREQUENCY = 521964330
[... 241 lines cut ...]
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.10, Current v5 delivery system: DVBT
Supported delivery systems: 
    [DVBT]
     DVBT2
     DVBC/ANNEX_A
     DVBC/ANNEX_B
FREQUENCY = 521833330
MODULATION = QAM/AUTO
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = AUTO
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = AUTO
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.75dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
`

chrisfudge, 2017-06-16:
Hi
 Thanks for replying. 
I'm using a Terrestrial aerial(U.K), I've attached two text files, created using dvb-fe-tool to show the differences. Tbs-tbs.txt is output when the driver from TBS is used. Tbs-Github.txt is the output when the driver from Github tbsdtv/linux_media is used. I believe this will provides Signal Strength and S/N ratio for each adaptor on the card.
Regards 
Chris
**tbs-Github**
Linux linux-steam 4.11.4-1-default #1 SMP PREEMPT Wed Jun 7 15:14:28 UTC 2017 (cba98ee) x86_64 x86_64 x86_64 GNU/Linux
Device TurboSight TBS 6205 DVB-T/T2/C  (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_MUTE_TS
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.10, Current v5 delivery system: DVBT
Supported delivery systems: 
    [DVBT]
     DVBT2
     DVBC/ANNEX_A
     DVBC/ANNEX_B
FREQUENCY = 521833330
MODULATION = QAM/AUTO
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = AUTO
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = AUTO
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT
Lock   (0x1f) Signal= -61.00dBm C/N= 32.25dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.75dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.50dB UCB= 152 postBER= 0
Lock   (0x1f) Signal= -61.00dBm C/N= 32.25dB UCB= 152 postBER= 0
-------------------------------------------------------------------------------------------------
tbs-tbs
**tbs-tbs**
linux-steam:/home/Myth/WorkArea # ./tbs6205-test.sh
Linux linux-steam 4.10.13-1-default #1 SMP PREEMPT Thu Apr 27 12:23:31 UTC 2017 (e5d11ce) x86_64 x86_64 x86_64 GNU/Linux
Device TurboSight TBS 6205 DVBT/T2 frontend (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_4_5
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_RECOVER
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.3, Current v5 delivery system: DVBT
Supported delivery system: 
    [DVBT]
Warning: new delivery systems like ISDB-T, ISDB-S, DMB-TH, DSS, ATSC-MH will be miss-detected by a DVBv5.4 or earlier API call
FREQUENCY = 521964330
MODULATION = QAM/64
BANDWIDTH_HZ = 8000000
INVERSION = OFF
CODE_RATE_HP = 3/4
CODE_RATE_LP = AUTO
GUARD_INTERVAL = 1/32
TRANSMISSION_MODE = 8K
HIERARCHY = NONE
DELIVERY_SYSTEM = DVBT
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0
Lock   (0x1f) Signal= 81.96% C/N= 99.95% UCB= 65500 postBER= 0