tbsdtv/linux_media #50: TBS6903 / Linux 4.10.12 Fedora 25 x86_64 kernel tried to execute NX-protected page - exploit attempt? (uid: 0)

lwandrebeck, 2017-04-28:
Hi,

Compiled drivers with latest git (even though patches application do fail).

Trying
```
femon -H
```

Gives:
1) femon is killed instantly by the system.
2) dmesg gives:

```
[10259.314438] kernel tried to execute NX-protected page - exploit attempt? (uid: 0)
[10259.314563] BUG: unable to handle kernel paging request at ffff96b4c6fc0068
[10259.314665] IP: 0xffff96b4c6fc0068
[10259.314715] PGD 3c437d067 
[10259.314717] PUD 80d045063 
[10259.314757] PMD 406e50063 
[10259.314796] PTE 8000000406fc0163

[10259.314906] Oops: 0011 [#1] SMP
[10259.314953] Modules linked in: ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_raw ip6table_security ip6table_mangle ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 iptable_raw iptable_security iptable_mangle iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ebtable_filter ebtables ip6table_filter ip6_tables stv6120(OE) stv0910(OE) amd64_edac_mod edac_mce_amd tbsecp3(OE) edac_core kvm_amd kvm tas2101(OE) irqbypass dvb_core sp5100_tco i2c_mux i2c_piix4 k10temp shpchp acpi_cpufreq tpm_tis tpm_tis_core tpm nfsd auth_rpcgss nfs_acl lockd grace sunrpc xfs libcrc32c ata_generic pata_acpi mgag200 drm_kms_helper ttm drm igb serio_raw ptp pata_atiixp pps_core dca 3w_9xxx i2c_algo_bit fjes
[10259.315920] CPU: 8 PID: 2543 Comm: femon Tainted: G           OE   4.10.12-200.fc25.x86_64 #1
[10259.316033] Hardware name: Supermicro H8DGU/H8DGU, BIOS 1.0a    07/01/2010
[10259.316126] task: ffff96b4c29c0000 task.stack: ffffb9f60705c000
[10259.316207] RIP: 0010:0xffff96b4c6fc0068
[10259.316263] RSP: 0018:ffffb9f60705fa60 EFLAGS: 00010202
[10259.316336] RAX: ffff96b4c6dee960 RBX: ffff96b4c6fc0050 RCX: 00000000ffff00f2
[10259.316430] RDX: 0000000000000002 RSI: 0000000000000002 RDI: ffff96b4c6fc0050
[10259.316525] RBP: ffffb9f60705fa80 R08: ffffb9f607060000 R09: 0000000000000000
[10259.316619] R10: 0000000000000002 R11: 0000000000000001 R12: ffffb9f60705fab0
[10259.316714] R13: 0000000000000002 R14: ffffb9f60705fad5 R15: ffff96b4c6d18400
[10259.316810] FS:  00007f73a07f6700(0000) GS:ffff96b8df680000(0000) knlGS:0000000000000000
[10259.316917] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[10259.316995] CR2: ffff96b4c6fc0068 CR3: 0000000809593000 CR4: 00000000000006e0
[10259.317090] Call Trace:
[10259.317137]  ? i2c_transfer+0x83/0xc0
[10259.317197]  read_status+0xc2/0x1740 [stv0910]
[10259.317265]  ? __enqueue_entity+0x6c/0x70
[10259.317339]  dvb_frontend_ioctl_legacy.isra.9+0xaa/0xbc0 [dvb_core]
[10259.317426]  ? check_preempt_wakeup+0x192/0x220
[10259.317491]  ? check_preempt_curr+0x79/0x90
[10259.317550]  ? ttwu_do_wakeup+0x19/0xe0
[10259.317606]  ? ttwu_do_activate+0x6f/0x80
[10259.317663]  ? try_to_wake_up+0x59/0x3d0
[10259.317720]  ? wake_up_process+0x15/0x20
[10259.317779]  ? insert_work+0xa2/0xf0
[10259.317831]  ? __queue_work+0x16a/0x460
[10259.321055]  ? tty_flip_buffer_push+0x2b/0x30
[10259.324282]  dvb_frontend_ioctl+0xb4/0x130 [dvb_core]
[10259.327507]  dvb_usercopy+0x5c/0x1d0 [dvb_core]
[10259.330721]  ? dvb_frontend_ioctl_properties+0xeb0/0xeb0 [dvb_core]
[10259.333918]  ? set_next_entity+0xc3/0x1a0
[10259.337056]  ? pick_next_task_fair+0x4a2/0x4d0
[10259.340135]  ? __switch_to+0x227/0x460
[10259.343140]  dvb_generic_ioctl+0x23/0x40 [dvb_core]
[10259.346122]  do_vfs_ioctl+0xa3/0x5f0
[10259.349113]  ? xen_build_mfn_list_list+0x255/0x420
[10259.352135]  SyS_ioctl+0x79/0x90
[10259.355181]  entry_SYSCALL_64_fastpath+0x1a/0xa9
[10259.358256] RIP: 0033:0x7f73a0114787
[10259.361332] RSP: 002b:00007fffab830958 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[10259.364513] RAX: ffffffffffffffda RBX: 0000000000000029 RCX: 00007f73a0114787
[10259.367633] RDX: 00007fffab830960 RSI: 0000000080046f45 RDI: 0000000000000003
[10259.370674] RBP: 000056103aad5060 R08: 0000000000000000 R09: 0000000000000029
[10259.373629] R10: 0000000000000073 R11: 0000000000000246 R12: 0000000000000029
[10259.376487] R13: 0000000000000001 R14: 00007f73a03d85e0 R15: 00007f73a03d43c0
[10259.379255] Code: 96 ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 0b fc c6 b4 96 ff ff 60 d0 6e c0 ff ff ff ff <60> ac 0a cc b4 96 ff ff 60 e9 de c6 b4 96 ff ff 3c 96 8a c0 ff 
[10259.384882] RIP: 0xffff96b4c6fc0068 RSP: ffffb9f60705fa60
[10259.387610] CR2: ffff96b4c6fc0068
[10259.403014] ---[ end trace 3ecf7e2a4ff3bc2a ]---
```

Launching again femon -H after that makes the command just being stuck.

Could someone help ?

lwandrebeck, 2017-05-02:
Hi again,

Same problem with 4.10.13.

Here is complete build log:
```
[root@eumetcast ~]# git clone https://github.com/tbsdtv/media_build.git
Clonage dans 'media_build'...
remote: Counting objects: 3596, done.
remote: Total 3596 (delta 0), reused 0 (delta 0), pack-reused 3596
Réception d'objets: 100% (3596/3596), 677.02 KiB | 0 bytes/s, fait.
Résolution des deltas: 100% (2561/2561), fait.
Vérification de la connectivité... fait.
[root@eumetcast ~]#  git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
Clonage dans './media'...
remote: Counting objects: 60881, done.
remote: Compressing objects: 100% (57295/57295), done.
remote: Total 60881 (delta 5360), reused 17681 (delta 2683), pack-reused 0
Réception d'objets: 100% (60881/60881), 160.09 MiB | 7.96 MiB/s, fait.
Résolution des deltas: 100% (5360/5360), fait.
Vérification de la connectivité... fait.
Extraction des fichiers: 100% (57373/57373), fait.
[root@eumetcast ~]#  cd media_build
[root@eumetcast media_build]#  make dir DIR=../media
make -C linux/ dir DIR="../../media"
make[1] : on entre dans le répertoire « /root/media_build/linux »
rm -rf drivers firmware include sound .patches_applied .linked_dir .git_log.md5 git_log kernel_version.h
Searching in ../../media/Makefile for kernel version.
./use_dir.pl ../../media
sync file: include/linux/dma-buf.h
sync file: firmware/av7110/Boot.S
sync file: sound/pci/bt87x.c
sync file: include/linux/fixp-arith.h
sync file: include/uapi/linux/v4l2-mediabus.h
sync file: include/linux/pci_ids.h
sync file: include/uapi/linux/media.h
sync file: firmware/vicam/firmware.H16
sync file: include/linux/via-core.h
sync file: include/uapi/linux/cec-funcs.h
sync file: include/uapi/linux/ivtv.h
sync file: include/uapi/linux/v4l2-dv-timings.h
sync file: include/uapi/linux/v4l2-common.h
sync file: firmware/ihex2fw.c
sync file: firmware/dvb-fe-mxl5xx.fw
sync file: firmware/av7110/bootcode.bin.ihex
sync file: include/linux/ti_wilink_st.h
sync file: include/trace/events/v4l2.h
sync file: include/uapi/linux/ivtvfb.h
sync file: include/linux/dma-fence.h
sync file: include/uapi/linux/v4l2-subdev.h
sync file: include/uapi/linux/smiapp.h
sync file: firmware/ttusb-budget/dspbootcode.bin.ihex
sync file: include/linux/mmc/sdio_ids.h
sync file: include/uapi/linux/v4l2-controls.h
sync file: include/misc/altera.h
sync file: include/uapi/linux/uvcvideo.h
sync file: include/uapi/linux/videodev2.h
sync file: include/linux/of_graph.h
sync file: include/uapi/linux/meye.h
sync file: include/uapi/linux/media-bus-format.h
sync file: include/uapi/linux/cec.h
sync file: include/linux/dma/xilinx_dma.h
sync file: include/linux/hdmi.h
sync file: include/uapi/linux/usb/video.h
sync file: include/linux/compiler-gcc.h
sync file: include/uapi/linux/xilinx-v4l2-controls.h
sync file: include/sound/aci.h
sync file: include/trace/events/vb2.h
sync file: firmware/cpia2/stv0672_vp4.bin.ihex
sync file: include/linux/kconfig.h
[... 3421 lines cut ...]
[mar. mai  2 15:59:28 2017]  dvb_usercopy+0x5c/0x1d0 [dvb_core]
[mar. mai  2 15:59:28 2017]  ? dvb_frontend_ioctl_properties+0xeb0/0xeb0 [dvb_core]
[mar. mai  2 15:59:28 2017]  ? tty_write_unlock+0x31/0x40
[mar. mai  2 15:59:28 2017]  ? tty_ldisc_deref+0x16/0x20
[mar. mai  2 15:59:28 2017]  ? tty_write+0x1d2/0x2f0
[mar. mai  2 15:59:28 2017]  ? n_tty_open+0xd0/0xd0
[mar. mai  2 15:59:28 2017]  dvb_generic_ioctl+0x23/0x40 [dvb_core]
[mar. mai  2 15:59:28 2017]  do_vfs_ioctl+0xa3/0x5f0
[mar. mai  2 15:59:28 2017]  SyS_ioctl+0x79/0x90
[mar. mai  2 15:59:28 2017]  entry_SYSCALL_64_fastpath+0x1a/0xa9
[mar. mai  2 15:59:28 2017] RIP: 0033:0x7f0a94767787
[mar. mai  2 15:59:28 2017] RSP: 002b:00007ffe0ac1af58 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[mar. mai  2 15:59:28 2017] RAX: ffffffffffffffda RBX: 00007f0a94a2b5e0 RCX: 00007f0a94767787
[mar. mai  2 15:59:28 2017] RDX: 00007ffe0ac1af60 RSI: 0000000080046f45 RDI: 0000000000000003
[mar. mai  2 15:59:28 2017] RBP: 00007f0a94a273c0 R08: 0000000000000000 R09: 0000000000000029
[mar. mai  2 15:59:28 2017] R10: 0000000000000073 R11: 0000000000000246 R12: 00005635927c5232
[mar. mai  2 15:59:28 2017] R13: 0000000000000004 R14: 0000000000000d68 R15: 00007f0a94a26880
[mar. mai  2 15:59:28 2017] Code: 8f ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 0b ae 0c 8f 8f ff ff 60 10 58 c0 ff ff ff ff <40> 0b 49 0c 8f 8f ff ff 20 27 38 0b 8f 8f ff ff 3c 76 67 c0 ff 
[mar. mai  2 15:59:28 2017] RIP: 0xffff8f8f0cae0068 RSP: ffffb62a44f4fa60
[mar. mai  2 15:59:28 2017] CR2: ffff8f8f0cae0068
[mar. mai  2 15:59:28 2017] ---[ end trace 1221e99153817a83 ]---
```

I’ll be glad to try patches.
Keep up the good work !

lwandrebeck, 2017-05-11:
Same problem with 4.10.14 and 4.10.15, will try to find a 4.9 kernel and test.

lwandrebeck, 2017-05-11:
Works fine with 4.9.14 (apparently at least, now I need to find some signal ^^).

lwandrebeck, 2017-05-11:
Still no luck :(

femon output:
```
Problem retrieving frontend information: Resource temporarily unavailable
status       | signal 0000 | snr 0000 | ber 00000000 | unc 00000001 | 
<snip>
```

femon gives in dmesg for both adapters:
```
[jeu. mai 11 16:27:00 2017] TBSECP3 driver 0000:05:00.0: DVB: adapter 0 frontend 0 frequency limits undefined - fix the driver
[jeu. mai 11 16:27:00 2017] TBSECP3 driver 0000:05:00.0: dvb_frontend_ioctl_legacy: doesn't know how to handle a DVBv3 call to delivery system 0
[jeu. mai 11 16:28:39 2017] TBSECP3 driver 0000:05:00.0: DVB: adapter 1 frontend 0 frequency limits undefined - fix the driver
[jeu. mai 11 16:28:39 2017] TBSECP3 driver 0000:05:00.0: dvb_frontend_ioctl_legacy: doesn't know how to handle a DVBv3 call to delivery system 0
[jeu. mai 11 16:29:15 2017] TBSECP3 driver 0000:05:00.0: DVB: adapter 0 frontend 0 frequency limits undefined - fix the driver
[jeu. mai 11 16:29:15 2017] TBSECP3 driver 0000:05:00.0: dvb_frontend_ioctl_legacy: doesn't know how to handle a DVBv3 call to delivery system 0
[jeu. mai 11 16:30:36 2017] TBSECP3 driver 0000:05:00.0: DVB: adapter 0 frontend 0 frequency limits undefined - fix the driver
[jeu. mai 11 16:30:36 2017] TBSECP3 driver 0000:05:00.0: dvb_frontend_ioctl_legacy: doesn't know how to handle a DVBv3 call to delivery system 0
[jeu. mai 11 16:30:44 2017] TBSECP3 driver 0000:05:00.0: DVB: adapter 1 frontend 0 frequency limits undefined - fix the driver
[jeu. mai 11 16:30:44 2017] TBSECP3 driver 0000:05:00.0: dvb_frontend_ioctl_legacy: doesn't know how to handle a DVBv3 call to delivery system 0
```

With /etc/channels.conf being:
```
Eutelsat 10A:11263:hC35S1M5O20:0:33000::::
```

```
szap -c /etc/channels.conf -n 1
reading channels from file '/etc/channels.conf'
zapping to 1 'Eutelsat 10A':
sat 0, frequency = 11263 MHz H, symbolrate 33000000, vpid = 0x1fff, apid = 0x1fff sid = 0x0000
using '/dev/dvb/adapter0/frontend0' and '/dev/dvb/adapter0/demux0'
Version: 5.10  	 FE_CAN {  }
Not a valid QPSK device!
```

In dmesg:

```
[jeu. mai 11 17:01:36 2017] BUG: unable to handle kernel NULL pointer dereference at 0000000000000010
[jeu. mai 11 17:01:36 2017] IP: [<ffffffff9d664d0e>] i2c_transfer+0xe/0xc0
[jeu. mai 11 17:01:36 2017] PGD 0 

[jeu. mai 11 17:01:36 2017] Oops: 0000 [#1] SMP
[jeu. mai 11 17:01:36 2017] Modules linked in: ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_broute bridge stp llc ebtable_nat ip6table_mangle ip6table_security ip6table_raw ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 iptable_mangle iptable_security iptable_raw iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ebtable_filter ebtables ip6table_filter ip6_tables stv6120(OE) stv0910(OE) tbsecp3(OE) amd64_edac_mod edac_mce_amd edac_core tas2101(OE) kvm_amd kvm i2c_mux dvb_core irqbypass sp5100_tco i2c_piix4 shpchp acpi_cpufreq k10temp tpm_tis tpm_tis_core tpm nfsd auth_rpcgss nfs_acl lockd grace sunrpc xfs libcrc32c mgag200 drm_kms_helper ttm drm igb ata_generic pata_acpi serio_raw ptp pata_atiixp pps_core dca 3w_9xxx i2c_algo_bit fjes
[jeu. mai 11 17:01:36 2017] CPU: 2 PID: 1985 Comm: kdvb-ad-0-fe-0 Tainted: G           OE   4.9.14-200.fc25.x86_64 #1
[jeu. mai 11 17:01:36 2017] Hardware name: Supermicro H8DGU/H8DGU, BIOS 1.0a    07/01/2010
[jeu. mai 11 17:01:36 2017] task: ffff9bd70ac33a80 task.stack: ffffafa483160000
[jeu. mai 11 17:01:36 2017] RIP: 0010:[<ffffffff9d664d0e>]  [<ffffffff9d664d0e>] i2c_transfer+0xe/0xc0
[jeu. mai 11 17:01:36 2017] RSP: 0018:ffffafa483163db0  EFLAGS: 00010246
[jeu. mai 11 17:01:36 2017] RAX: ffffafa483163ded RBX: ffff9bd30a082000 RCX: 0000000000000006
[jeu. mai 11 17:01:36 2017] RDX: 0000000000000001 RSI: ffffafa483163dd8 RDI: 0000000000000000
[jeu. mai 11 17:01:36 2017] RBP: ffffafa483163dc8 R08: ffffafa483163e80 R09: 0000000000000000
[jeu. mai 11 17:01:36 2017] R10: ffffafa483a6b908 R11: ffff9bd70ac33b00 R12: 000000000000f702
[jeu. mai 11 17:01:36 2017] R13: 000000000000f705 R14: 00000000000000ff R15: ffff9bd30a023c00
[jeu. mai 11 17:01:36 2017] FS:  0000000000000000(0000) GS:ffff9bd30fa80000(0000) knlGS:0000000000000000
[jeu. mai 11 17:01:36 2017] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[jeu. mai 11 17:01:36 2017] CR2: 0000000000000010 CR3: 0000000404408000 CR4: 00000000000006e0
[jeu. mai 11 17:01:36 2017] Stack:
[jeu. mai 11 17:01:36 2017]  ffff9bd30a082000 000000000000f702 000000000000f705 ffffafa483163df8
[jeu. mai 11 17:01:36 2017]  ffffffffc0434298 ffff000300000010 ffffafa483163ded 0602f70000000216
[jeu. mai 11 17:01:36 2017]  000000006498646f ffffafa483163e28 ffffffffc04346f3 0000000000000b01
[jeu. mai 11 17:01:36 2017] Call Trace:
[jeu. mai 11 17:01:36 2017]  [<ffffffffc0434298>] i2c_write_reg16+0x58/0x80 [stv0910]
[jeu. mai 11 17:01:36 2017]  [<ffffffffc04346f3>] send_burst+0x63/0xd0 [stv0910]
[jeu. mai 11 17:01:36 2017]  [<ffffffffc0464668>] dvb_frontend_thread+0x248/0x700 [dvb_core]
[jeu. mai 11 17:01:36 2017]  [<ffffffff9d0e8230>] ? prepare_to_wait_event+0x100/0x100
[jeu. mai 11 17:01:36 2017]  [<ffffffffc0464420>] ? dtv_set_frontend+0x420/0x420 [dvb_core]
[... 88 lines cut ...]
[jeu. mai 11 17:01:56 2017] RBP: ffffafa4843a7e78 R08: 0000000000000000 R09: 0000000000000000
[jeu. mai 11 17:01:56 2017] R10: 00000000ffffffff R11: ffff9bd30ba19dc0 R12: ffffafa4843a7f18
[jeu. mai 11 17:01:56 2017] R13: 0000000000000282 R14: 0000000000000001 R15: 0000000000000003
[... cut ...]
[jeu. mai 11 17:01:56 2017] Code: 0f 1f 44 00 00 55 48 89 e5 41 57 41 56 41 55 41 54 4c 8d 67 08 53 41 89 f7 48 83 ec 10 89 55 cc 48 8b 57 08 4c 89 45 d0 49 39 d4 <48> 8b 32 74 40 48 8d 42 e8 4c 8d 6e e8 41 89 ce 8b 18 48 8b 4d 
[jeu. mai 11 17:01:56 2017] RIP  [<ffffffff9d0e7bdb>] __wake_up_common+0x2b/0x80
[jeu. mai 11 17:01:56 2017]  RSP <ffffafa4843a7e40>
[jeu. mai 11 17:01:56 2017] CR2: 0000000048012587
[jeu. mai 11 17:01:56 2017] ---[ end trace b62c91824e5c4c40 ]---
[jeu. mai 11 17:01:56 2017] Fixing recursive fault but reboot is needed!
```

Tried to reboot the box, which does not answer anymore, have to go to DC. More news later.

crazycat69, 2017-05-11:
Very strange, now test under 4.9 kernel (Debian 9 RC) and TBS 6903 - all ok.