tbsdtv/linux_media #51: linux-4.12-rc1: saa716x failes to build

bas-t, 2017-05-14:
linux-4.12-rc1: saa716x failes to build due to

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/pci/msi.c?h=v4.12-rc1&id=4244de1c64ded7f5438717bdce3fa074efd20efb

Failing with:

drivers/media/pci/saa716x/saa716x_pci.c: In function ‘saa716x_enable_msix’:
drivers/media/pci/saa716x/saa716x_pci.c:46:8: error: implicit declaration of function ‘pci_enable_msix’ [-Werror=implicit-function-declaration]
  ret = pci_enable_msix(pdev, saa716x->msix_entries, SAA716x_MSI_MAX_VECTORS);
        ^~~~~~~~~~~~~~~

I really hope that you guys can patch saa716x so it can live past linux-4.12.x.
Kind regards,
Tycho.

bas-t, 2017-05-17:
The above mentioned patch is part 4 of 4. Maybe the other ones are interesting too:

https://patchwork.kernel.org/patch/9667013/

https://patchwork.kernel.org/patch/9667023/

https://patchwork.kernel.org/patch/9667027/

Regards, Tycho.

crazycat69, 2017-06-25:
saa716x: replace pci_enable_msix() with pci_enable_msix_exact() 
fb349ffe41d5e198088bd046bd1c82dacf4eb8c4

bas-t, 2017-06-26:
@crazycat69 Thanks, applied in https://github.com/bas-t/saa716x/commit/75eb1e63111a74b2135dd01ab524cee7f37e063b and compiling right now.
Will report back.
Cheers!

bas-t, 2017-06-26:
@crazycat69 
Works OK for me.
Going to test backwards compatibility with a regular Debian Stretch stock kernel (4.9.whatever)
If that test succeeds too, I'll close the issue.
Thanks again and regards,
Tycho.

bas-t, 2017-06-26:
Backwards compatibility with a regular Debian Stretch stock kernel confirmed.

closed 2017-06-26 by bas-t