tbsdtv/linux_media #53: TBS 5520 SE

cyberkrel, 2017-05-18:
Help to understand the kernel failure when connecting the tuner.

[   34.282523@2] usb 1-2: new high-speed USB device number 2 using xhci-hcd
[   34.422594@2] usb 1-2: New USB device found, idVendor=734c, idProduct=5521
[   34.422607@2] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   34.443628@3] media: Linux media interface: v0.10
[   34.449909@3] WARNING: You are using an experimental version of the media stack.
[   34.449909@3]        As the driver is backported to an older kernel, it doesn't offer
[   34.449909@3]        enough quality for its usage in production.
[   34.449909@3]        Use it with care.
[   34.449909@3] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
[   34.449909@3]
[   34.455481@3] WARNING: You are using an experimental version of the media stack.
[   34.455481@3]        As the driver is backported to an older kernel, it doesn't offer
[   34.455481@3]        enough quality for its usage in production.
[   34.455481@3]        Use it with care.
[   34.455481@3] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
[   34.455481@3]
[   34.463765@3] dvb-usb: found a 'TBS 5520se USB2.0' in cold state, will try to load a firmware
[   34.465779@3] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5521.fw'
[   34.465784@3] tbs5520se: start downloading TBS5520se firmware
[   34.472065@3] usb 1-2: USB disconnect, device number 2
[   34.572369@2] dvb-usb: found a 'TBS 5520se USB2.0' in warm state.
[   34.573702@3] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   34.573818@3] dvbdev: DVB: registering new adapter (TBS 5520se USB2.0)
[   34.573845@3] usb 1-2: media controller created
[   34.573854@3] tbs5520se: read eeprom failed.
[   34.573858@3] dvb-usb: MAC address reading failed.
[   34.575154@3] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[   34.593743@3] i2c i2c-1: Added multiplexed i2c bus 2
[   34.593749@3] si2183 1-0067: Silicon Labs Si2183 successfully attached
[   34.601839@3] si2157 2-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   34.611210@3] i2c i2c-2: av201x: Airoha Technology AV201x successfully attached
[   34.611228@3] usb 1-2: DVB: registering adapter 0 frontend 0 (TBS 5520se USB2.0)...
[   34.611253@3] dvbdev: dvb_create_media_entity: media entity 'TBS 5520se USB2.0' registered.
[   34.611843@3] usb 1-2: DVB: registering adapter 0 frontend 1 (Silicon Labs Si2183)...
[   34.611861@3] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2183' registered.
[   34.615440@3] dvb-usb: TBS 5520se USB2.0 successfully initialized and connected.
[   34.616975@3] usbcore: registered new interface driver tbs5520se
[   34.628599@0] dvb-usb: TBS 5520se USB2.0 successfully deinitialized and disconnected.
[   36.262366@2] usb 1-2: new high-speed USB device number 3 using xhci-hcd
[   36.402658@2] usb 1-2: New USB device found, idVendor=734c, idProduct=5521
[   36.402663@2] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   36.402666@2] usb 1-2: Product: TBS 5521
[   36.402669@2] usb 1-2: Manufacturer: TBS-Tech
[   36.405542@3] dvb-usb: found a 'TBS 5520se USB2.0' in cold state, will try to load a firmware
[   36.405564@3] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5521.fw'
[   36.405568@3] tbs5520se: start downloading TBS5520se firmware
[   36.512341@2] dvb-usb: found a 'TBS 5520se USB2.0' in warm state.
[   36.512898@2] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   36.513008@2] dvbdev: DVB: registering new adapter (TBS 5520se USB2.0)
[   36.513021@2] usb 1-2: media controller created
[   36.594002@2] dvb-usb: MAC address: 00:44:ab:c0:26:d2
[   36.594943@2] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[   36.600709@2] i2c i2c-1: Added multiplexed i2c bus 2
[   36.600716@2] si2183 1-0067: Silicon Labs Si2183 successfully attached
[   36.606201@2] si2157 2-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   36.606275@2] i2c i2c-2: av201x: Airoha Technology AV201x successfully attached
[   36.666605@2] usb 1-2: DVB: registering adapter 0 frontend 0 (TBS 5520se USB2.0)...
[   36.666626@2] dvbdev: dvb_create_media_entity: media entity 'TBS 5520se USB2.0' registered.
[   36.666907@0] usb 1-2: DVB: registering adapter 0 frontend 1 (Silicon Labs Si2183)...
[   36.666922@0] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2183' registered.
[   36.667176@0] ------------[ cut here ]------------
[   36.667192@0] WARNING: CPU: 0 PID: 1183 at fs/sysfs/dir.c:52 sysfs_warn_dup+0x8c/0xb0()
[   36.667193@0] sysfs: cannot create duplicate filename '/class/dvb/dvb0.frontend1'
[   36.667219@0] Modules linked in: av201x(O) si2157(O) si2183(O) dvb_usb_tbs5520se(O) dvb_usb(O) dvb_core(O) rc_core(O) media(O) 8189es 8021q cfg80211 wifi_dummy mali(O) aml_nftl_dev(P) fbcon bitblit softcursor font
[   36.667224@0] CPU: 0 PID: 1183 Comm: khubd Tainted: P           O 3.14.29 #1
[   36.667226@0] Call trace:
[   36.667233@0] [<ffffffc001087e08>] dump_backtrace+0x0/0x128
[   36.667236@0] [<ffffffc001087f50>] show_stack+0x20/0x30
[   36.667242@0] [<ffffffc0017e0d54>] dump_stack+0x78/0x94
[   36.667249@0] [<ffffffc001095714>] warn_slowpath_common+0x94/0xc0
[   36.667252@0] [<ffffffc0010957a0>] warn_slowpath_fmt+0x60/0x78
[   36.667257@0] [<ffffffc0011fa1a8>] sysfs_warn_dup+0x88/0xb0
[   36.667260@0] [<ffffffc0011fa68c>] sysfs_do_create_link_sd.isra.0+0xec/0x100
[   36.667262@0] [<ffffffc0011fa6dc>] sysfs_create_link+0x3c/0x68
[   36.667267@0] [<ffffffc0013e93a0>] device_add+0x420/0x520
[   36.667270@0] [<ffffffc0013e9724>] device_create_groups_vargs+0x124/0x138
[   36.667274@0] [<ffffffc0013e9834>] device_create+0x84/0xa0
[   36.667284@0] [<ffffffbffc25ca7c>] dvb_register_device+0x224/0x638 [dvb_core]
[   36.667291@0] [<ffffffbffc2652cc>] dvb_register_frontend+0x194/0x1f0 [dvb_core]
[   36.667296@0] [<ffffffbffc27cd70>] dvb_usb_adapter_frontend_init+0xb0/0x228 [dvb_usb]
[   36.667300@0] [<ffffffbffc27bb40>] dvb_usb_device_init+0x4a0/0x6f0 [dvb_usb]
[   36.667305@0] [<ffffffbffc2850cc>] tbs5520se_probe+0x34/0x50 [dvb_usb_tbs5520se]
[   36.667311@0] [<ffffffc001477cb0>] usb_probe_interface+0xd8/0x1e8
[   36.667315@0] [<ffffffc0013ec3ec>] really_probe+0x74/0x230
[   36.667318@0] [<ffffffc0013ec610>] __device_attach+0x68/0x78
[   36.667321@0] [<ffffffc0013ea61c>] bus_for_each_drv+0x64/0xa8
[   36.667325@0] [<ffffffc0013ec344>] device_attach+0xa4/0xd8
[   36.667328@0] [<ffffffc0013eb7ec>] bus_probe_device+0x9c/0xc8
[   36.667330@0] [<ffffffc0013e92f8>] device_add+0x378/0x520
[   36.667334@0] [<ffffffc0014765c8>] usb_set_configuration+0x440/0x7c8
[   36.667338@0] [<ffffffc0014808b8>] generic_probe+0x38/0xc0
[   36.667340@0] [<ffffffc001476a24>] usb_probe_device+0x24/0x38
[   36.667344@0] [<ffffffc0013ec3ec>] really_probe+0x74/0x230
[   36.667347@0] [<ffffffc0013ec610>] __device_attach+0x68/0x78
[   36.667350@0] [<ffffffc0013ea61c>] bus_for_each_drv+0x64/0xa8
[   36.667353@0] [<ffffffc0013ec344>] device_attach+0xa4/0xd8
[   36.667356@0] [<ffffffc0013eb7ec>] bus_probe_device+0x9c/0xc8
[   36.667359@0] [<ffffffc0013e92f8>] device_add+0x378/0x520
[   36.667362@0] [<ffffffc00146d71c>] usb_new_device+0x1e4/0x320
[   36.667364@0] [<ffffffc00146ed9c>] hub_thread+0xc54/0x1028
[   36.667370@0] [<ffffffc0010b95fc>] kthread+0xd4/0xf0
[   36.667372@0] ---[ end trace bc4a632bd562acf6 ]---
[   36.667592@0] dvbdev: dvb_register_device: failed to create device dvb0.frontend1 (-17)
[   36.668750@2] dvb-usb: TBS 5520se USB2.0 successfully initialized and connected.

crazycat69, 2017-05-18:
some dvb_create_media_entity() issues with registration second virtual frontend instance.

crazycat69, 2017-05-19:
Please check with latest changes
2f87a1257b1e2b18121dcaf7a4d7844688115ead
00c1eb060dd39aa20b29140ebd6efa3b9c48840e

cyberkrel, 2017-05-20:
The error repeats, you could not see through SSH
Thank you.

cyberkrel, 2017-05-20:
Start in cold state
----------------------------------------------------------------------------------------------------------
[  118.495290] media: Linux media interface: v0.10
[  118.497106] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                00c1eb060dd39aa20b29140ebd6efa3b9c48840e tbs5520se: Fix frontend names.
[  118.499740] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                00c1eb060dd39aa20b29140ebd6efa3b9c48840e tbs5520se: Fix frontend names.
[  118.501484] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[  118.501838] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[  118.501839] tbs5520se: start downloading TBS5520se firmware
[  118.614240] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[  118.614434] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  118.614502] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)
[  118.695294] dvb-usb: MAC address: 00:22:ab:c0:44:d2
[  118.698607] i2c i2c-8: Added multiplexed i2c bus 9
[  118.698608] si2183 8-0067: Silicon Labs Si2183 successfully attached
[  118.701963] si2157 9-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[  118.703780] i2c i2c-9: av201x: Airoha Technology AV201x successfully attached
[  118.764155] usb 3-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[  118.764360] usb 3-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[  118.764657] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[  118.764736] usbcore: registered new interface driver tbs5520se

----------------------------------------------------------------------------------------------------------
Replug USB
----------------------------------------------------------------------------------------------------------

[  196.280347] usb 3-1: USB disconnect, device number 2
[  196.281381] dvb-usb: TurboSight TBS 5520SE successfully deinitialized and disconnected.
[  203.970625] usb 3-1: new high-speed USB device number 3 using xhci_hcd
[  204.111236] usb 3-1: New USB device found, idVendor=734c, idProduct=5521
[  204.111241] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  204.111244] usb 3-1: Product: TBS 5521
[  204.111246] usb 3-1: Manufacturer: TBS-Tech
[  204.111842] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[  204.111847] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[  204.111850] tbs5520se: start downloading TBS5520se firmware
[  204.222572] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[  204.222786] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  204.222891] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)
[  204.305968] dvb-usb: MAC address: 00:22:ab:c0:44:d2
[  204.307554] i2c i2c-8: Added multiplexed i2c bus 9
[  204.307557] si2183 8-0067: Silicon Labs Si2183 successfully attached
[  204.310425] si2157 9-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[  204.310434] i2c i2c-9: av201x: Airoha Technology AV201x successfully attached
[  204.370662] usb 3-1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[  204.370773] usb 3-1: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[  204.370783] ------------[ cut here ]------------
[  204.370788] WARNING: CPU: 2 PID: 44 at /build/linux-lz1RHE/linux-4.10.0/fs/sysfs/dir.c:31 sysfs_warn_dup+0x62/0x80
[  204.370789] sysfs: cannot create duplicate filename '/class/dvb/dvb0.frontend1'

cyberkrel, 2017-05-21:
So works, I will check

diff --git a/drivers/media/usb/dvb-usb/dvb-usb-dvb.c b/drivers/media/usb/dvb-usb/dvb-usb-dvb.c
index 5f265e68..c70406c6 100644
--- a/drivers/media/usb/dvb-usb/dvb-usb-dvb.c
+++ b/drivers/media/usb/dvb-usb/dvb-usb-dvb.c
@@ -356,6 +356,8 @@ int dvb_usb_adapter_frontend_exit(struct dvb_usb_adapter *adap)
                if (adap->fe_adap[i].fe != NULL) {
                        dvb_unregister_frontend(adap->fe_adap[i].fe);
                        dvb_frontend_detach(adap->fe_adap[i].fe);
+                        dvb_unregister_frontend(adap->fe_adap[i].fe2);
+                        dvb_frontend_detach(adap->fe_adap[i].fe2);
                }
        }
        adap->num_frontends_initialized = 0;

ckuethe, 2017-11-03:
Your patch works with `4.13.11-041311-generic #201711020532 SMP Thu Nov 2 09:33:54 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux` and driver cf382935d479868da551bf1593b3ee24afd5d54d

closed 2017-11-04 by crazycat69