tbsdtv/linux_media #54: TBS 5881 DVB-T/T2/C + CI TVHeadend  poll TIMEOUT

m-kad, 2017-05-23:
Everything worked find with closed source drivers, but with open source...

dmesg
```4.688346] dvb-usb: found a 'TurboSight TBS 5881 DVB-T/T2/C + CI' in cold state, will try to load a firmware
[    4.688350] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5881.fw'
[    4.688352] tbs5881: start downloading TBS5881 CI firmware
[    4.817293] dvb-usb: found a 'TurboSight TBS 5881 DVB-T/T2/C + CI' in warm state.
[    4.817388] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    4.817490] dvbdev: DVB: registering new adapter (TurboSight TBS 5881 DVB-T/T2/C + CI)
[    5.005150] dvb-usb: MAC address: 00:22:ab:b1:04:0d
[    5.013445] i2c i2c-5: Added multiplexed i2c bus 6
[    5.013451] si2168 5-0064: Silicon Labs Si2168-A20 successfully identified
[    5.013455] si2168 5-0064: firmware version: A 2.0.3
[    5.017332] IPv6: ADDRCONF(NETDEV_UP): enp0s31f6: link is not ready
[    5.022446] si2157 6-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.042951] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[    5.043243] tbs5881: CI initialized.
[    5.043649] tbs5881: CI disabled.
[    5.043659] usb 1-6: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5881 DVB-T/T2/C + CI)...
[    5.044135] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-6/input/input20
[    5.044269] dvb-usb: schedule remote query interval to 150 msecs.
[    5.044273] dvb-usb: TurboSight TBS 5881 DVB-T/T2/C + CI successfully initialized and connected.
[    6.068127] [drm] RC6 on
[    7.829621] e1000e: enp0s31f6 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[    7.829681] IPv6: ADDRCONF(NETDEV_CHANGE): enp0s31f6: link becomes ready
[    7.927795] igb 0000:04:00.0 enp4s0: igb: enp4s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[    8.047497] IPv6: ADDRCONF(NETDEV_CHANGE): enp4s0: link becomes ready
[   14.125239] tun: Universal TUN/TAP device driver, 1.6
[   14.125239] tun: (C) 1999-2004 Max Krasnyansky <maxk@qualcomm.com>
[   14.133571] NFSD: starting 90-second grace period (net ffffffff81ac2640)
[   14.237285] usb 1-5: reset full-speed USB device number 2 using xhci_hcd
[   14.290154] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
[   17.926417] si2168 5-0064: firmware version: \xffffff81 2.0.35
[   17.935248] si2157 6-0060: found a 'Silicon Labs Si2158-A20'
[   17.935512] si2157 6-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[   18.457780] si2157 6-0060: firmware version: 2.1.9
[   18.457821] usb 1-6: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)
[   18.460944] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
[   22.137475] si2168 5-0064: firmware version: \xffffff81 2.0.35
```
TVHeadend
```2017-05-23 21:18:28.888 linuxdvb: TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0 - poll TIMEOUT
2017-05-23 21:18:36.548 linuxdvb: TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0 - poll TIMEOUT
2017-05-23 21:18:44.208 linuxdvb: TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0 - poll TIMEOUT
2017-05-23 21:18:51.869 linuxdvb: TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0 - poll TIMEOUT
2017-05-23 21:18:59.529 linuxdvb: TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0 - poll TIMEOUT
```

crazycat69, 2017-05-23:
Dvb-c annex a or b ? Modulation, SR ?

m-kad, 2017-05-23:
Modulation: QAM 256
SR: 6,875 Msym/s
Annex: A? should be (i'm in europe)

marcinulan, 2017-09-09:
I have same situation.
`dmesg | grep dvb`
`[  174.597856] dvb-usb: found a 'TurboSight TBS 5881 DVB-T/T2/C + CI' in cold state, will try to load a firmware
[  174.598385] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5881.fw'
[  174.712088] dvb-usb: found a 'TurboSight TBS 5881 DVB-T/T2/C + CI' in warm state.
[  174.713683] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  174.713861] dvbdev: DVB: registering new adapter (TurboSight TBS 5881 DVB-T/T2/C + CI)
[  174.817331] dvb-usb: MAC address: 00:22:ab:b1:03:26
[  174.906624] dvb-usb: schedule remote query interval to 150 msecs.
[  174.906634] dvb-usb: TurboSight TBS 5881 DVB-T/T2/C + CI successfully initialized and connected.
[  174.921248] si2168 12-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  175.923301] si2157 13-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[  176.616762] si2168 12-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  178.184222] dvb_ca_en50221: dvb_ca adapter 0: DVB CAM detected and initialised successfully
`
tvheadend:
`017-09-09 22:43:59.765 mpegts: 682MHz in JMDI - tuning on TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0
2017-09-09 22:43:59.766 subscription: 0036: "scan" subscribing to mux "682MHz", weight: 5, adapter: "TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0", network: "JMDI", service: "Raw PID Subscription"
2017-09-09 22:44:00.719 linuxdvb: TurboSight TBS 5881 DVB-T/T2/C + CI #0 : DVB-C #0 - poll TIMEOUT
2017-09-09 22:44:04.763 mpegts: 682MHz in JMDI - scan no data, failed
2017-09-09 22:44:04.764 subscription: 0036: "scan" unsubscribing`