tbsdtv/linux_media #55: driver not compiling on ubuntu 14.04.5 4.4.0-31-generic

husamaga, 2017-06-02:
Hi, i get issue with the driver on ubuntu 14.04.5 kernel:  4.4.0-31-generic 
card type tbs 6209.

/root/driver/github/media_build/v4l/dvb_demux.c:23:32: fatal error: linux/sched/signal.h: No such file or directory
 #include <linux/sched/signal.h>
                                ^
compilation terminated.
  CC [M]  /root/driver/github/media_build/v4l/dvb_ca_en50221.o
  CC [M]  /root/driver/github/media_build/v4l/dvb_frontend.o
  CC [M]  /root/driver/github/media_build/v4l/dvb_net.o
/root/driver/github/media_build/v4l/dvb_ca_en50221.c:38:32: fatal error: linux/sched/signal.h: No such file or directory
 #include <linux/sched/signal.h>
                                ^
compilation terminated.
/root/driver/github/media_build/v4l/dvb_frontend.c:33:32: fatal error: linux/sched/signal.h: No such file or directory
 #include <linux/sched/signal.h>
                                ^
compilation terminated.
make[3]: *** [/root/driver/github/media_build/v4l/dvb_demux.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [/root/driver/github/media_build/v4l/dvb_ca_en50221.o] Error 1
make[3]: *** [/root/driver/github/media_build/v4l/dvb_frontend.o] Error 1
make[2]: *** [_module_/root/driver/github/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-4.4.0-31-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/root/driver/github/media_build/v4l'
make: *** [all] Error 2

Thanks

nightshadow36, 2017-06-28:
I am having the same error message on Ubuntu 14.04  Kernel 3.13.0-123-generic.
Is there a fix for this. Its running on a live IPTV server so i went back to kernel 3.13.0.120

husamaga, 2017-07-07:
Hi,

Yes it works, you need to compile using the opensource driver.

You can use this as a guide.

https://www.tbsdtv.com/forum/viewtopic.php?f=86&t=9875

TBSDTV Community Forum • View topic - How to install linux ...<https://www.tbsdtv.com/forum/viewtopic.php?f=86&t=9875>
www.tbsdtv.com
TBSDTV Forum is a community to discuss Digital TV Tuner and industry news, reviews, guides. It provides solutions of watching DVB satellite television, terrestrial TV ...

CvH, 2017-07-07:
> You can use this as a guide.

@husamaga 
that guide explain how to install the old drivers that are only available for x86

the *new* guide is here https://github.com/tbsdtv/linux_media/wiki

nightshadow36, 2017-07-08:
Dont work on the latest kernal.been using that guide on every kernal update

nightshadow36, 2017-08-08:
@crazycat69 
Any fix for this i get the same install error on the latest ubuntu 14.04 kernel.
I have reinstalled many times when there has been a kernel up date but it will not install any more.
Something I am missing?

crazycat69, 2017-08-08:
3.13 kernel ?

nightshadow36, 2017-08-08:
@crazycat69 
Yes last time i tried was on  kernel 3.13.0-123 this is when i got the make: *** [all] Error 2
I am using an older kernel now the last one that worked for me 3.13.0-123
Thanks

crazycat69, 2017-08-09:
f51964675c995c22ea6ce6b77ad2ed635864acb1

Success build on my development PC with Kubuntu 14.04 and 3.13.0-126 + Xenial LTS 4.4.0-75 kernels.
If you have some problems with media_build patches - use **./build --main-git** for first-time build (instead ["magic recipe" from wiki](https://github.com/tbsdtv/linux_media/wiki)).

P.S. Anyway i recommend install [4.4.0 Xenial LTS kernel for *buntu 14.04](https://wiki.ubuntu.com/Kernel/LTSEnablementStack#Ubuntu_14.04_LTS_-_Trusty_Tahr). Some frontend modules not compiled for <4.4 kernels (cxd2820r for example for old TBS ter/cable devices).

nightshadow36, 2017-08-24:
@crazycat69 
[image]

I am still getting some errors.

crazycat69, 2017-08-24:
look like your local linux_media sources not updated with my latest commit
f51964675c995c22ea6ce6b77ad2ed635864acb1

nightshadow36, 2017-08-26:
@crazycat69 
"If you have some problems with media_build patches - use ./build --main-git for first-time build"
Sorry i don`t understand. Is there other instructions I can use?
Thank you.

crazycat69, 2017-08-26:
This usual build practice, see README for media_build.

nightshadow36, 2017-08-26:
Read me changed location and cannot find media_build.
You say to use o./build --main-git and not the Wiki instructions.
Can i send you some money via paypal to post or send the instructions that will work?
Thank you.

crazycat69, 2017-08-26:
```
git clone https://github.com/tbsdtv/media_build.git
cd media_build
./build --main-git
make install

```

nightshadow36, 2017-08-27:
@crazycat69 
Thank you all three servers updated and working.
You are a Star.
:)