tbsdtv/linux_media #56: TBS 6985 not detected after driver install

utelisysadmin, 2017-06-14:
There are:
4 6991 cards
2 6985 cards
in the server. 

They all work with proprietary driver, but the server freezes every 4.2 hours. So I decided to give the OS driver a go.

uname -a
Linux tv-box 4.8.0-54-generic #57~16.04.1-Ubuntu SMP Wed May 24 16:22:28 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

lspci -v | grep --after-context=10 7160
0e:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
	Subsystem: Device 6991:0002
	Flags: bus master, fast devsel, latency 0, IRQ 47
	Memory at de600000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
	Capabilities: [50] Express Endpoint, MSI 00
	Capabilities: [74] Power Management version 2
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel driver in use: SAA716x Budget
	Kernel modules: saa716x_budget
--
0f:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
	Subsystem: Device 6991:0002
	Flags: bus master, fast devsel, latency 0, IRQ 49
	Memory at de500000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
	Capabilities: [50] Express Endpoint, MSI 00
	Capabilities: [74] Power Management version 2
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel driver in use: SAA716x Budget
	Kernel modules: saa716x_budget
--
12:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
	Subsystem: Device 6991:0002
	Flags: bus master, fast devsel, latency 0, IRQ 50
	Memory at de800000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
	Capabilities: [50] Express Endpoint, MSI 00
	Capabilities: [74] Power Management version 2
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel driver in use: SAA716x Budget
	Kernel modules: saa716x_budget
--
13:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
	Subsystem: Device 6991:0002
	Flags: bus master, fast devsel, latency 0, IRQ 52
	Memory at de700000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
	Capabilities: [50] Express Endpoint, MSI 00
	Capabilities: [74] Power Management version 2
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel driver in use: SAA716x Budget
	Kernel modules: saa716x_budget
--
15:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)
	Subsystem: Device 6985:0002
	Physical Slot: 6
	Flags: fast devsel, IRQ 16
	Memory at de900000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
	Capabilities: [50] Express Endpoint, MSI 00
	Capabilities: [74] Power Management version 2
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel modules: saa716x_budget
[... 51 lines cut ...]
Jun 14 17:38:12 tv-box kernel: [ 1144.028257] i2c i2c-21: av201x: Airoha Technology AV201x successfully attached
Jun 14 17:38:12 tv-box kernel: [ 1144.028262] SAA716x Budget 0000:12:00.0: DVB: registering adapter 5 frontend 0 (Tmax TAS2101)...
Jun 14 17:38:12 tv-box kernel: [ 1144.056452] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
Jun 14 17:38:12 tv-box kernel: [ 1144.057018] i2c i2c-23: Added multiplexed i2c bus 24
Jun 14 17:38:12 tv-box kernel: [ 1144.057144] i2c i2c-23: Added multiplexed i2c bus 25
Jun 14 17:38:12 tv-box kernel: [ 1144.164242] i2c i2c-25: av201x: Airoha Technology AV201x successfully attached
Jun 14 17:38:12 tv-box kernel: [ 1144.164248] SAA716x Budget 0000:13:00.0: DVB: registering adapter 6 frontend 0 (Tmax TAS2101)...
Jun 14 17:38:12 tv-box kernel: [ 1144.164494] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
Jun 14 17:38:12 tv-box kernel: [ 1144.165078] i2c i2c-22: Added multiplexed i2c bus 26
Jun 14 17:38:12 tv-box kernel: [ 1144.165210] i2c i2c-22: Added multiplexed i2c bus 27
Jun 14 17:38:12 tv-box kernel: [ 1144.272230] i2c i2c-27: av201x: Airoha Technology AV201x successfully attached
Jun 14 17:38:12 tv-box kernel: [ 1144.272235] SAA716x Budget 0000:13:00.0: DVB: registering adapter 7 frontend 0 (Tmax TAS2101)...

I also have tbs.conf in /etc/modprobe.d/:
options tbs_pcie-dvb tbs_int_type=1
options saa716x_tbs-dvb int_type=1

lsmod | grep budget
saa716x_budget         40960  0
tas2101                24576  9 saa716x_budget
cx24117                36864  1 saa716x_budget
saa716x_core           73728  1 saa716x_budget
dvb_core              131072  2 saa716x_budget,saa716x_core

Followed the instructions to install driver from wiki.

crazycat69, 2017-06-14:
8 adapters detected, so maybe need increase max number of adapters ?

utelisysadmin, 2017-06-15:
It was my understanding that this was increased to 48, but I think I am mixing the closed source repo with OS one. 

Increased to 64 an all cards detected now, however none of the hardware CAMs are detected.

utelisysadmin, 2017-06-15:
Actually getting kernel traces:

[kernel.txt](https://github.com/tbsdtv/linux_media/files/1077100/kernel.txt)

utelisysadmin, 2017-06-15:
Wiped the driver and re-did the installation. Now all cards and CAMs are detected, however when I start more than 4 cards they all start throwing discontinuities.

crazycat69, 2017-06-15:
Try disable MSI interrupt

utelisysadmin, 2017-06-15:
I did. Its the same situation. I went back to the closed-source driver to confirm. There I get discontinuities as well, but not as many.