tbsdtv/linux_media #58: TBS6814 Signal attributes

victor-accarini, 2017-06-20:
Hi,

I have three different boards (6528, 6522 and 6814) and when I try to retrieve their signal attributes from the same source (Brazilian ISDB-T, 6Mhz bandwidth, ~641Mhz) I get the following data:

- 6528/6522
```
> femon -H
status SCVYL | signal  50% | snr  77% | ber 0 | unc 0 | FE_HAS_LOCK
> dvbv5-zap
          (0x00) Signal=-100dBm                            ** Without signal **
Lock   (0x1f) Signal= -50.00dBm C/N= 34.50dB ** With signal **
Lock   (0x1f) Signal= -50.00dBm C/N= 34.50dB
Lock   (0x1f) Signal= -50.00dBm C/N= 38.00dB
```
- 6814 
```
>femon -H
status SCVYL | signal  99% | snr  40% | ber 0 | unc 1 | FE_HAS_LOCK
> dvbv5-zap
Timeout(0x20) Signal= 99.18% C/N= 0.00% postBER= 10 ** Without signal **
Timeout(0x20) Signal= 99.18% C/N= 0.00% postBER= 10
Lock   (0x1f) Signal= 99.18% C/N= 37.84% postBER= 0     ** With signal **
```

For the 6528/6522 I believe those values are *good enough* but for the 6814 the signal level is a little weird since when I remove the antenna I get 99%.

I did some research in the Brazilian Telecommunication Agency website and I found a graph with the optimal relationship between signal level(SL) and C/N [article (in Portuguese)](http://www.anatel.gov.br/Portal/verificaDocumentos/documento.asp?numeroPublicacao=237823&assuntoPublicacao=null&caminhoRel=null&filtro=1&documentoPath=biblioteca/publicacao/tvdigital_cp216/parte_25.pdf).

It basically states that the useful signal level would be between -75dBm and -30dBm and noise level(NL) should be 19dB below from signal levels from -30dBm to -65dBm. For signals less than -65dBm it should lower (I believe) exponentially.

`CN = SL - NL`

| SL (dBm) | NL (dBm) *maximum value | CN (dB) *minimal value |
|----|----|----|
|-30| -49 | 19 |
|-40| -59 | 19 |
|-50| -69 | 19 |
|-60| -79 | 19 |
|-73| -105 | 32 |
*All those values are approximation made from the article's graph.*

I don't know if those values apply for the ISDB-T signal in Japan or if they can help, but can you guys take a look into the drivers to make the information retrieved from the boards a little more uniform?

Thanks

crazycat69, 2017-06-20:
Extended signal statistic (Strength in dBm, SNR in dB) not implemented for mtv23x demod driver (6814). Legacy statistic (%) also look like incorrect.

nrmmota, 2017-06-20:
I would be careful with TBS6814. One of the 2016 batches was faulty.
Couldn't get it to stream properly no matter what.

victor-accarini, 2017-06-21:
@crazycat69 Can you guys fix the legacy statistics?
@nrmmota Thanks for the warning

crazycat69, 2017-06-21:
If you can provide remote access to your linux PC with installed development tools and some free hdd space. + ter antenna with real isdb-t signal must be connected to 6814 card.

crazycat69, 2017-06-23:
f563cede33499e3055ab9898c484e067bd087e0a

closed 2017-06-23 by crazycat69