tbsdtv/linux_media #59: TBS5990+RPi 2+Libreelec v8.0.2

next2non, 2017-06-22:
Hi

I'm having an issue with TBS5990, tvheadend on Libreelec v8.0.2 detects it fine and I can add mux's fine. DVB-S SD channels stream well and no issues even HD channels with modulation QPSK, but HD channels with modulation 8PSK are not watchable at all and won't even run or stream. I approached Libreelec forum for help but in the end was guided here. Please find file that I executed using the following shell "dmesg | paste" http://sprunge.us/OGfD.

Also attaching video of how 8PSK file look like when I try to play it from tvheadend recording (if not from recording it won't run (get this error from tvheadend "2017-06-20 18:04:16.390 TS: Bader/10730V/LBC HD: H264 @ #371 Continuity counter error (total 1)"). 

My satellite that I'm using are Arabsat Badr 26E and Nilesat 7W.

Regards

[Recording sample.zip](https://github.com/tbsdtv/linux_media/files/1095856/Recording.sample.zip)

crazycat69, 2017-06-23:
use USB Bulk transfer mode instead ISO
`options cx231xx transfer_mode=0`

closed 2017-06-23 by crazycat69

CvH, 2017-06-23:
@crazycat69 shouldn't the driver do it? The same problem should appear at other devices too with this card !?

crazycat69, 2017-06-23:
Problem on high-bitrate  muxes > 70mbit.
All another cx2310x based device for DVB-T/C and maximum bitrate 50-60mbit.

CvH, 2017-06-23:
@crazycat69 tx for clarification