tbsdtv/linux_media #61: Can't compile on ARM RPI3

sharky1337, 2017-06-26:
I'm not able to  compile linux media on my raspberrypi 3

I followed the wiki 

https://github.com/tbsdtv/linux_media/wiki

but it fails. 

`make -C /home/pi/media_build/v4l 
make[1]: Entering directory '/home/pi/media_build/v4l'
Updating/Creating .config
make[2]: Entering directory '/home/pi/media_build/linux'
make[2]: Entering directory '/home/pi/media_build/linux'
Syncing with dir ../../media
Sync'ing file drivers/media/media-devnode.c (orig = 16233eda657927f1f4783982ee1b9f82c97029da, copy = 71ab3050de3ddb40777ca69893e17ac820be041d, patched = )
Sync'ing file drivers/media/media-device.c (orig = 7ba6759470146fd4cc6e18fd383da6aa1b462492, copy = ad5cb46662735abaa8d5ce9030cf464138266e51, patched = )
Sync'ing file drivers/media/common/tveeprom.c (orig = 08334f9839409d397f91fb4b7811f6a8ff2d5150, copy = 835d4681018ec2ba904bb60d2b74642861afc52b, patched = )
Sync'ing file drivers/media/common/saa7146/saa7146_fops.c (orig = ba5ed6008814f17dc65d42e2f3edbfaaa1afabfc, copy = e3af61d1315eafe0dfe3e919a943dc6e03ef8211, patched = )
Sync'ing file drivers/media/common/saa7146/saa7146_video.c (orig = e6242ff6dbb76f17a2b176c31fe7c150eca98951, copy = f3473fc291237784108a3cfaad4089b3810afcf9, patched = )
Sync'ing file drivers/media/common/saa7146/saa7146_hlp.c (orig = ba8adb432846b816f288f7e571a47de8cc7d3650, copy = 16d1af965e91f287e0b36cbf2fc40d7972da7500, patched = )
Sync'ing file drivers/media/common/saa7146/saa7146_core.c (orig = bd4d09a883930166981c3f61d7e6fec9b8acbadf, copy = 053241a8ba2c0f362a9cc655c1a215e68389375a, patched = )
Sync'ing file drivers/media/common/saa7146/saa7146_i2c.c (orig = 02a0ad96ea0a209b559f95894080dce428114873, copy = 82f05a4ba775ed7402cff8066a2dc3b7000bac7c, patched = )
Sync'ing file drivers/media/common/siano/smscoreapi.h (orig = f299fd0b3d279e5ac8842e29ecec297a9f2f0743, copy = 626919b45880ea69097f5a53e4b58b87d50c21b0, patched = )
Sync'ing file drivers/media/usb/au0828/au0828.h (orig = 81f3f032057b7e7498e57c43fae272b408a3cc76, copy = 958c103717eef02743e9f2b6bef426372f30edfb, patched = )
Sync'ing file drivers/media/usb/go7007/go7007-usb.c (orig = 4124109add45a5696af6088bf74c3833bf1acab0, copy = 79dd02df3859726cd24e8c3707b79e74e0e08ff0, patched = )
Sync'ing file drivers/media/usb/gspca/kinect.c (orig = 23ef0d32487b99c290e3f7025728300f3b72b175, copy = cf7a9444db02140a9b519ba78ef3cc7213d375ce, patched = )
Sync'ing file drivers/media/usb/gspca/spca501.c (orig = 9ab5c387d146e00bf1fe2e77b317a0d7ca2ca4ae, copy = bf4052152ea02fdf03c8b00c379d5d9e2d516792, patched = )
Sync'ing file drivers/media/usb/gspca/stk1135.c (orig = 5a292e37762ee2bbb239af6331bedeeb9e5ec2de, copy = 33d1857b3567bdfbafb60212561c1f4064af559f, patched = )
Sync'ing file drivers/media/usb/gspca/vc032x.c (orig = 4ba029e90f595316a6615b6d252ebf2e78edd265, copy = 6726a052f4d1676c23c156e955b67e71b729d696, patched = )
Sync'ing file drivers/media/usb/gspca/vicam.c (orig = eec8a0b4d010b52a9a32eed78f6861ed04f44702, copy = afc613a41b033896d21e27adf8c8b425c2487e91, patched = )
Sync'ing file drivers/media/usb/gspca/dtcs033.c (orig = 549c738569b55686dd61f91c4d369ba56c120914, copy = f40b19f624a72fae354d8c05739181b0945a36fa, patched = )
Sync'ing file drivers/media/usb/gspca/conex.c (orig = dac6a04ad8e8fbafbd0975e7735379c565b1e715, copy = 811381f11fb1ff83323f96ce1e21c743a94df86d, patched = )
Sync'ing file drivers/media/usb/gspca/w996Xcf.c (orig = f5038a8a85f91f9467896dd989a4638798db64f6, copy = f668830c815ea09b0e7d7a29507a2c7aff83a919, patched = )
Sync'ing file drivers/media/usb/gspca/stk014.c (orig = 848326d02aaa1a80a56c65edf46122c50a6f4312, copy = 1676622c7e427ec0c5c1aa8b3d37e42d58d75b45, patched = )
Sync'ing file drivers/media/usb/gspca/ov534_9.c (orig = 951b31c11cdc4fb28330b735ece6c0951cd9a5e0, copy = 16ce2631e572f9fc3c6706fc210a30b17dfd4cd8, patched = )
Sync'ing file drivers/media/usb/gspca/cpia1.c (orig = 2f4bdd3ac16ce67de3e178f886a30e1d6c7e2aac, copy = 02080d652bbda6f597c4d039ea75b0192f539fb1, patched = )
Sync'ing file drivers/media/usb/gspca/sq930x.c (orig = 5fbab39edd524fee19ceb6b8fa3e89b84f064970, copy = 6205bae9608c7338028d04c8bda31567f896817e, patched = )
Sync'ing file drivers/media/usb/gspca/zc3xx.c (orig = cfa2b272842b22f16118eb2e4fb0c7ce889b4cf0, copy = 82875acdcc353c558fd90980f09e31a9588988df, patched = )
Sync'ing file drivers/media/usb/gspca/pac207.c (orig = 067b6d4512080fd8e9439b04f7208f887f981c0d, copy = 8194251890f4657fbe9efb1237ede4f3d0190064, patched = )
Sync'ing file drivers/media/usb/gspca/spca1528.c (orig = a16576e78e0e969df667ff2c7d7146b8b65efde1, copy = 1a983b7c228d99f7e1a9002c3dc2babfd663db89, patched = )
Sync'ing file drivers/media/usb/gspca/jeilinj.c (orig = b11407e250a717ddf89d013baf180be63f2fbb8a, copy = cce0e0c30b7479005612b1a1ed04401d42487aff, patched = )
Sync'ing file drivers/media/usb/gspca/topro.c (orig = 0139d4794be3db74b454bbd31d98bcb726556ba3, copy = fd98594ecec71f4f63aadbc4de30c0de1bfc1ede, patched = )
Sync'ing file drivers/media/usb/gspca/mr97310a.c (orig = d09d6888c7a7253fdd6175bed3b1d8d78ce7a7dc, copy = fd70e82885adf57b2845a260e7767b1a923a4b56, patched = )
Sync'ing file drivers/media/usb/gspca/ov534.c (orig = c1e2b5da1e82c13090b131eb14a1d733d3f50db1, copy = 80f196aa5e0a722dda7f2e2b891565e859394ade, patched = )
Sync'ing file drivers/media/usb/gspca/ov519.c (orig = 8f98f55dde34a99d6630d408589cdad602fe04ba, copy = ba0621fbb2766db4d19757b2946549853ef141cd, patched = )
Sync'ing file drivers/media/usb/gspca/sq905c.c (orig = 28c8ce413b05b5033971d55e3445a3ae90935202, copy = 2182199390b96366c4e65622fc02bee03b1f0d69, patched = )
Sync'ing file drivers/media/usb/gspca/xirlink_cit.c (orig = 64e03cd9b29954afd49a012b7ee65998cd38eb9f, copy = 237fbe9e31d6bfa49a2284b1aaeeba6660588b01, patched = )
Sync'ing file drivers/media/usb/gspca/finepix.c (orig = 55dcff5a9b7d90c739e7b0f2df693dc4f4644e4b, copy = c58d3951dc7e8a04795807b7cb397f002be6414f, patched = )
Sync'ing file drivers/media/usb/gspca/gspca.c (orig = 056f8b361ff8417682b5780e0daf5754c84fffbc, copy = 868eaf1edbe40167aa2c4cceb57172fb90c366fd, patched = )
Sync'ing file drivers/media/usb/gspca/spca561.c (orig = df503a3a3118fa9061330ec0b13ab2ae47b1d0e6, copy = d013d9d0af5c0de0d68804f669d9a051d75ea3b7, patched = )
Sync'ing file drivers/media/usb/gspca/pac7311.c (orig = 592c6b2b6b913942e5b6131332f70d0919b68658, copy = 958a5787e12c49cfab4c10a3d2ccef377073615d, patched = )
Sync'ing file drivers/media/usb/gspca/spca508.c (orig = 4168621d78e4790a775ca30baa90c44e1577f33f, copy = 748930f8b53f285f80694b90382f68e3e795d66f, patched = )
Sync'ing file drivers/media/usb/gspca/pac7302.c (orig = 4ce1b280049e824ce45724ac2c473e0854278627, copy = cb0f4fe2c9601205362fa6b27e55d2bf667d83b0, patched = )
Sync'ing file drivers/media/usb/gspca/sn9c20x.c (orig = b807f675848e30d96787018a2eeb804f727bf527, copy = 52012a567d0dfe9bb32475e76139df741b1f880f, patched = )
Sync'ing file drivers/media/usb/gspca/nw80x.c (orig = 51cbf124ac2eed47c96e2dbd4616fa9a93928faf, copy = fa97c747d9fb3908613f3908199cdb1dd7463e22, patched = )
Sync'ing file drivers/media/usb/gspca/etoms.c (orig = c7ed2c12b99dc2ca256ca4af25acce59b214ba1d, copy = 5b14ea25e87c7a835bd0622b093136bf10f9ea71, patched = )
Sync'ing file drivers/media/usb/gspca/sn9c2028.c (orig = 75bd497de5b32a4ddb6f3a996e3ff980d9522487, copy = a5735cf7cdf3e036ed42d8f0419fb6c4642ad5f1, patched = )
Sync'ing file drivers/media/usb/gspca/sq905.c (orig = d06b45708b147e8fe17447107de9334ff5f2cbbf, copy = 4137ec6be5dd52776159b8a99aea46f9aa145644, patched = )
Sync'ing file drivers/media/usb/gspca/benq.c (orig = ba9c454768740c385472b591686ee1fe26296d9b, copy = aa2149bbf4818a2ac11db054cbe87cac1d3ff769, patched = )
Sync'ing file drivers/media/usb/gspca/sonixj.c (orig = ecb16d1caab23222246ea4d7c9858ab7c6aa45cd, copy = 5442ab497ee4a7d5cf08710bfe8f5c06d1033cc9, patched = )
Sync'ing file drivers/media/usb/gspca/sunplus.c (orig = ee70442284a7ea18a4153c382082d5d694b2bdc1, copy = f0ff10ee0807ce7ac2c5c539523aafe3c33ad9d7, patched = )
Sync'ing file drivers/media/usb/gspca/se401.c (orig = 7e73ed0b63f521ef0a88a35bcd39870b546ae363, copy = e767837d5cf8033772b4cf4e301558293b059929, patched = )
Sync'ing file drivers/media/usb/gspca/t613.c (orig = 98c2ea27db44113e33d78e3cfa478ab8a3e96f26, copy = cc959a88e950ab35fe0247fb90c5b644d1ce5dea, patched = )
Sync'ing file drivers/media/usb/gspca/spca505.c (orig = fa9733d378127330a38ef1736c0da1269acff236, copy = 8f9d355a34a9b2449d1d859293958dbc4393e452, patched = )
[... cut ...]
bluetooth 326105 22 bnep,btbcm,hci_uart
brcmfmac 186403 0
brcmutil 5661 1 brcmfmac
cfg80211 428871 1 brcmfmac
rfkill 16037 3 cfg80211,bluetooth
snd_bcm2835 20447 0
snd_pcm 75762 1 snd_bcm2835
snd_timer 19288 1 snd_pcm
snd 51908 3 snd_bcm2835,snd_timer,snd_pcm
bcm2835_gpiomem 2976 0
bcm2835_wdt 3225 0
uio_pdrv_genirq 3164 0
uio 8000 1 uio_pdrv_genirq
ipv6 347620 53

pi@raspberrypi:~ $ dmesg | grep TBS
...

root@raspberrypi:#apt list --installed | grep header
raspberrypi-kernel-headers/stable,now 1.20170515-1 armhf [installed]

cwsutherland, 2017-06-27:
You have 4.9 kernel headers installed 4.4.50 kernel running, do an update reboot and check again. It's not going to work like this.

sharky1337, 2017-06-27:
Which kernel-headers should I use  ? Git or via apt-get ?

sharky1337, 2017-06-27:
Using the following kernel files.

‘/root/linux-3e22e20f80899847293ef835b083e36ecb866d2d.tar.gz

updated via rpi-source

  CC [M]  /home/pi/media_build/v4l/altera-lpt.o
  CC [M]  /home/pi/media_build/v4l/altera-jtag.o
  CC [M]  /home/pi/media_build/v4l/altera-comp.o
make[3]: *** No rule to make target '/home/pi/media_build/v4l/frame_vector.c', needed by '/home/pi/media_build/v4l/frame_vector.o'.  Stop.
make[3]: *** Waiting for unfinished jobs....
  CC [M]  /home/pi/media_build/v4l/altera.o
Makefile:1490: recipe for target '_module_/home/pi/media_build/v4l' failed
make[2]: *** [_module_/home/pi/media_build/v4l] Error 2
make[2]: Leaving directory '/root/linux-3e22e20f80899847293ef835b083e36ecb866d2d'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

cwsutherland, 2017-06-27:
It won't build properly by installing the kernel source either, not without much tweaking.

You need a clean install of raspbian minimal

builds ok with 4.9 as well, so let it upgrade the kernel. (sudo apt-get upgrade)

sudo apt-get install raspberrypi-kernel-headers
must reboot

then you should be ok

sharky1337, 2017-06-27:
reinstalling raspbian fixed the problem thank you ! @cwsutherland

closed 2017-06-27 by sharky1337