tbsdtv/linux_media #67: Build fails on 4.9.35 on Rpi 3

cwsutherland, 2017-07-20:
See below.

Patches for 4.9.35-v7+ already applied.
make[2]: Leaving directory '/home/pi/media_build/linux'
make -C /lib/modules/4.9.35-v7+/build SUBDIRS=/home/pi/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-4.9.35-v7+'
  CC [M]  /home/pi/media_build/v4l/msp3400-driver.o
  CC [M]  /home/pi/media_build/v4l/msp3400-kthreads.o
  CC [M]  /home/pi/media_build/v4l/et8ek8_mode.o
  CC [M]  /home/pi/media_build/v4l/et8ek8_driver.o
In file included from /home/pi/media_build/v4l/msp3400-driver.c:53:0:
/home/pi/media_build/v4l/../linux/include/media/v4l2-ioctl.h:15:32: fatal error: linux/sched/signal.h: No such file or directory
 #include <linux/sched/signal.h>
                                ^
compilation terminated.
  CC [M]  /home/pi/media_build/v4l/cx25840-core.o
  CC [M]  /home/pi/media_build/v4l/cx25840-audio.o
  CC [M]  /home/pi/media_build/v4l/cx25840-firmware.o
scripts/Makefile.build:293: recipe for target '/home/pi/media_build/v4l/msp3400-driver.o' failed
make[3]: *** [/home/pi/media_build/v4l/msp3400-driver.o] Error 1
make[3]: *** Waiting for unfinished jobs....
Makefile:1490: recipe for target '_module_/home/pi/media_build/v4l' failed
make[2]: *** [_module_/home/pi/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.9.35-v7+'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/pi/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

cwsutherland, 2017-08-26:
OK now with the changes, build works installs, however huge delay in starting, some channels work most do not, reporting poor signal quality, or no signal.  

[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.9.35-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1014 SMP Fri Jun 30 14:47:43 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3c800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 249856
[    0.000000] free_area_init_node: node 0, pgdat 80c6ee40, node_mem_map bbf62000
[    0.000000]   Normal zone: 2196 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 249856 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @bbf1d000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 247660
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 968780K/999424K available (7168K kernel code, 484K rwdata, 2012K rodata, 1024K init, 778K bss, 22452K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbd800000 - 0xff800000   (1056 MB)
    lowmem  : 0x80000000 - 0xbd000000   ( 976 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c79014   ( 485 kB)
       .bss : 0x80c7b000 - 0x80d3da64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000010] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000032] Switching to timer-based delay loop, resolution 52ns
[    0.000380] Console: colour dummy device 80x30
[    0.001347] console [tty1] enabled
[    0.001405] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001483] pid_max: default: 32768 minimum: 301
[    0.001928] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001980] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003290] Disabling cpuset control group subsystem
[    0.003570] CPU: Testing write buffer coherency: ok
[    0.003670] ftrace: allocating 22390 entries in 66 pages
[    0.059141] CPU0: update cpu_capacity 1024
[    0.059219] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.059314] Setting up static identity map for 0x100000 - 0x100034
[    0.062536] CPU1: update cpu_capacity 1024
[    0.062548] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.063625] CPU2: update cpu_capacity 1024
[    0.063635] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.064618] CPU3: update cpu_capacity 1024
[    0.064628] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.064743] Brought up 4 CPUs
[    0.064943] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.064976] CPU: All CPU(s) started in HYP mode.
[    0.065005] CPU: Virtualization extensions available.
[    0.066192] devtmpfs: initialized
[    0.079466] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.079929] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.080011] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.080819] pinctrl core: initialized pinctrl subsystem
[    0.082139] NET: Registered protocol family 16
[... 249 lines cut ...]
[  366.571949] [<80121354>] (do_exit) from [<80121b68>] (do_group_exit+0x4c/0xe4)
[  366.571966] [<80121b68>] (do_group_exit) from [<8012cc34>] (get_signal+0x370/0x6d4)
[  366.571987] [<8012cc34>] (get_signal) from [<8010b2cc>] (do_signal+0x74/0x3e4)
[  366.572009] [<8010b2cc>] (do_signal) from [<8010b824>] (do_work_pending+0xb8/0xd0)
[  366.572028] [<8010b824>] (do_work_pending) from [<801080e8>] (slow_work_pending+0xc/0x20)
[  489.451573] INFO: task v4l_id:218 blocked for more than 120 seconds.
[  489.451587]       Tainted: G           O    4.9.35-v7+ #1014
[... cut ...]
[  489.452332] [<8013ab9c>] (task_work_run) from [<80121354>] (do_exit+0x370/0xaec)
[  489.452349] [<80121354>] (do_exit) from [<80121b68>] (do_group_exit+0x4c/0xe4)
[  489.452365] [<80121b68>] (do_group_exit) from [<8012cc34>] (get_signal+0x370/0x6d4)
[  489.452387] [<8012cc34>] (get_signal) from [<8010b2cc>] (do_signal+0x74/0x3e4)
[  489.452408] [<8010b2cc>] (do_signal) from [<8010b824>] (do_work_pending+0xb8/0xd0)
[  489.452428] [<8010b824>] (do_work_pending) from [<801080e8>] (slow_work_pending+0xc/0x20)
[  559.767988] cx231xx 1-1.4:1.1: Unknown tuner type configuring SIF

cwsutherland, 2017-08-26:
After rolling back to 0519, I get both adapters working for about 5-6 minutes fine then.

[  909.079327] cx231xx 1-1.4:1.1: Unknown tuner type configuring SIF
[ 1012.899081] cx231xx 1-1.4:1.1: Unknown tuner type configuring SIF
[ 1073.820970] cx231xx 1-1.4:1.1: Unknown tuner type configuring SIF
[ 1176.254046] tbscxci: Adapter 0 CI slot disabled
[ 1231.694340] tbscxci: Adapter 0 CI slot disabled
[ 1356.691076] cx231xx 1-1.4:1.1: Unknown tuner type configuring SIF
[ 1382.895396] tbscxci: Adapter 0 CI slot disabled
[ 1390.653335] cx231xx 1-1.4:1.1: cx231xx_send_usb_command: failed with status --32
[ 1390.653358] i2c i2c-9: av201x: i2c wrm err(-32) @0x00 (len=5)
[ 1392.512951] cx231xx 1-1.4:1.1: cx231xx_send_usb_command: failed with status --32
[ 1392.512975] i2c i2c-9: av201x: i2c wrm err(-32) @0x00 (len=5)
[ 1394.373339] cx231xx 1-1.4:1.1: cx231xx_send_usb_command: failed with status --32
[ 1394.373362] i2c i2c-9: av201x: i2c wrm err(-32) @0x00 (len=5)
[ 1396.278802] cx231xx 1-1.4:1.1: cx231xx_send_usb_command: failed with status --32
[ 1396.278826] i2c i2c-9: av201x: i2c wrm err(-32) @0x00 (len=5)
[ 1398.133740] cx231xx 1-1.4:1.1: cx231xx_send_usb_command: failed with status --32
[ 1398.133763] i2c i2c-9: av201x: i2c wrm err(-32) @0x00 (len=5)
[ 1418.834961] i2c i2c-7: tas2101_send_diseqc_msg() timeout sending burst
[ 1425.625033] i2c i2c-7: tas2101_send_diseqc_msg() timeout sending burst
[ 1443.375709] tbscxci: Adapter 0 CI slot disabled

And both adapter freeze, requiring  a hard reboot. Works more or less ok if only Adapter 0 used exclusively.

marcinulan, 2017-09-09:
On current version, , I got 
'make -j4'
'make -C /home/pi/tbs/media_build/v4l
make[1]: Entering directory "/home/pi/tbs/media_build/v4l"
make -C firmware prep
creating symbolic links...
make[2]: Entering directory "/home/pi/tbs/media_build/v4l/firmware"
make[2]: Leaving directory "/home/pi/tbs/media_build/v4l/firmware"
make -C firmware
make[2]: Entering directory "/home/pi/tbs/media_build/v4l/firmware"
make[2]: Nothing to be done for "default".
make[2]: Leaving directory "/home/pi/tbs/media_build/v4l/firmware"
Kernel build directory is /lib/modules/4.9.41-v7+/build
make -C ../linux apply_patches
make[2]: Entering directory "/home/pi/tbs/media_build/linux"
Syncing with dir ../../media
Patches for 4.9.41-v7+ already applied.
make[2]: Leaving directory "/home/pi/tbs/media_build/linux"
make -C /lib/modules/4.9.41-v7+/build SUBDIRS=/home/pi/tbs/media_build/v4l  modules
make[2]: Entering directory "/home/pi/linux-4153f509b449f1c1c816cf124c314975c3daa824"
Makefile:628: arch/arm/Makefile: No such file or directory
make[2]: *** No rule to make target "arch/arm/Makefile".  Stop.
make[2]: Leaving directory "/home/pi/linux-4153f509b449f1c1c816cf124c314975c3daa824"
Makefile:51: recipe for target "default" failed
make[1]: *** [default] Error 2
make[1]: Leaving directory "/home/pi/tbs/media_build/v4l"
Makefile:26: recipe for target "all" failed
make: *** [all] Error 2'

cwsutherland, 2017-09-10:
Looks like you are missing raspberrypi-kernel-headers, but in any case these drivers are unreliable on the Raspberry PI II/III.

The only one which works semi-reliable is [https://www.tbsiptv.com/download/common/tbs-linux-drivers_v170330.zip](url)

However, using two adapters simultaneously causes a freeze, so you should disable the second adapter otherwise you will need to do a hard reset to get it working again.