tbsdtv/linux_media #68: TBS 5520SE si2183 failed

cyberkrel, 2017-07-20:
Уважаемый CrazyCat я столкнулся с одной проблемой при запуске tvheadend периодически возникает ошибка начинает работать только после многократного перезапуска tvheadend.
Причём тюнер работает великолепно DVB-T2 DVB-S2 если tvheadend запустился проблем не возникает до перезагрузки.

```
[   13.082825] media: Linux media interface: v0.10
[   13.084925] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[   13.085251] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[   13.085254] tbs5520se: start downloading TBS5520se firmware
[   13.191369] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[   13.191504] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   13.191564] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)
[   13.270077] dvb-usb: MAC address: 00:23:ab:c0:55:d2
[   13.272171] i2c i2c-8: Added multiplexed i2c bus 9
[   13.272173] si2183 8-0067: Silicon Labs Si2183 successfully attached
[   13.275818] si2157 9-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   13.276587] i2c i2c-9: av201x: Airoha Technology AV201x successfully attached
[   13.341039] usb 3-3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   13.341103] usb 3-3: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   13.341253] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[   13.341273] usbcore: registered new interface driver tbs5520se
[   13.373489] si2183 8-0067: found a 'Silicon Labs Si2183-B60'
[   13.373811] si2183 8-0067: downloading firmware from file 'dvb-demod-si2183-b60-01.fw'
[   13.606938] si2183 8-0067: firmware version: 6.0.2
[   13.620393] si2157 9-0061: found a 'Silicon Labs Si2157-A30'
[   13.643478] si2157 9-0061: firmware version: 3.0.5
[   13.643491] usb 3-3: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[   14.359508] usb 3-3: DVB: adapter 0 frontend 1 frequency 0 out of range (950000..2150000)
[   17.159729] si2183 8-0067: read_status failed=-11
[   17.359742] si2183 8-0067: read_status failed=-11
[   17.559765] si2183 8-0067: read_status failed=-11
[   17.759771] si2183 8-0067: read_status failed=-11
[   17.959787] si2183 8-0067: read_status failed=-11
[   18.059816] si2183 8-0067: read_status failed=-11
[   18.159824] si2183 8-0067: read_status failed=-11
[   18.259824] si2183 8-0067: read_status failed=-11
[   18.359832] si2183 8-0067: read_status failed=-11
[   18.459842] si2183 8-0067: read_status failed=-11
```
```
[    7.919311] si2183 8-0067: read_status failed=-11
[    8.436733] si2183 8-0067: set_tone failed=-110
[    8.436753] si2183 8-0067: read_status failed=-11
[    8.961073] si2183 8-0067: err set agc mode
[    9.464785] si2183 8-0067: err set dvb-t mode
[    9.968793] si2183 8-0067: err restarting dsp
[   45.132746] si2183 8-0067: set_tone failed=-110
[   45.644697] si2183 8-0067: err set agc mode
[   46.148604] si2183 8-0067: err set dvb-t mode
[   46.652441] si2183 8-0067: err restarting dsp
[   81.763740] si2183 8-0067: set_tone failed=-110
[   82.275901] si2183 8-0067: err set agc mode
[   82.779900] si2183 8-0067: err set dvb-t mode
[   83.283949] si2183 8-0067: err restarting dsp
[  118.403254] si2183 8-0067: set_tone failed=-110
[  118.915274] si2183 8-0067: err set agc mode
[  119.419482] si2183 8-0067: err set dvb-t mode
[  119.923331] si2183 8-0067: err restarting dsp
[  155.042780] si2183 8-0067: set_tone failed=-110
[  155.554910] si2183 8-0067: err set agc mode
```

mxl9236, 2017-07-24:
same here,waiting for help.

cyberkrel, 2017-07-29:
@mxl9236 Try it.

```
diff --git a/drivers/media/usb/dvb-usb/dvb-usb-dvb.c b/drivers/media/usb/dvb-usb/dvb-usb-dvb.c
index 5f265e6..c70406c 100644
--- a/drivers/media/usb/dvb-usb/dvb-usb-dvb.c
+++ b/drivers/media/usb/dvb-usb/dvb-usb-dvb.c
@@ -356,6 +356,8 @@ int dvb_usb_adapter_frontend_exit(struct dvb_usb_adapter *adap)
                if (adap->fe_adap[i].fe != NULL) {
                        dvb_unregister_frontend(adap->fe_adap[i].fe);
                        dvb_frontend_detach(adap->fe_adap[i].fe);
+                        dvb_unregister_frontend(adap->fe_adap[i].fe2);
+                        dvb_frontend_detach(adap->fe_adap[i].fe2);
                }
        }
        adap->num_frontends_initialized = 0;
diff --git a/drivers/media/usb/dvb-usb/tbs5520se.c b/drivers/media/usb/dvb-usb/tbs5520se.c
index d3b5ffa..d5bb605 100644
--- a/drivers/media/usb/dvb-usb/tbs5520se.c
+++ b/drivers/media/usb/dvb-usb/tbs5520se.c
@@ -88,7 +88,7 @@ static int tbs5520se_i2c_transfer(struct i2c_adapter *adap,

                tbs5520se_op_rw(d->udev, 0x90, 0, 0,
                                        buf6, 3, TBS5520se_WRITE_MSG);
-               //msleep(5);
+               msleep(5);
                tbs5520se_op_rw(d->udev, 0x91, 0, 0,
                                        inbuf, buf6[0], TBS5520se_READ_MSG);
                memcpy(msg[1].buf, inbuf, msg[1].len);
@@ -117,7 +117,7 @@ static int tbs5520se_i2c_transfer(struct i2c_adapter *adap,
                                        inbuf, buf6[0], TBS5520se_READ_MSG);
                                memcpy(msg[0].buf, inbuf, msg[0].len);
                        }
-                       //msleep(3);
+                       msleep(3);
                break;
                case (TBS5520se_VOLTAGE_CTRL):
                        buf6[0] = 3;
@@ -130,9 +130,9 @@ static int tbs5520se_i2c_transfer(struct i2c_adapter *adap,
                                        buf6, 4, TBS5520se_READ_MSG);
                        msg[0].buf[0] = buf6[2];
                        msg[0].buf[1] = buf6[3];
-                       //msleep(3);
-                       //info("TBS5520se_RC_QUERY %x %x %x %x\n",
-                       //              buf6[0],buf6[1],buf6[2],buf6[3]);
+                       msleep(3);
+                       info("TBS5520se_RC_QUERY %x %x %x %x\n",
+                                       buf6[0],buf6[1],buf6[2],buf6[3]);
                        break;
                }
```

TomBZombie, 2018-04-19:
I'm having the same issue albeit a slightly different error:
```
16.224118] si2157: probe of 6-0061 failed with error -110
[   16.224251] dvb-usb: no frontend was attached by 'TBS 5520se USB2.0'
[   16.224332] dvb-usb: TBS 5520se USB2.0 successfully initialized and connected.

```

Unfortunately the patch also gives me an error:

```
File to patch: drivers/media/usb/dvb-usb/tbs5520se.c
patching file drivers/media/usb/dvb-usb/tbs5520se.c
Hunk #1 FAILED at 88.
Hunk #2 FAILED at 117.
Hunk #3 FAILED at 130.
```

Arch linux, 4.16 kernel