tbsdtv/linux_media #72: TBS 5881 - no frontend was attached

tschreibe, 2017-08-31:
Hi,
I compiled the drivers successfully but when I plug the USB I get the following:

[ 6947.892078] usb 1-1: new high-speed USB device number 12 using xhci_hcd
[ 6948.063633] dvb-usb: found a 'TBS 5881 CI USB2.0' in cold state, will try to load a firmware
[ 6948.063639] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5881.fw'
[ 6948.063641] tbs5881: start downloading TBS5881 CI firmware
[ 6948.182049] dvb-usb: found a 'TBS 5881 CI USB2.0' in warm state.
[ 6948.182161] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[ 6948.182309] DVB: registering new adapter (TBS 5881 CI USB2.0)
[ 6948.400109] dvb-usb: MAC address: 00:22:ab:b1:02:87
[ 6948.400840] dvb-usb: no frontend was attached by 'TBS 5881 CI USB2.0'
[ 6948.401093] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-1/input/input23
[ 6948.401200] dvb-usb: schedule remote query interval to 150 msecs.
[ 6948.401204] dvb-usb: TBS 5881 CI USB2.0 successfully initialized and connected.

I'm on Arch Linux with lts kernel 4.9.45, I used the aur package (https://aur.archlinux.org/packages/tbs-linux_media-git) which compiles this repo.
(I also got the same error with 4.12.8 kernel)

I don't know why but this works perfectly with LibreELEC, here is the dmesg output for LibreELEC:

[  499.889330] usb 1-3: new high-speed USB device number 6 using xhci_hcd
[  500.020892] dvb-usb: found a 'TBS 5881 CI USB2.0' in cold state, will try to load a firmware
[  500.020917] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5881.fw'
[  500.020930] tbs5881: start downloading TBS5881 CI firmware
[  500.132713] dvb-usb: found a 'TBS 5881 CI USB2.0' in warm state.
[  500.133202] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  500.133359] dvbdev: DVB: registering new adapter (TBS 5881 CI USB2.0)
[  500.225063] dvb-usb: MAC address: 00:22:ab:b1:02:87
[  500.236374] i2c i2c-8: Added multiplexed i2c bus 9
[  500.236378] si2168 8-0064: Silicon Labs Si2168-B40 successfully identified
[  500.236381] si2168 8-0064: firmware version: B 4.0.2
[  500.238859] si2157 9-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[  500.259366] tbs5881: CI initialized.
[  500.259560] tbs5881: CI disabled.
[  500.259570] usb 1-3: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[  500.259941] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb1/1-3/input/input15
[  500.260232] dvb-usb: schedule remote query interval to 150 msecs.
[  500.260237] dvb-usb: TBS 5881 CI USB2.0 successfully initialized and connected.
[  500.333087] si2168 8-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  500.689778] si2168 8-0064: firmware version: B 4.0.11
[  500.694325] usb 1-3: DVB: adapter 0 frontend 0 frequency 0 out of range (42000000..870000000)

Looks like "Silicon Labs Si2168" is found on LibreELEC but not on Arch... I even tried "modprobe si2168" but it did not help. Yet the si2168 firmware is in /lib/firmware...

lsmod | grep dvb

dvb_usb_tbs5881        24576  0
dvb_usb                28672  1 dvb_usb_tbs5881
dvb_core               94208  2 dvb_usb_tbs5881,dvb_usb
rc_core                24576  8 ir_rc6_decoder,ir_nec_decoder,ir_mce_kbd_decoder,ir_sony_decoder,ir_lirc_codec,dvb_usb,ir_jvc_decoder,ir_rc5_decoder
usbcore               208896  6 usbhid,dvb_usb_tbs5881,xhci_pci,dvb_usb,btusb,xhci_hcd

lsusb

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 005: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 004: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 0cf3:e300 Qualcomm Atheros Communications 
Bus 001 Device 012: ID 734c:5881 TBS Technologies China 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

ls /dev/dvb/adapter0 -l

total 0
crw-rw----+ 1 root video 212, 0 31 août  18:25 demux0
crw-rw----+ 1 root video 212, 1 31 août  18:25 dvr0
crw-rw----+ 1 root video 212, 2 31 août  18:25 net0

Probably same issue than https://github.com/tbsdtv/linux_media/issues/47

Any help would be appreciated.
Thanks.

crazycat69, 2017-08-31:
So ask Arch Linux package maintainer :)
si2168 module present ? `modinfo si2168`

tschreibe, 2017-08-31:
My bad... You're right, the arch build script got confused by switching kernels and modules ended up in the wrong extramodules directory.. And si2168 was given by the kernel and not the one provided here.
So now it's all good :) Thanks.

closed 2017-08-31 by tschreibe

AlexanderS, 2017-09-01:
@tschreibe There is also a dkms version available: https://aur.archlinux.org/packages/tbs-linux_media-git-dkms/ (gets rebuild automatically on kernel updates).