tbsdtv/linux_media #76: ADD THIS DRIVER IN KERNEL FOR FEDORA 26 or high....?

sajjads24, 2017-09-15:
hi
please add this driver to kernel 4.13 or make repo rpm package for fedora 26 , centos7, arch, opensuse ,.... 
For Fedora, a Coper repo is build
So users can update the driver easily after updating to new kernels .
please build repo for rpm distro ?

thanks

sajjads24, 2017-09-15:
please add driver tbs6983 to kernel 4.13 or high ....?
or make rpm repo for driver for fedora , centos , arch , opensuse ....

thanks

Hoeze, 2018-01-14:
This will never happen (most likely)...

This repository is not just a bunch of additional drivers, it's the whole driver part of the linux kernel patched with TBS-specific changes.
This means that we won't get official support for any Linux distro as long as TBS does not upstream its changes to the official kernel.

See also https://github.com/tbsdtv/linux_media/issues/87

Davin622, 2018-01-15:
Hi
this driver already  support  to 4.13. 
@Hoeze  I understand your mean, we just make the media directory, the other part we do not care. in fact, we alway got the current version from the offical web. if you want update other part, you can make the patch. or you can refer to"Upgrading sources and re-installing"from wiki(https://github.com/tbsdtv/linux_media/wiki).

Best Regards

Hoeze, 2018-01-15:
Oh, sorry, i see:
I was talking about distro support, not kernel 4.13 support. I clarified my comment.