tbsdtv/linux_media #78: Error  Upgrading sources and re-installing after update kernel 4.12.12 ?!

sajjads24, 2017-09-16:
hi 
how fix this error for upgrade driver when use make command on kernel 4.12.12 ?
please see this error:
`[sajjad@localhost ~]$ cd media
[sajjad@localhost media]$ git remote update
error: cannot open .git/FETCH_HEAD: Permission denied
[sajjad@localhost media]$ su
Password:
[root@localhost media]# git remote update
Fetching origin
[root@localhost media]# cd ../media_build
[root@localhost media_build]# git remote update
Fetching origin
[root@localhost media_build]# git pull
Already up-to-date.
[root@localhost media_build]# make
make -C /home/sajjad/media_build/v4l
make[1]: Entering directory '/home/sajjad/media_build/v4l'
creating symbolic links...
make -C firmware prep
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
Kernel build directory is /lib/modules/4.12.11-300.fc26.x86_64/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/sajjad/media_build/linux'
Syncing with dir ../../media
Patches for 4.12.11-300.fc26.x86_64 already applied.
make[2]: Leaving directory '/home/sajjad/media_build/linux'
make -C /lib/modules/4.12.11-300.fc26.x86_64/build SUBDIRS=/home/sajjad/media_build/v4l  modules
make[2]: Entering directory '/home/sajjad/media_build/v4l'
make[2]: *** /lib/modules/4.12.11-300.fc26.x86_64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/sajjad/media_build/v4l'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/sajjad/media_build/v4l'
make: *** [Makefile:26: all] Error 2
[root@localhost media_build]#`

sajjads24, 2017-09-18:
how fix error ?
`[sajjad@localhost ~]$ cd media
[sajjad@localhost media]$ git remote update
error: cannot open .git/FETCH_HEAD: Permission denied
[sajjad@localhost media]$ su
Password: 
[root@localhost media]# git remote update
Fetching origin
[root@localhost media]# git pull
Already up-to-date.
[root@localhost media]# cd ../media_build
[root@localhost media_build]# git remote update
Fetching origin
[root@localhost media_build]# git pull
Already up-to-date.
[root@localhost media_build]# make
make -C /home/sajjad/media_build/v4l 
make[1]: Entering directory '/home/sajjad/media_build/v4l'
creating symbolic links...
make -C firmware prep
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
Kernel build directory is /lib/modules/4.12.11-300.fc26.x86_64/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/sajjad/media_build/linux'
Syncing with dir ../../media
Patches for 4.12.11-300.fc26.x86_64 already applied.
make[2]: Leaving directory '/home/sajjad/media_build/linux'
make -C /lib/modules/4.12.11-300.fc26.x86_64/build SUBDIRS=/home/sajjad/media_build/v4l  modules
make[2]: Entering directory '/home/sajjad/media_build/v4l'
make[2]: *** /lib/modules/4.12.11-300.fc26.x86_64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/sajjad/media_build/v4l'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/sajjad/media_build/v4l'
make: *** [Makefile:26: all] Error 2
[root@localhost media_build]# rpm -qa | grep -i kernel
kernel-tools-libs-4.12.12-300.fc26.x86_64
kernel-rpm-macros-63-1.fc26.noarch
kernel-4.12.12-300.fc26.x86_64
kernel-cross-headers-4.12.12-300.fc26.x86_64
kernel-core-4.12.9-300.fc26.x86_64
kernel-devel-4.12.12-300.fc26.x86_64
kernelshark-2.6.1-1.fc26.x86_64
kernel-tools-4.12.12-300.fc26.x86_64
kernel-core-4.12.12-300.fc26.x86_64
kernel-headers-4.12.12-300.fc26.x86_64
kernel-modules-extra-4.12.12-300.fc26.x86_64
kernel-modules-extra-4.12.9-300.fc26.x86_64
kernel-modules-4.12.9-300.fc26.x86_64
libreport-plugin-kerneloops-2.9.1-2.fc26.x86_64
kernel-4.12.9-300.fc26.x86_64
abrt-addon-kerneloops-2.10.3-1.fc26.x86_64
kernel-tools-libs-devel-4.12.12-300.fc26.x86_64
kernel-modules-4.12.12-300.fc26.x86_64
erlang-kernel-19.3.6.2-1.fc26.x86_64
[root@localhost media_build]# `

crazycat69, 2017-09-18:
Look like kernel header not installed, but anyway current sources not merged with lastest 4.13,x media_tree.

P.S. Check wiki info about CentOS/RedHat compatibility/installation - https://github.com/tbsdtv/linux_media/wiki

sajjads24, 2017-09-21:
can not run kaffeine after update driver for kernel 4.12.12 !?
https://pastebin.com/09bKytA8
how fix kaffeine and update driver for new kernel (4.12.12)?

crazycat69, 2017-09-21:
First say what DVB device you have ? Next - see kernel log with dmesg cmd.

P.S. Kafeine is 'dumb' software and not related to this repository.

sajjads24, 2017-09-22:
That's right, since I did not run the driver after that, so I wrote here.
So the values and logs that you said:
https://paste.fedoraproject.org/paste/o9ScSPg-5D-vZAYsc~bFXw
kernel list :
https://pastebin.com/yAkHZH6n
My card is tbs6983.
Thanks for the help

crazycat69, 2017-09-22:
look like dvb drivers not installed.
you must re-build this drivers for new kernel :) don't ask about some dynamic-compilable tbs driver rpm packet :)

sajjads24, 2017-09-22:
i am rebuild and re install for two kernel old and new but not fix !?:
`[sajjad@localhost ~]$ kaffeine
22-09-17 20:21:49.384 [Critical] "Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
[sajjad@localhost ~]$ dmesg | tail
[  103.633143] RAX: ffffffffffffffda RBX: 00007f95946c0258 RCX: 00007f958fabf5e7
[  103.633178] RDX: 0000000000000002 RSI: 0000000000006f43 RDI: 000000000000001c
[  103.633212] RBP: 00007ffd61483be0 R08: 0000000000000000 R09: 00007ffd61483e18
[  103.633247] R10: 00007ffd614842f0 R11: 0000000000000246 R12: 00007ffd61483d98
[  103.635327] R13: 00007f95946c0258 R14: 00007ffd61483bf0 R15: 00000000fbad8001
[  103.637402] Code: a5 bf f5 eb cb e8 a7 d2 5f f5 0f 1f 80 00 00 00 00 66 66 66 66 90 55 48 89 e5 41 56 41 55 41 54 53 41 89 f6 48 8b 87 30 03 00 00 <48> 8b 40 38 80 b8 09 05 00 00 01 4c 8b a8 00 05 00 00 4c 89 ef 
[  103.639680] RIP: saa716x_tbs6983_set_voltage+0x1a/0xe0 [saa716x_tbs_dvb] RSP: ffffb175c8cbfaf8
[  103.642008] CR2: 0000000000000038
[  103.656548] ---[ end trace dc7e560c05488936 ]---
[  172.325073] tun: Universal TUN/TAP device driver, 1.6
[sajjad@localhost ~]$ dmesg -k | fpaste
Uploading (80.3KiB)...
https://paste.fedoraproject.org/paste/YBq~CUJmuhIE9G0NwVHIgA
[sajjad@localhost ~]$`

sajjads24, 2017-09-22:
\nRemoving obsolete files from /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/media/video:

\nRemoving obsolete files from /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/media/video/gspca:

\nRemoving obsolete files from /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/media/video/saa7164:

\nRemoving obsolete files from /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/media/video/gspca/stv06xx:

Installing kernel modules under /lib/modules/4.12.13-300.fc26.x86_64/kernel/drivers/media/:
/sbin/depmod -a 4.12.13-300.fc26.x86_64 
make -C firmware install
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
Installing firmwares at /lib/firmware: vicam/firmware.fw ttusb-budget/dspbootcode.bin cpia2/stv0672_vp4.bin av7110/bootcode.bin 
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
make[1]: Leaving directory '/home/sajjad/media_build/v4l'
[root@localhost media_build]# dmesg | fpaste
Uploading (77.3KiB)...
https://paste.fedoraproject.org/paste/G~2nP-4PW1MEElbgJ3qabQ
[root@localhost media_build]#

AlexanderS, 2017-09-25:
@sajjads24 You have to reboot (or manually reloading the modules) after the installation of the driver.

sajjads24, 2017-09-26:
i am install closesource driver on kernel 4.12.13 so  after install opensource driver from github wiki but not use opensource firmware :
Firmware
wget http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2
tar jxvf tbs-tuner-firmwares_v1.0.tar.bz2 -C /lib/firmware/
so after reboot fixed kaffeine and installed driver on kernel 4.12.13 now working fine for me.

```
[sajjad@localhost ~]$ uname -r
4.12.13-300.fc26.x86_64
[sajjad@localhost ~]$ ls /dev/dvb
adapter0  adapter1  adapter2
[sajjad@localhost ~]$ lspci | grep Multimedia
08:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
[sajjad@localhost ~]$ dmesg | grep -i dvb
[   13.255541] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[   15.065864] SAA716x Budget 0000:08:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6983 DVB-S/S2)...
[   15.065966] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[   15.066472] SAA716x Budget 0000:08:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6983 DVB-S/S2)...
[   16.239201] dvbdev: DVB: registering new adapter (Siano Nice Digital Receiver)
[   16.239231] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[   16.239406] usb 2-1.7: DVB: registering adapter 2 frontend 0 (Siano Mobile Digital MDTV Receiver)...
[   16.239408] dvbdev: dvb_create_media_entity: media entity 'Siano Mobile Digital MDTV Receiver' registered.
[   16.239468] smsdvb:smsdvb_hotplug: DVB interface registered.
[sajjad@localhost ~]$
```