tbsdtv/linux_media #79: TBS-6285: Fedora 26 - Kernel 4.12.13 Cannot Read from Transport

petem67, 2017-09-23:
Hi,
I'm in the process of upgrading my system from Fedora 21 to Fedora 26 and am having trouble with the TBS drivers with the latest linux kernels.

The issue is that whilst I can tune to a frequency it is not possible to read from the transports.

I have successfully installed the the latest drivers from the media repository (commit 2cacea5c44b45c46712837dd2afdbdde18447e0c) and dmesg confirms they are loaded:

[   16.837572] media: Linux media interface: v0.10
[   16.856979] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	2cacea5c44b45c46712837dd2afdbdde18447e0c Merge branch 'tbsdtv_linux_media/master' into tbsdtv_linux_media/latest
[   16.925840] SAA716x Budget 0000:09:00.0: enabling device (0000 -> 0002)
[   16.935307] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[   16.981315] i2c i2c-9: Added multiplexed i2c bus 10
[   16.981316] si2168 9-0064: Silicon Labs Si2168-B40 successfully identified
[   16.981317] si2168 9-0064: firmware version: B 4.0.2
[   16.995707] RAPL PMU: API unit is 2^-32 Joules, 4 fixed counters, 655360 ms ovfl timer
[   16.995708] RAPL PMU: hw unit of domain pp0-core 2^-14 Joules
[   16.995708] RAPL PMU: hw unit of domain package 2^-14 Joules
[   16.995708] RAPL PMU: hw unit of domain dram 2^-14 Joules
[   16.995709] RAPL PMU: hw unit of domain pp1-gpu 2^-14 Joules
[   17.033484] si2157 10-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   17.033802] SAA716x Budget 0000:09:00.0: TurboSight TBS 6285  MAC[0]=ff:ff:ff:ff:ff:ff
[   17.033804] SAA716x Budget 0000:09:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   17.033869] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[   17.036809] i2c i2c-9: Added multiplexed i2c bus 11
[   17.036810] si2168 9-0066: Silicon Labs Si2168-B40 successfully identified
[   17.036811] si2168 9-0066: firmware version: B 4.0.2
[   17.038448] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   17.038767] SAA716x Budget 0000:09:00.0: TurboSight TBS 6285  MAC[1]=ff:ff:ff:ff:ff:ff
[   17.038769] SAA716x Budget 0000:09:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   17.038849] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[   17.041768] i2c i2c-8: Added multiplexed i2c bus 12
[   17.041770] si2168 8-0064: Silicon Labs Si2168-B40 successfully identified
[   17.041771] si2168 8-0064: firmware version: B 4.0.2
[   17.043386] si2157 12-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   17.043716] SAA716x Budget 0000:09:00.0: TurboSight TBS 6285  MAC[2]=ff:ff:ff:ff:ff:ff
[   17.043717] SAA716x Budget 0000:09:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   17.043786] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[   17.046675] i2c i2c-8: Added multiplexed i2c bus 13
[   17.046677] si2168 8-0066: Silicon Labs Si2168-B40 successfully identified
[   17.046678] si2168 8-0066: firmware version: B 4.0.2
[   17.048287] si2157 13-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   17.048619] SAA716x Budget 0000:09:00.0: TurboSight TBS 6285  MAC[3]=ff:ff:ff:ff:ff:ff
[   17.048620] SAA716x Budget 0000:09:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...

However when I try to make use of the tuner I hit issues:

dvbv5-scan returns the following (for the Crystal palace transmitter):
[root@europaTest media]# dvbv5-scan ~ppfm/initial.conf 
Scanning frequency #1 490000000
Lock   (0x1f) C/N= 21.25dB UCB= 0 postBER= 325x10^-3
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #2 514000000
Lock   (0x1f) Signal= -26.00dBm C/N= 28.75dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #3 570000000
Lock   (0x1f) Signal= -35.00dBm C/N= 30.25dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #4 506000000
Lock   (0x1f) Signal= -35.00dBm C/N= 28.00dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #5 482000000
Lock   (0x1f) Signal= -23.00dBm C/N= 22.50dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #6 529833000
Lock   (0x1f) Signal= -23.00dBm C/N= 22.25dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
Scanning frequency #7 545833000
Lock   (0x1f) Signal= -23.00dBm C/N= 32.25dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table

I get similar behaviour from w_scan

w_scan -X -vv -ft -c gb 
w_scan version 20161022 (compiled for DVB API 5.10)
using settings for UNITED KINGDOM
DVB aerial
DVB-T GB
scan type TERRESTRIAL, channellist 6
output format czap/tzap/szap/xine
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
   check TurboSight TBS 6285 DVB-T/T2/C:
           DVB-C ann.B
           DVB-C ann.A
           DVB-T2
           DVB-T
	/dev/dvb/adapter0/frontend0 -> TERRESTRIAL "TurboSight TBS 6285 DVB-T/T2/C": very good :-))

Using TERRESTRIAL frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_ 
Using DVB API 5.10
frontend 'TurboSight TBS 6285 DVB-T/T2/C' supports
DVB-T2
INVERSION_AUTO
QAM_AUTO
TRANSMISSION_MODE_AUTO
GUARD_INTERVAL_AUTO
HIERARCHY_AUTO
FEC_AUTO
BANDWIDTH_AUTO not supported, trying 6/7/8 MHz.
FREQ (42.00MHz ... 870.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_ 
   check TurboSight TBS 6285 DVB-T/T2/C:
           DVB-C ann.B
           DVB-C ann.A
           DVB-T2
           DVB-T
Scanning DVB-T...
Scanning 8MHz frequencies...
474000: (time: 00:00.838) 
474167: (time: 00:02.842) 
473833: (time: 00:04.890) 
482000: (time: 00:06.936) 
        (0.256sec): SCL (0x1F)
        (0.256sec) signal
        (0.256sec) lock
        signal ok:	QAM_AUTO f = 482000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        **initial PAT lookup..
        Info: no data from PAT after 2 seconds**
        deleting (QAM_AUTO f = 482000 kHz I999B8C999D999T999G999Y999 (0:0:0))
482167: (time: 00:09.707) 
        (0.254sec): SC (0x3)
        (0.254sec) signal
        (0.306sec): SCL (0x1F)
        (0.306sec) lock
        signal ok:	QAM_AUTO f = 482167 kHz I999B8C999D999T999G999Y999 (0:0:0)
        **initial PAT lookup..
        Info: no data from PAT after 2 seconds**
        deleting (QAM_AUTO f = 482167 kHz I999B8C999D999T999G999Y999 (0:0:0))
etc...

This is telling me that the card is successfully tuning to the appropriate frequency but nothing is actually being read from the mux!

From dmesg when the tuner is accessed I see the following:
[  160.206766] si2168 9-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  160.620852] si2168 9-0064: firmware version: B 4.0.25
[  160.625769] si2157 10-0062: found a 'Silicon Labs Si2158-A20'
[  160.652430] si2157 10-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[  160.919183] si2157 10-0062: firmware version: 2.1.9
[  160.921169] si2168 9-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'

So the firmware does get loaded for the card.

As a last resort I tried to access the tuners via mythtv. This returns the following when I try to scan:
2017-09-23 01:36:08.737867 E  DVBChan[1](/dev/dvb/adapter0/frontend0): Getting Frontend status failed.
			eno: Remote I/O error (121)
2017-09-23 01:36:08.737875 E  DVBSigMon[-1](/dev/dvb/adapter0/frontend0): Cannot read DVB status
			eno: Remote I/O error (121)
2017-09-23 01:36:08.772138 E  DVBChan: wait_for_backend: Failed to get status
			eno: Remote I/O error (121)

chrisfudge, 2017-10-03:
Hi
I may have had a similar problem with Opensuse Leap42.3. I altered the permissions on "/dvb/......." so all users could RWX not just user:root group:root. mythtv and kaffeine were then able to access the adaptors from a user login.

closed 2017-10-06 by crazycat69

petem67, 2017-10-06:
Thanks Chris,
sadly it is not a permissions issue. I just checked that and in any case when I was testing with w_scan and dvbv5-scan I was using root.

I have just tried the latest 4.13.4-200 and this show the same issue. Basically the TBS6285 drivers are not working with the latest kernels.

I'm not sure why this ticket has been closed as the issue still exists.

Incidentally as a short term solution I have patched the old official TBS drivers such that they will build and install. This is not ideal but at least I am up and running until this is resolved!

petem67, 2017-10-06:
Another strange behaviour seen with the TBS6285:
using adapter 1 with dvbv5-scan causes the tool to crash. This only happens with adapter 1:

dvbv5-scan -a 1 initial_single_mux.conf 
Scanning frequency #1 490000000
Lock   (0x1f) C/N= 20.75dB UCB= 0 postBER= 1.00
ERROR    dvb_read_sections: no data read on section filter
ERROR    error while waiting for PAT table
*** Error in `dvbv5-scan': double free or corruption (fasttop): 0x0000562ede489180 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x7c8dc)[0x7f69f48ee8dc]
/lib64/libc.so.6(+0x87789)[0x7f69f48f9789]
/lib64/libc.so.6(cfree+0x16e)[0x7f69f48ff0ee]
/lib64/libdvbv5.so.0(free_dvb_dev+0x12)[0x7f69f4c5ac42]
/lib64/libdvbv5.so.0(dvb_dev_free_devices+0x2b)[0x7f69f4c5acdb]
/lib64/libdvbv5.so.0(dvb_dev_free+0x4b)[0x7f69f4c5afeb]
dvbv5-scan(+0x1f62)[0x562ede004f62]
/lib64/libc.so.6(__libc_start_main+0xea)[0x7f69f489250a]
dvbv5-scan(+0x221a)[0x562ede00521a]
======= Memory map: ========
562ede003000-562ede007000 r-xp 00000000 fd:00 2500831                    /usr/bin/dvbv5-scan
562ede207000-562ede208000 r--p 00004000 fd:00 2500831                    /usr/bin/dvbv5-scan
562ede208000-562ede209000 rw-p 00005000 fd:00 2500831                    /usr/bin/dvbv5-scan
562ede46c000-562ede48d000 rw-p 00000000 00:00 0                          [heap]
7f69e8000000-7f69e8021000 rw-p 00000000 00:00 0 
7f69e8021000-7f69ec000000 ---p 00000000 00:00 0 
7f69ecbd8000-7f69f3777000 r--p 00000000 fd:00 2491825                    /usr/lib/locale/locale-archive
7f69f3777000-7f69f377a000 r-xp 00000000 fd:00 2498865                    /usr/lib64/libdl-2.25.so
7f69f377a000-7f69f3979000 ---p 00003000 fd:00 2498865                    /usr/lib64/libdl-2.25.so
7f69f3979000-7f69f397a000 r--p 00002000 fd:00 2498865                    /usr/lib64/libdl-2.25.so
7f69f397a000-7f69f397b000 rw-p 00003000 fd:00 2498865                    /usr/lib64/libdl-2.25.so
7f69f397b000-7f69f39ec000 r-xp 00000000 fd:00 2499369                    /usr/lib64/libpcre.so.1.2.9
7f69f39ec000-7f69f3bec000 ---p 00071000 fd:00 2499369                    /usr/lib64/libpcre.so.1.2.9
7f69f3bec000-7f69f3bed000 r--p 00071000 fd:00 2499369                    /usr/lib64/libpcre.so.1.2.9
7f69f3bed000-7f69f3bee000 rw-p 00072000 fd:00 2499369                    /usr/lib64/libpcre.so.1.2.9
7f69f3bee000-7f69f3c07000 r-xp 00000000 fd:00 2499316                    /usr/lib64/libpthread-2.25.so
7f69f3c07000-7f69f3e07000 ---p 00019000 fd:00 2499316                    /usr/lib64/libpthread-2.25.so
7f69f3e07000-7f69f3e08000 r--p 00019000 fd:00 2499316                    /usr/lib64/libpthread-2.25.so
7f69f3e08000-7f69f3e09000 rw-p 0001a000 fd:00 2499316                    /usr/lib64/libpthread-2.25.so
7f69f3e09000-7f69f3e0d000 rw-p 00000000 00:00 0 
7f69f3e0d000-7f69f3e23000 r-xp 00000000 fd:00 2495772                    /usr/lib64/libgcc_s-7-20170915.so.1
7f69f3e23000-7f69f4022000 ---p 00016000 fd:00 2495772                    /usr/lib64/libgcc_s-7-20170915.so.1
7f69f4022000-7f69f4023000 r--p 00015000 fd:00 2495772                    /usr/lib64/libgcc_s-7-20170915.so.1
7f69f4023000-7f69f4024000 rw-p 00016000 fd:00 2495772                    /usr/lib64/libgcc_s-7-20170915.so.1
7f69f4024000-7f69f4028000 r-xp 00000000 fd:00 2500152                    /usr/lib64/libcap.so.2.25
7f69f4028000-7f69f4227000 ---p 00004000 fd:00 2500152                    /usr/lib64/libcap.so.2.25
7f69f4227000-7f69f4228000 r--p 00003000 fd:00 2500152                    /usr/lib64/libcap.so.2.25
7f69f4228000-7f69f4229000 rw-p 00004000 fd:00 2500152                    /usr/lib64/libcap.so.2.25
7f69f4229000-7f69f424e000 r-xp 00000000 fd:00 2499789                    /usr/lib64/libselinux.so.1
7f69f424e000-7f69f444d000 ---p 00025000 fd:00 2499789                    /usr/lib64/libselinux.so.1
7f69f444d000-7f69f444e000 r--p 00024000 fd:00 2499789                    /usr/lib64/libselinux.so.1
7f69f444e000-7f69f444f000 rw-p 00025000 fd:00 2499789                    /usr/lib64/libselinux.so.1
7f69f444f000-7f69f4451000 rw-p 00000000 00:00 0 
7f69f4451000-7f69f4467000 r-xp 00000000 fd:00 2499409                    /usr/lib64/libresolv-2.25.so
7f69f4467000-7f69f4666000 ---p 00016000 fd:00 2499409                    /usr/lib64/libresolv-2.25.so
7f69f4666000-7f69f4667000 r--p 00015000 fd:00 2499409                    /usr/lib64/libresolv-2.25.so
7f69f4667000-7f69f4668000 rw-p 00016000 fd:00 2499409                    /usr/lib64/libresolv-2.25.so
7f69f4668000-7f69f466a000 rw-p 00000000 00:00 0 
7f69f466a000-7f69f4671000 r-xp 00000000 fd:00 2499441                    /usr/lib64/librt-2.25.so
7f69f4671000-7f69f4870000 ---p 00007000 fd:00 2499441                    /usr/lib64/librt-2.25.so
7f69f4870000-7f69f4871000 r--p 00006000 fd:00 2499441                    /usr/lib64/librt-2.25.so
7f69f4871000-7f69f4872000 rw-p 00007000 fd:00 2499441                    /usr/lib64/librt-2.25.so
7f69f4872000-7f69f4a39000 r-xp 00000000 fd:00 2498769                    /usr/lib64/libc-2.25.so
7f69f4a39000-7f69f4c39000 ---p 001c7000 fd:00 2498769                    /usr/lib64/libc-2.25.so
7f69f4c39000-7f69f4c3d000 r--p 001c7000 fd:00 2498769                    /usr/lib64/libc-2.25.so
7f69f4c3d000-7f69f4c3f000 rw-p 001cb000 fd:00 2498769                    /usr/lib64/libc-2.25.so
7f69f4c3f000-7f69f4c43000 rw-p 00000000 00:00 0 
7f69f4c43000-7f69f4c7f000 r-xp 00000000 fd:00 2500808                    /usr/lib64/libdvbv5.so.0.0.0
[... cut ...]
7f69f53c9000-7f69f53cb000 rw-p 00027000 fd:00 2502156                    /usr/lib64/ld-2.25.so
7ffc1cb28000-7ffc1cb49000 rw-p 00000000 00:00 0                          [stack]
7ffc1cbed000-7ffc1cbf0000 r--p 00000000 00:00 0                          [vvar]
7ffc1cbf0000-7ffc1cbf2000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted (core dumped)

If I try the same thing with my patched official drivers then everything works correctly.

drphilth, 2017-10-19:
I suspect this is the same issue I am having - same card, same errors when trying to tune with w_scan. My issue is listed at https://www.tbsdtv.com/forum/viewtopic.php?f=87&t=24704

I can upload the zip file of an old version of the  ljalves driver that does work on my hardware if that is helpful for anyone?

drphilth, 2017-10-27:
@petem67 Does commit the most recent fix (https://github.com/tbsdtv/linux_media/commit/cf382935d479868da551bf1593b3ee24afd5d54d) solve your issue?

petem67, 2017-10-29:
@drphilth Yes I have just tried that fix and the tuners are now working correctly!