tbsdtv/linux_media #81: Compile error: cxd2820r_core.c:681:18: error: 'struct gpio_chip' has no member named 'dev'

Martin-Weiss, 2017-10-03:
Hi *,

when trying to compile I get this error:

```
  CC [M]  /root/tbs-open/media_build/v4l/cxd2820r_c.o
/root/tbs-open/media_build/v4l/cxd2820r_core.c: In function 'cxd2820r_probe':
/root/tbs-open/media_build/v4l/cxd2820r_core.c:681:18: error: 'struct gpio_chip' has no member named 'dev'
   priv->gpio_chip.dev = &client->dev;
                  ^
/usr/src/linux-4.4.82-6.9/scripts/Makefile.build:259: recipe for target '/root/tbs-open/media_build/v4l/cxd2820r_core.o' failed
make[5]: *** [/root/tbs-open/media_build/v4l/cxd2820r_core.o] Error 1
make[5]: *** Waiting for unfinished jobs....
/usr/src/linux-4.4.82-6.9/Makefile:1431: recipe for target '_module_/root/tbs-open/media_build/v4l' failed
make[4]: *** [_module_/root/tbs-open/media_build/v4l] Error 2
Makefile:150: recipe for target 'sub-make' failed
make[3]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make[2]: *** [__sub-make] Error 2
make[2]: Leaving directory '/usr/src/linux-4.4.82-6.9-obj/x86_64/default'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/root/tbs-open/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2
```

Any idea what might be causing this?

Thanks!
Martin

crazycat69, 2017-10-05:
What linux distro used ? Look like some wrong with kernel header.
Drivers compiled without any problems on 4.4 kernels, including cxd2820r module.

Martin-Weiss, 2017-10-06:
I see this issue on openSUSE 42.3 and SLES 12 SP3 using kernel version "kernel-default-4.4.82-6.9.1.x86_64".

In the meantime I got some help from TBS support and it seems that the compile issue is solved in this package: http://www.tbsdtv.com/download/document/linux/media_build-susu-4.4.87.tar.bz2 however - I have no idea what the differences are included, there.

Is there a chance to incorporate the adjustments in that package also here?