tbsdtv/linux_media #82: Generic Stream support?

racerxdl, 2017-10-04:
I wonder if Generic Stream support is planned (or supported, couldn't find it) for the linux drivers.

I have a GOES-16 Satellite Ingestor for GRB Stream that uses Generic Stream and would be good to have a way to feed the stream to my ingestor.

crazycat69, 2017-10-05:
Official V4L DVB can handle only DVB TS or ATSC TS.
Proper Generic continous stream support require specialized hardware (some stream processor) which handle special extra signaling from demod (start of frame, frame info, padding).
Current PC DVB devices just pass all data from TS bus via some DMA transfer.

racerxdl, 2017-10-05:
Is there any spec that I can follow to implement it myself?

Btw there are few TBS cards that supports Generic Stream just fine.

weather01089, 2017-10-06:
Crazycat, I sent him a large file saved with TBSrecorder in GSE mode with at TBS6903. The file was error free and contained some nice data from GOES16. TBS must be doing something with that windows driver so it works.

crazycat69, 2017-10-06:
No any specific GSE mode, generic continous stream always in frame-mode streaming. GS mode checkbox just force frame-mode streaming for DVB-S2 transport or generic paketized stream ( disable packet delineator.

Parsing frame-mode stream must be performed in user-space. So TBS provide hardware + BDA driver. Next - is your application. So please stop this disscussion not related to V4L DVB.

closed 2017-10-06 by crazycat69

racerxdl, 2017-10-30:
So getting back to V4L DVB. I want to dump raw TS data. I tried with dvbsnoop:
```bash
sudo dvbsnoop -s ts -tsraw -b > test.bin
```
How can I be sure that is outputing the inner data from BBFrames? 

I checked for CCSDS Frame Sync mark (0x1A 0xCF 0xFC 0x1D) and I can find it several times, the content looks like its working, I just don't know if anything is being inserted or removed by dvbsnoop or linux-media driver. As I understood the linux-media just passthrough everything to userspace, that's where dvbsnoop start playing with. Am I right?

crazycat69, 2017-10-30:
V4L DVB core pass only TS packets. Some experimental BBFrame support available in Updatelee linux_media ( but i think only saa7160 based cards like 6925, 6983 ) - https://bitbucket.org/updatelee/v4l-updatelee
Some experimental dvbsnoop with BBFrame dumping support in my repo - https://bitbucket.org/CrazyCat/dvbsnoop

This all unofficial, experimental and not supported by TBS.  Based on original patches for old linux-tbs-drivers for 6925. Read more about this old story :-) 
https://www.linuxtv.org/pipermail/linux-dvb/2007-December/022217.html
http://www.spinics.net/lists/linux-media/msg42112.html

racerxdl, 2017-10-30:
Sure, I will try it! I have a TBS 6903 card, I can also try to reverse engineer the windows driver to check how they enable the raw framing.

Thanks!

crazycat69, 2017-10-30:
Windows driver is complete different story :-) All DMA traffic passed to user-space.
You can modify tbsecp3 adapter driver and disable TS sync check in DMA tasklet. But first need BBFrame filtering support in dvb-core.

racerxdl, 2017-10-30:
Sure, I will try that. I was checking yesterday few old patches that was around the mail lists that was using a tsout module parameter to passthrough everything. I might adopt a similar behaviour (I haven't checked how the updateleeo implements that)