tbsdtv/linux_media #84: TBS6902 and clone TBS6922 do not work correctly with no lnb settings

bastard79, 2017-10-13:
Hi, i'm using tt budget s2-4100 (clone TBS6922) with your latest drivers on ubuntu 17.04 64bit with tvheadend and kodi and everything works ok, few days ago i bought TBS 6902 to connect 2 more tvs and it doesnt work properly - there is no lnb option for both adapters. 
when i plugin booth card, i have only lnb option for TBS6922, TBS6902 no lnb option :(, when i plugin only TBS6902 i have still one lnb option i think it might be drivers problem.

more information in tvheadend page https://tvheadend.org/boards/4/topics/27660?r=28839#message-28839

please help to solve this issue.

crazycat69, 2017-10-13:
Hi,
Please provide kernel log (dmesg). Some wrong on frontend attach.

P.S. saa716x and tbsecp3 drivers have some issues with current not-LTS kernels >4.9. I plan install and verify *buntu 17.04, but no time since spring 2017 :-) So i recomend LTS Ubuntu 16.04 or 14.04.5 with 4.4 kernel. Or Debian 9 with 4.9 kernel.

bastard79, 2017-10-13:
here is full dmesg https://pastebin.com/wAztuwcy

may i install only lts kernel from latest lts ubuntu 16.04?  i can install from scratch 16.04 or debian 9 but there is no sure to resolve my problem. i bought tbs6902 becasue i thought it working with exist TBS6922 clone with any issue on the same compiled drivers

bastard79, 2017-10-13:
installed 4.4 and 4.9, compiled drivers for 4.4 result is the same as in 4.10
full dmesg https://pastebin.com/AF6B6pFR
now compiling for 4.9

crazycat69, 2017-10-13:
I not see any errors in logs, all three dvb adapter registered. You not see LNB option for adapter 2 because you configure it as slave to adapter 1.

crazycat69, 2017-10-13:
You need fix adapter enumeration with module options. Put this lines to /etc/modprobe.d/dvb.conf:
```
options saa716x_core adapter_nr=0
options tbsecp3 adapter_nr=1,2
```

bastard79, 2017-10-15:
I SOLVE THE ISSUE !!!
https://tvheadend.org/boards/4/topics/27660?r=28896#message-28896

crazycat69, 2017-10-15:
Ok :-)

closed 2017-10-15 by crazycat69

steven1758, 2026-02-06:
> 老师们，我是新手，刚买tbs6902，有教程安装吗？

https://www.tbsiptv.com/help_center/  windows 和 linux dou you 安装指导