tbsdtv/linux_media #87: driver and firmware upstreaming

cz172638, 2017-11-13:
is there anything in the way of upstreaming drivers to mainline kernel?
devel-like way of building drivers against running kernel is anoying.
same for firmware blobs.
can they be put to linux-firmware repo?

CvH, 2017-11-13:
Us was told (as these repo went up) these driver should be up-streamed to get proper mainline support, sadly nothing happens till date :( 
At the meantime every bigger vendor besides TBS has their drivers at the kernel (even DD drivers are at the kernel due help of hobbyists). The only other missing one, Hauppauge, is currently semi official starting to push their drivers.
I understand that not every driver is able to get upstream due NDAs, copyright problems and whatnot, but zero ? 
Its a mess for the users, its a mess downstream and a clear competitive disadvantage if the hardware is not supported out of the box at Linux and "every" other available hardware works ootb.

CHBMB, 2017-12-18:
@CvH I agree wholeheartedly, and it's a shame because I really like the TBS hardware.

Hoeze, 2017-12-19:
I'll have to buy four new tuners in the next year and I am seriously considering buying other cars (not from TBS) because I am really annoyed from compiling the TBS driver on every kernel update.

CvH, 2018-01-16:
"All" Hauppauge devices are now soon to get merged at mainline kernel, so TBS is now the only vendor (besides those ultra cheep crap) that has no proper Linux support :( (installing an out of tree driver is no proper support)
I had hoped that someone who is in charge at least post a statement if this is still the goal or if this ends up like the old binary driver.

Hoeze, 2018-10-25:
Are there any updates on this?