tbsdtv/linux_media #88: No way to select ISDB-T network under TVHeadend because driver uses old Linux DVBAPI

mpaolino, 2017-11-13:
I have a TBS6814 and I have installed and tested the last driver available with MythTV and it works fine.
I'm using kernel 4.8.0-30-generic on a 64bit Ubuntu.

The problem is that it doesn't work with TVHeadend because the drivers report the card as a DVB-T card, and the software does not allow to select a ISDB-T network to use. Apparetky this is because the drivers are using an old Linux DVBAPI. BTW, MythTV also reports the card as DVB-T.

[image]

[The related issue in TVHeadend bug tracker and response from developers](https://tvheadend.org/issues/4146).

crazycat69, 2017-11-13:
You use old semi-closed source linux-tbs-drivers.
Install open-source tbsdtv drivers and your 6814 must  be ISDB-T tuner + modern DVB-API 5.10

mpaolino, 2017-11-13:
[I don't see it in the list of supported cards](https://github.com/tbsdtv/linux_media/wiki), but I'll give it a try and let you know.

crazycat69, 2017-11-13:
added to wiki list :)

mpaolino, 2017-11-14:
I can confirm it works, thanks.

closed 2017-11-14 by mpaolino