tbsdtv/linux_media #89: Fails to build with 4.14 kernel

belliash, 2017-11-27:
/usr/src/media_build/v4l/vim2m.c: In function 'job_ready':
/usr/src/media_build/v4l/vim2m.c:353:41: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  if (v4l2_m2m_num_src_bufs_ready(ctx->fh.m2m_ctx) < ctx->translen
                                         ^
/usr/src/media_build/v4l/vim2m.c:354:44: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
      || v4l2_m2m_num_dst_bufs_ready(ctx->fh.m2m_ctx) < ctx->translen) {
                                            ^
/usr/src/media_build/v4l/vim2m.c: In function 'device_run':
/usr/src/media_build/v4l/vim2m.c:382:41: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  src_buf = v4l2_m2m_next_src_buf(ctx->fh.m2m_ctx);
                                         ^
/usr/src/media_build/v4l/vim2m.c:383:41: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  dst_buf = v4l2_m2m_next_dst_buf(ctx->fh.m2m_ctx);
                                         ^
/usr/src/media_build/v4l/vim2m.c: In function 'device_isr':
/usr/src/media_build/v4l/vim2m.c:405:47: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  src_vb = v4l2_m2m_src_buf_remove(curr_ctx->fh.m2m_ctx);
                                               ^
/usr/src/media_build/v4l/vim2m.c:406:47: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  dst_vb = v4l2_m2m_dst_buf_remove(curr_ctx->fh.m2m_ctx);
                                               ^
/usr/src/media_build/v4l/vim2m.c:419:55: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
   v4l2_m2m_job_finish(vim2m_dev->m2m_dev, curr_ctx->fh.m2m_ctx);
                                                       ^
/usr/src/media_build/v4l/vim2m.c: In function 'vidioc_g_fmt':
/usr/src/media_build/v4l/vim2m.c:486:30: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  vq = v4l2_m2m_get_vq(ctx->fh.m2m_ctx, f->type);
                              ^
/usr/src/media_build/v4l/vim2m.c: In function 'vidioc_s_fmt':
/usr/src/media_build/v4l/vim2m.c:593:30: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  vq = v4l2_m2m_get_vq(ctx->fh.m2m_ctx, f->type);
                              ^
/usr/src/media_build/v4l/vim2m.c: In function 'vim2m_buf_queue':
/usr/src/media_build/v4l/vim2m.c:787:28: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  v4l2_m2m_buf_queue(ctx->fh.m2m_ctx, vbuf);
                            ^
/usr/src/media_build/v4l/vim2m.c: In function 'vim2m_stop_streaming':
/usr/src/media_build/v4l/vim2m.c:807:42: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
    vbuf = v4l2_m2m_src_buf_remove(ctx->fh.m2m_ctx);
                                          ^
/usr/src/media_build/v4l/vim2m.c:809:42: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
    vbuf = v4l2_m2m_dst_buf_remove(ctx->fh.m2m_ctx);
                                          ^
/usr/src/media_build/v4l/vim2m.c: In function 'vim2m_open':
/usr/src/media_build/v4l/vim2m.c:926:9: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  ctx->fh.m2m_ctx = v4l2_m2m_ctx_init(dev->m2m_dev, ctx, &queue_init);
         ^
/usr/src/media_build/v4l/vim2m.c:928:20: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  if (IS_ERR(ctx->fh.m2m_ctx)) {
                    ^
/usr/src/media_build/v4l/vim2m.c:929:23: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
   rc = PTR_ERR(ctx->fh.m2m_ctx);
                       ^
In file included from /usr/src/media_build/v4l/../linux/include/media/v4l2-subdev.h:24:0,
                 from /usr/src/media_build/v4l/../linux/include/media/v4l2-device.h:25,
                 from /usr/src/media_build/v4l/vim2m.c:28:
/usr/src/media_build/v4l/vim2m.c:941:15: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
   ctx, ctx->fh.m2m_ctx);
               ^
/usr/src/media_build/v4l/../linux/include/media/v4l2-common.h:69:44: note: in definition of macro 'v4l2_printk'
  printk(level "%s: " fmt, (dev)->name , ## arg)
                                            ^~~
/usr/src/media_build/v4l/vim2m.c:71:2: note: in expansion of macro 'v4l2_dbg'
  v4l2_dbg(1, debug, &dev->v4l2_dev, "%s: " fmt, __func__, ## arg)
  ^~~~~~~~
/usr/src/media_build/v4l/vim2m.c:940:2: note: in expansion of macro 'dprintk'
  dprintk(dev, "Created instance: %p, m2m_ctx: %p\n",
  ^~~~~~~
/usr/src/media_build/v4l/vim2m.c: In function 'vim2m_release':
/usr/src/media_build/v4l/vim2m.c:959:30: error: 'struct v4l2_fh' has no member named 'm2m_ctx'
  v4l2_m2m_ctx_release(ctx->fh.m2m_ctx);
                              ^
make[3]: *** [scripts/Makefile.build:321: /usr/src/media_build/v4l/vim2m.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1503: _module_/usr/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-4.14.2-gentoo'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/usr/src/media_build/v4l'
make: *** [Makefile:26: all] Error 2

crazycat69, 2017-12-06:
check with updated media_build and media_tree

CvH, 2017-12-07:
4.14.0 and 4.14.4 worked without an problem here

CHBMB, 2017-12-18:
Yeah, working for me too on `4.14.4`

kunxiang1, 2018-03-21:
same error on 4.14.23. how to fix it?I have updated media_build and media_tree,still the error

kunxiang1, 2018-03-22:
I find out how to fix it now,with this patch:
--- vim2m.c.orig	2018-03-05 18:38:24.000000000 +0800
+++ vim2m.c	2018-03-22 23:21:05.908814972 +0800
@@ -29,6 +29,7 @@
 #include <media/v4l2-ioctl.h>
 #include <media/v4l2-ctrls.h>
 #include <media/v4l2-event.h>
+#include <media/v4l2-fh.h>
 #include <media/videobuf2-vmalloc.h>
 
 MODULE_DESCRIPTION("Virtual device for mem2mem framework testing");

--- v4l2-fh.h.orig	2018-03-13 15:29:35.000000000 +0800
+++ v4l2-fh.h	2018-03-22 23:19:39.831736509 +0800
@@ -57,9 +57,8 @@
 	unsigned int		navailable;
 	u32			sequence;
 
-#if IS_ENABLED(CONFIG_V4L2_MEM2MEM_DEV)
 	struct v4l2_m2m_ctx	*m2m_ctx;
-#endif
+
 };
 
 /**